UZ-SLAMLab/ORB_SLAM3 #1: whether ORB_SLAM3 would support the ROS

Sunchanghao, 2020-07-24:
looking forward to!

saimouli, 2020-07-24:
@Sunchanghao I see that authors have already provided a ROS example: https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/master/Examples/ROS/ORB_SLAM3/src/ros_stereo.cc

Sunchanghao, 2020-07-24:
Wow  Thx!

| |
孙长浩
|
|
邮箱：sunchanghaosch@163.com
|

签名由 网易邮箱大师 定制

On 07/24/2020 23:36, Sai wrote:

@Sunchanghao I see that authors have already provided a ROS example: https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/master/Examples/ROS/ORB_SLAM3/src/ros_stereo.cc

dimaxano, 2020-07-24:
As I see this code uses `ORB_SLAM2` class, not `ORB_SLAM3`.

SteveMacenski, 2020-07-24:
Also support for ROS2 (I'm maintainer on ROS2 navigation2, would love to do testing with this!)

Sunchanghao, 2020-07-25:
> As I see this code uses `ORB_SLAM2` class, not `ORB_SLAM3`.

Have you tested the effect?

Sunchanghao, 2020-07-25:
> Also support for ROS2 (I'm maintainer on ROS2 navigation2, would love to do testing with this!) would be great. I'd like to make formally supported integrations with this and ROS2 with navigation2 (https://github.com/ros-planning/navigation2).
> 
> I see a few ORB-SLAM2 references that weren't changed: https://github.com/UZ-SLAMLab/ORB_SLAM3/search?p=1&q=ORB_SLAM2&unscoped_q=ORB_SLAM2
> 
> Do you intend to support this and improve on it or is this a one-and-done research project? I ask because if we do some integrations with navigation2 we'll likely make improvements and want to know if we should fork this for the ROS world or submit PRs back here if there's support.

good job! 
therefor, the interface in the current vision is ORB_SLAM2?

zhijianglu, 2020-07-25:
just change ORBSLAM2::System into ORBSLAM3::System, I can compile the code by catkin_make

Sunchanghao, 2020-07-25:
> just change ORBSLAM2::System into ORBSLAM3::System, I can compile the code by catkin_make

哈 那效果呢 是新版本的效果了吧 不知道会不会有什么问题

liuzhenboo, 2020-07-26:
ros代码里面的那些命名空间还是ORBSLAM2

Sunchanghao, 2020-07-26:
> ros代码里面的那些命名空间还是ORBSLAM2

是啊 所以不知道啥意思

liuzhenboo, 2020-07-26:
ORBSLAM3 haven't provide source code of vio based ros.

liuzhenboo, 2020-07-26:
关于VIO的ros代码作者没有提供；不过由于兼容了ORBSLAM2的一些函数，所以orbslam3中基于ros的视觉里程计代码理论上还是能跑的，不过要改一些变量名，命名空间杀啥的。。

Sunchanghao, 2020-07-26:
> 关于VIO的ros代码作者没有提供；不过由于兼容了ORBSLAM2的一些函数，所以orbslam3中基于ros的视觉里程计代码理论上还是能跑的，不过要改一些变量名，命名空间杀啥的。。

好的吧 再等等先

zang09, 2020-07-27:
I change the code 'ORBSLAM2' to 'ORBSLAM3' and success build and launch.
But there is no mono-inertial or stereo-inertial code in Example/ROS

Sunchanghao, 2020-07-27:
> I change the code 'ORBSLAM2' to 'ORBSLAM3' and success build and launch.
> But there is no mono-inertial or stereo-inertial code in Example/ROS

There also seems to be no odometer~

xiefei2929, 2020-07-27:
I commited two visual-inertial ROS interfaces Mono_inertial and Stereo_inertial at https://github.com/xiefei2929/ORB_SLAM3-ROS-Interface

closed 2020-08-10 by richard-elvira