UZ-SLAMLab/ORB_SLAM3 #3: Does anyone run the demo without 'Segmentation fault (core dumped)' error?

anakin0620, 2020-07-24:
Does anyone run the demo without 'Segmentation fault (core dumped)' error? 

Seems like an issue with the OpenCV version (currently OpenCV3.4 and Ubuntu18.04). 

Is there anyone here tested it successfully with Ubuntu18.04 and OpenCV3.4?

dimaxano, 2020-07-24:
Yep, run Euroc example successfully. Post, please, orb-slam output here

mirellameelo, 2020-07-24:
@anakin0620 I just built with Opencv 3.4.0 and Ubuntu 18.04. All good. 
Then I downloaded the MH_01_easy dataset (EuRoC) and pointed to its folder (just like the README say to do), but nothing happens after executing ./euroc_examples.sh 
@dimaxano could you please confirm the folder you pointed to in "euroc_examples.sh"?
Tks in advance.

anakin0620, 2020-07-24:
@dimaxano 
Here is the output. My error was from mono_cc. 

./Examples/Monocular/mono_tum /home/pt/ORB3/ORB_SLAM3/Vocabulary/ORBvoc.txt /home/pt/ORB3/ORB_SLAM3/Examples/Monocular/TUM1.yaml /home/pt/Downloads/rgbd_dataset_freiburg1_desk 

ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Monocular

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name: 

Camera Parameters: 
- fx: 517.306
- fy: 516.469
- cx: 318.643
- cy: 255.314
- bf: 0
- k1: 0
- k2: 0
- p1: 0
- p2: 0
- k3: 1.16331
- fps: 30
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 1000
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

-------
Start processing sequence ...
Images in the sequence: 613

Segmentation fault (core dumped)

anakin0620, 2020-07-25:
[New Thread 0x7fffc672f700 (LWP 87287)]
[New Thread 0x7fffc5f2e700 (LWP 87288)]
[New Thread 0x7fffc572d700 (LWP 87289)]

Failed to load image at: /home/Downloads/MH01/mav0/cam0/data/1403636579763555584.png
/home/ORB3/ORB_SLAM3/Examples/Monocular/mono_euroc: symbol lookup error: /usr/lib/x86_64-linux-gnu/libGLX_mesa.so.0: undefined symbol: XGetXCBConnection
[Thread 0x7fffc572d700 (LWP 87289) exited]
[Thread 0x7fffc5f2e700 (LWP 87288) exited]
[Thread 0x7fffc672f700 (LWP 87287) exited]
[Inferior 1 (process 87282) exited with code 0177]
(gdb)

anakin0620, 2020-07-25:
@dimaxano 
[New Thread 0x7fffc672f700 (LWP 87287)]
[New Thread 0x7fffc5f2e700 (LWP 87288)]
[New Thread 0x7fffc572d700 (LWP 87289)]

Failed to load image at: /home/Downloads/MH01/mav0/cam0/data/1403636579763555584.png
/home/ORB3/ORB_SLAM3/Examples/Monocular/mono_euroc: symbol lookup error: /usr/lib/x86_64-linux-gnu/libGLX_mesa.so.0: undefined symbol: XGetXCBConnection
[Thread 0x7fffc572d700 (LWP 87289) exited]
[Thread 0x7fffc5f2e700 (LWP 87288) exited]
[Thread 0x7fffc672f700 (LWP 87287) exited]
[Inferior 1 (process 87282) exited with code 0177]
(gdb)

michaelnguyen11, 2020-07-25:
coredump happens when run stereo_kitti, while stereo_inertial_tum_tum_vi run successfully.

```
#0  0x00007f679f40cfe4 in ORB_SLAM3::EdgeSE3ProjectXYZOnlyPose::computeError() ()
    at /media/hiep/Ubuntu/working/SLAM/ORB_SLAM3/lib/libORB_SLAM3.so
#1  0x00007f679d2eed61 in g2o::SparseOptimizer::computeActiveErrors() ()
    at /media/hiep/Ubuntu/working/SLAM/ORB_SLAM3/Thirdparty/g2o/lib/libg2o.so
#2  0x00007f679d2f831e in g2o::OptimizationAlgorithmLevenberg::solve(int, bool) ()
    at /media/hiep/Ubuntu/working/SLAM/ORB_SLAM3/Thirdparty/g2o/lib/libg2o.so
#3  0x00007f679d2f02ad in g2o::SparseOptimizer::optimize(int, bool) ()
    at /media/hiep/Ubuntu/working/SLAM/ORB_SLAM3/Thirdparty/g2o/lib/libg2o.so
#4  0x00007f679f3d2d29 in ORB_SLAM3::Optimizer::PoseOptimization(ORB_SLAM3::Frame*) ()
    at /media/hiep/Ubuntu/working/SLAM/ORB_SLAM3/lib/libORB_SLAM3.so
#5  0x00007f679f322a2e in ORB_SLAM3::Tracking::TrackReferenceKeyFrame() () at /media/hiep/Ubuntu/working/SLAM/ORB_SLAM3/lib/libORB_SLAM3.so
```

12138youbrotherdong, 2020-07-25:
I also have this kind of mistake, how to solve it?

Ckq-Sugar, 2020-07-25:
when I run rgb_d datasets(rgbd_dataset_freiburg1_xyz), I also had the same mistake?

./Examples/RGB-D/rgbd_tum ./Vocabulary/ORBvoc.txt ./Examples/RGB-D/TUM1.yaml /home/mk/data/rgbd_dataset_freiburg1_xyz /home/mk/data/rgbd_dataset_freiburg1_xyz/associations.txt
ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: RGB-D

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name: 

Camera Parameters: 
- fx: 517.306
- fy: 516.469
- cx: 318.643
- cy: 255.314
- bf: 40
- k1: 0
- k2: 0
- p1: 0
- p2: 0
- k3: 1.16331
- fps: 30
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 1000
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

Depth Threshold (Close/Far Points): 3.09294

-------
Start processing sequence ...
Images in the sequence: 792

Segmentation fault (core dumped)

anakin0620, 2020-07-25:
@Ckq-Sugar I have the same output as yours.

anakin0620, 2020-07-26:
#8 similar problem here

aiyouaiyouhehehe, 2020-07-26:
the same problem

squaresoft2015, 2020-07-26:
I encounter the same problem:
ailed to load image at: ~/EuRoC/MH01/mav0/cam0/data/1403636579763555584.png
./euroc_examples.sh: line 7:  2782 Segmentation fault      (core dumped) ./Examples/Monocular/mono_euroc ./Vocabulary/ORBvoc.txt ./Examples/Monocular/EuRoC.yaml "$pathDatasetEuroc"/MH01 ./Examples/Monocular/EuRoC_TimeStamps/MH01.txt dataset-MH01_mono
Launching MH02 with Monocular sensor
num_seq = 1
file name: dataset-MH02_mono
Loading images for sequence 0...LOADED!
It appears that the program does not read the image sequence correctly. The image names contained in file "MH01.txt" do not match with the actual file names in EuRoC dataset. Do I need to reset the file content manually?

yan99033, 2020-07-27:
Hi everyone,

I think I just found the error that causing the `Segmentation fault (core dumped)`. I ran valgrind memcheck and found that the code was accessing an invalid memory address.

Specifically, it was caused by `static_cast<Pinhole*>(mpCamera)->toK();` when instantiating a `Frame` (` mCurrentFrame = Frame(mImGray,timestamp,mpIniORBextractor,mpORBVocabulary,mpCamera,mDistCoef,mbf,mThDepth);` in Tracking.cc). 

To solve it, you need to add `Camera.type: "PinHole"` (assuming you are using PinHole camera model) in the yaml file in which the camera parameters are kept so that the initialization of `Tracking` would also initialize a `Pinhole` for camera tracking.

anakin0620, 2020-07-27:
> Hi everyone,
>
>
>
> I think I just found the error that causing the `Segmentation fault (core dumped)`. I ran valgrind memcheck and found that the code was accessing an invalid memory address. [...]

Thank you! I will test it later and close this. It's really helpful for many of us experienced these errors :)

anakin0620, 2020-07-28:
Add Camera.type: "PinHole" (assuming you are using PinHole camera model) in the yaml file in which the camera parameters are kept so that the initialization of Tracking would also initialize a Pinhole for camera tracking.
Thanks to @yan99033  ！

closed 2020-07-28 by anakin0620

BeardyMan37, 2020-09-05:
> I encounter the same problem:
> ailed to load image at: ~/EuRoC/MH01/mav0/cam0/data/1403636579763555584.png
> ./euroc_examples.sh: line 7: 2782 Segmentation fault (core dumped) [...]

"~/EuRoC/MH01/mav0/cam0/data/1403636579763555584.png" This image path won't work because bash doesn't recognize ~. You have to use cannonical path.