UZ-SLAMLab/ORB_SLAM3 #6: TUM_VI Datasheet run problem in Monocular-Inertial mode

shanpenghui, 2020-07-25:
I try to run the example of mono_inertial_tum_vi, but error come out when start the .sh file.
The error like below:
`sph@sph-System-Product-Name:~/Documents/ORB_SLAM3$ ./tum_vi.sh 
Launching Room 4 with Monocular-Inertial sensor
num_seq = 1
file name: dataset-room4_512_monoi
Loading images for sequence 0.../home/sph/Downloads/dataset-room4_512_16/mav0/cam0/data
Examples/Monocular-Inertial/TUM_TimeStamps/dataset-room4_512.txt
LOADED!
Loading IMU for sequence 0...LOADED!

-------

ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Monocular-Inertial

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name: dataset-room4_512_monoi

Camera Parameters: 
- fx: 190.97846984863281
- fy: 190.97331237792969
- cx: 254.93170166015625
- cy: 256.89743041992188
- bf: 0
- k1: 0
- k2: 0
- p1: 0
- p2: 0
- k3: -0.0020532361231744289
- fps: 20
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 1500
- Scale Levels: 8
- Scale Factor: 1.2000000476837158
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

Left camera to Imu Transform (Tbc): 
[-0.99952501, 0.0075019184, -0.029890131, 0.045574836;
 0.029615344, -0.03439736, -0.99896932, -0.071161799;
 -0.008522328, -0.99938005, 0.034158852, -0.044681255;
 0, 0, 0, 1]

IMU frequency: 200 Hz
IMU gyro noise: 0.00015999999595806003 rad/s/sqrt(Hz)
IMU gyro walk: 2.2000000171829015e-05 rad/s^2/sqrt(Hz)
IMU accelerometer noise: 0.00279999990016222 m/s^2/sqrt(Hz)
IMU accelerometer walk: 0.00085999997099861503 m/s^3/sqrt(Hz)

.pnged to load image at: /home/sph/Downloads/dataset-room4_512_16/mav0/cam0/data/1520531124150444163
./tum_vi.sh: line 17: 11368 Segmentation fault      (core dumped) ./Examples/Monocular-Inertial/mono_inertial_tum_vi Vocabulary/ORBvoc.txt Examples/Monocular-Inertial/TUM_512.yaml "$pathDatasetTUM_VI"/dataset-room4_512_16/mav0/cam0/data Examples/Monocular-Inertial/TUM_TimeStamps/dataset-room4_512.txt Examples/Monocular-Inertial/TUM_IMU/dataset-room4_512.txt dataset-room4_512_monoi
sph@sph-System-Product-Name:~/Documents/ORB_SLAM3$ ./tum_vi.sh 
Launching Room 4 with Monocular-Inertial sensor
num_seq = 1
file name: dataset-room4_512_monoi
Loading images for sequence 0.../home/sph/Downloads/dataset-room4_512_16/mav0/cam0/data
Examples/Monocular-Inertial/TUM_TimeStamps/dataset-room4_512.txt
LOADED!
Loading IMU for sequence 0...LOADED!

-------

ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Monocular-Inertial

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name: dataset-room4_512_monoi

Camera Parameters: 
- fx: 190.97846984863281
- fy: 190.97331237792969
- cx: 254.93170166015625
- cy: 256.89743041992188
- bf: 0
- k1: 0
- k2: 0
- p1: 0
- p2: 0
- k3: -0.0020532361231744289
- fps: 20
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 1500
- Scale Levels: 8
- Scale Factor: 1.2000000476837158
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

Left camera to Imu Transform (Tbc): 
[-0.99952501, 0.0075019184, -0.029890131, 0.045574836;
 0.029615344, -0.03439736, -0.99896932, -0.071161799;
 -0.008522328, -0.99938005, 0.034158852, -0.044681255;
 0, 0, 0, 1]

IMU frequency: 200 Hz
IMU gyro noise: 0.00015999999595806003 rad/s/sqrt(Hz)
IMU gyro walk: 2.2000000171829015e-05 rad/s^2/sqrt(Hz)
IMU accelerometer noise: 0.00279999990016222 m/s^2/sqrt(Hz)
IMU accelerometer walk: 0.00085999997099861503 m/s^3/sqrt(Hz)

.pnged to load image at: /home/sph/Downloads/dataset-room4_512_16/mav0/cam0/data/1520531124150444163
./tum_vi.sh: line 17: 11882 Segmentation fault      (core dumped) ./Examples/Monocular-Inertial/mono_inertial_tum_vi Vocabulary/ORBvoc.txt Examples/Monocular-Inertial/TUM_512.yaml "$pathDatasetTUM_VI"/dataset-room4_512_16/mav0/cam0/data Examples/Monocular-Inertial/TUM_TimeStamps/dataset-room4_512.txt Examples/Monocular-Inertial/TUM_IMU/dataset-room4_512.txt dataset-room4_512_monoi
`
The tum_vi.sh file is 
`#!/bin/bash
pathDatasetTUM_VI='/home/sph/Downloads' #Example, it is necesary to change it by the dataset path
#------------------------------------
# Monocular-Inertial Examples
echo "Launching Room 4 with Monocular-Inertial sensor"
./Examples/Monocular-Inertial/mono_inertial_tum_vi Vocabulary/ORBvoc.txt Examples/Monocular-Inertial/TUM_512.yaml "$pathDatasetTUM_VI"/dataset-room4_512_16/mav0/cam0/data Examples/Monocular-Inertial/TUM_TimeStamps/dataset-room4_512.txt Examples/Monocular-Inertial/TUM_IMU/dataset-room4_512.txt dataset-room4_512_monoi`

The exec file is in the git
https://github.com/shanpenghui/my_orbslam3.git

Thanks for helping.

richard-elvira, 2020-07-29:
We cannot replicate this bug, it works properly in our computers.

Studenthc, 2020-07-31:
I have the same problem, tested on dataset-corridor1_512_16,i found that the picture name of dataset-corridor1_512.txt can not match file name of picture folder(data).I change the path cam0 to cam1,it work.Have a try!

shanpenghui, 2020-07-31:
> I have the same problem, tested on dataset-corridor1_512_16,i found that the picture name of dataset-corridor1_512.txt can not match file name of picture folder(data).I change the path cam0 to cam1,it work.Have a try!

I solve the problems by adding a function like below, before vstrImages.push_back 
```
string lltoString(string &input)
{
    string result;
    ostringstream ss;
    ss << input;
    istringstream is(ss.str());
    is >> result;
    return result;
}
```

And then:
```
            stringstream ss;
            string rlt = lltoString(s);
            string pre = strImagePath + "/";
            rlt += ".png";
            string fi = pre + rlt;
            //cout << fi << endl;
            vstrImages.push_back(fi);
```

closed 2020-07-31 by shanpenghui

hzm8341, 2020-08-29:
change the code by follow:

string lltoString(string &input)

{

    string result;

    ostringstream ss;

    ss << input;

    istringstream is(ss.str());

    is >> result;

    return result;

}

void LoadImages(const string &strImagePath, const string &strPathTimes,

                vector<string> &vstrImages, vector<double> &vTimeStamps)

{

    ifstream fTimes;

    fTimes.open(strPathTimes.c_str());

    vTimeStamps.reserve(5000);

    vstrImages.reserve(5000);

    while(!fTimes.eof())

    {

        string s;

        getline(fTimes,s);

        if(!s.empty())

        {

            stringstream ss;

            ss << s;

            string rlt = lltoString(s);

            string pre = strImagePath + "/";

            rlt += ".png";

            string fi = pre + rlt;

            //cout << fi << endl;

            vstrImages.push_back(fi);

            double t;

            ss >> t;

            vTimeStamps.push_back(t/1e9);

        }

    }

}