UZ-SLAMLab/ORB_SLAM3 #8: errors: ORB-SLAM3 seems not work well with an RGB-D mode?

cnqiangfu, 2020-07-25:


anakin0620, 2020-07-25:
Do you mean core dumped error?

cnqiangfu, 2020-07-25:
> Do you mean core dumped error?

ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: RGB-D

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name: 

Camera Parameters: 
- fx: 520.909
- fy: 521.007
- cx: 325.141
- cy: 249.702
- bf: 40
- k1: 0
- k2: 0
- p1: 0
- p2: 0
- k3: 0.917205
- fps: 30
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 1000
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

Depth Threshold (Close/Far Points): 3.07156

-------
Start processing sequence ...
Images in the sequence: 5057

./runtum.sh: line 2: 16582 Segmentation fault      ../Examples/RGB-D/rgbd_tum ../Vocabulary/ORBvoc.txt ../Examples/RGB-D/TUM2.yaml /home/healer/TUM-RGBD/rgbd_dataset_freiburg2_large_with_loop /home/healer/TUM-RGBD/rgbd_dataset_freiburg2_large_with_loop/associations.txt

Just error when run RGB-D example.

anakin0620, 2020-07-26:
> ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
> ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José [...]

Looks like the same error.

aiyouaiyouhehehe, 2020-07-26:
the same error. T_T

richard-elvira, 2020-07-27:
ORB-SLAM3 Beta version has been only tested in Monocular, Monocular-Inertial, Stereo, and Stereo-Inertial sensors, which are the sensors that we report in the article. RGB-D sensor may not work properly, we will provide complete support to all sensor in future versions.

yan99033, 2020-07-27:
Hi @cnqiangfu, while RGB-D may not work properly, feel free to try my posted solution in #3, i.e., adding `Camera.type: "PinHole"` in the yaml file and see if it helps.

cnqiangfu, 2020-07-27:
> Hi @cnqiangfu, while RGB-D may not work properly, feel free to try my posted solution in #3, i.e., adding `Camera.type: "PinHole"` in the yaml file and see if it helps.

Hi Shing, that is so nice of you. I tried it, but it still didn't work. It does not matter, because I have turned attention to VIO ~~~

closed 2020-07-31 by richard-elvira

aiyouaiyouhehehe, 2020-08-10:
It still can not run tum1 and tum2. Could anyone tell me how to fix it？Thanks very much！

aiyouaiyouhehehe, 2020-08-10:
> 
> 
> ORB-SLAM3 Beta version has been only tested in Monocular, Monocular-Inertial, Stereo, and Stereo-Inertial sensors, which are the sensors that we report in the article. RGB-D sensor may not work properly, we will provide complete support to all sensor in future versions.

I tried tum1 and tum2 dataset in RGB-D mode but it still has an error with core dumped. Could you please tell me how to fix it? Thanks very much!!!

aiyouaiyouhehehe, 2020-08-10:
> 
> 
> ORB-SLAM3 Beta version has been only tested in Monocular, Monocular-Inertial, Stereo, and Stereo-Inertial sensors, which are the sensors that we report in the article. RGB-D sensor may not work properly, we will provide complete support to all sensor in future versions.

Hi,@richard-elvira. tum3 runs successfully. I tried tum1 and tum2 dataset in RGB-D mode but it still has an error with core dumped. Could you please tell me how to fix it? Thanks very much!!!

niumaoyihao, 2024-04-24:
> ORB-SLAM3 Beta version has been only tested in Monocular, Monocular-Inertial, Stereo, and Stereo-Inertial sensors, which are the sensors that we report in the article. RGB-D sensor may not work properly, we will provide complete support to all sensor in future versions.

You can read the TUM1.yaml and your own TUMxxx.yaml，maybe some parameters' names are different. I run successful after changing my own TUMxxx.yaml using the same param name as TUM1.yaml.