UZ-SLAMLab/ORB_SLAM3 #9: ./build.sh

mengdejiang, 2020-07-26:
CMakeFiles/Makefile2:400: recipe for target 'CMakeFiles/ORB_SLAM3.dir/all' failed
make[1]: *** [CMakeFiles/ORB_SLAM3.dir/all] Error 2

dimaxano, 2020-07-26:
Hi! These lines are insufficient to provide some help. Paste, please, the whole script output

thuliangjz, 2020-07-27:
I suggest adding a line
~~~
set -e
~~~
at the begining of the build.sh and run `bash build.sh` instead of running `./build.sh` directly so that bash stops executing upon any error. Also as @dimaxano said you should copy the entire output of cmake to let us see what's wrong with building (unsatisified requirements or compilation error etc)

belal-ibrahim, 2020-08-07:
ake[2]: *** [CMakeFiles/ORB_SLAM3.dir/build.make:102: CMakeFiles/ORB_SLAM3.dir/src/LoopClosing.cc.o] Error 1
/home/beraru/ORB_SLAM3/src/Tracking.cc: In member function ‘void ORB_SLAM3::Tracking::Track()’:
/home/beraru/ORB_SLAM3/src/Tracking.cc:1155:13: warning: ‘bOK’ may be used uninitialized in this function [-Wmaybe-uninitialized]
 1155 |             if(bOK)
      |             ^~
make[2]: *** [CMakeFiles/ORB_SLAM3.dir/build.make:232: CMakeFiles/ORB_SLAM3.dir/src/Optimizer.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:390: CMakeFiles/ORB_SLAM3.dir/all] Error 2
make: *** [Makefile:84: all] Error 2

thuliangjz, 2020-08-10:
@belal-ibrahim it seems you have trouble building LoopClosing.cc and Optimizer.cc. However I cannot see what exactly the error is, please provide the **full** output of your building script output.

belal-ibrahim, 2020-08-10:
ORB_SLAM3/src/System.cc:26:10: fatal error: openssl/md5.h: No such file or directory
   26 | #include <openssl/md5.h>
      |          ^~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/ORB_SLAM3.dir/build.make:63: CMakeFiles/ORB_SLAM3.dir/src/System.cc.o] Error 1
/home/beraru/ORB_SLAM3/src/Optimizer.cc:5179:5: warning: "/*" within comment [-Wcomment]
 5179 |     /*std::cout << " Construction time: " << t_const << std::endl;
      |      
In file included from /home/beraru/ORB_SLAM3/include/Atlas.h:27,
                 from /home/beraru/ORB_SLAM3/include/LocalMapping.h:24,
                 from /home/beraru/ORB_SLAM3/include/LoopClosing.h:24,
                 from /home/beraru/ORB_SLAM3/src/LoopClosing.cc:20:
/home/beraru/ORB_SLAM3/include/CameraModels/KannalaBrandt8.h: In constructor ‘ORB_SLAM3::KannalaBrandt8::KannalaBrandt8(std::vector<float>)’:
/home/beraru/ORB_SLAM3/include/CameraModels/KannalaBrandt8.h:105:21: warning: ‘ORB_SLAM3::KannalaBrandt8::precision’ will be initialized after [-Wreorder]
  105 |         const float precision;
      |                     ^~~~~~~~~
/home/beraru/ORB_SLAM3/include/CameraModels/KannalaBrandt8.h:95:26: warning:   ‘std::vector<int> ORB_SLAM3::KannalaBrandt8::mvLappingArea’ [-Wreorder]
   95 |         std::vector<int> mvLappingArea;
      |                          ^~~~~~~~~~~~~
In file included from /home/beraru/ORB_SLAM3/include/Atlas.h:27,
                 from /home/beraru/ORB_SLAM3/include/LocalMapping.h:24,
                 from /home/beraru/ORB_SLAM3/include/LoopClosing.h:24,
                 from /home/beraru/ORB_SLAM3/src/LoopClosing.cc:20:
/home/beraru/ORB_SLAM3/include/CameraModels/KannalaBrandt8.h:53:9: warning:   when initialized here [-Wreorder]
   53 |         KannalaBrandt8(const std::vector<float> _vParameters) : GeometricCamera(_vParameters), precision(1e-6), mvLappingArea(2,0) ,tvr(nullptr) {
      |         ^~~~~~~~~~~~~~
In file included from /home/beraru/ORB_SLAM3/include/Atlas.h:27,
                 from /home/beraru/ORB_SLAM3/include/LocalMapping.h:24,
                 from /home/beraru/ORB_SLAM3/include/LoopClosing.h:24,
                 from /home/beraru/ORB_SLAM3/src/LoopClosing.cc:20:
/home/beraru/ORB_SLAM3/include/CameraModels/KannalaBrandt8.h: In constructor ‘ORB_SLAM3::KannalaBrandt8::KannalaBrandt8(std::vector<float>, float)’:
/home/beraru/ORB_SLAM3/include/CameraModels/KannalaBrandt8.h:105:21: warning: ‘ORB_SLAM3::KannalaBrandt8::precision’ will be initialized after [-Wreorder]
  105 |         const float precision;
      |                     ^~~~~~~~~
/home/beraru/ORB_SLAM3/include/CameraModels/KannalaBrandt8.h:95:26: warning:   ‘std::vector<int> ORB_SLAM3::KannalaBrandt8::mvLappingArea’ [-Wreorder]
   95 |         std::vector<int> mvLappingArea;
      |                          ^~~~~~~~~~~~~
/home/beraru/ORB_SLAM3/include/CameraModels/KannalaBrandt8.h:59:9: warning:   when initialized here [-Wreorder]
   59 |         KannalaBrandt8(const std::vector<float> _vParameters, const float _precision) : GeometricCamera(_vParameters),
      |         ^~~~~~~~~~~~~~
/home/beraru/ORB_SLAM3/include/CameraModels/KannalaBrandt8.h: In constructor ‘ORB_SLAM3::KannalaBrandt8::KannalaBrandt8(ORB_SLAM3::KannalaBrandt8*)’:
/home/beraru/ORB_SLAM3/include/CameraModels/KannalaBrandt8.h:105:21: warning: ‘ORB_SLAM3::KannalaBrandt8::precision’ will be initialized after [-Wreorder]
  105 |         const float precision;
      |                     ^~~~~~~~~
/home/beraru/ORB_SLAM3/include/CameraModels/KannalaBrandt8.h:95:26: warning:   ‘std::vector<int> ORB_SLAM3::KannalaBrandt8::mvLappingArea’ [-Wreorder]
   95 |         std::vector<int> mvLappingArea;
      |                          ^~~~~~~~~~~~~
/home/beraru/ORB_SLAM3/include/CameraModels/KannalaBrandt8.h:65:9: warning:   when initialized here [-Wreorder]
   65 |         KannalaBrandt8(KannalaBrandt8* pKannala) : GeometricCamera(pKannala->mvParameters), precision(pKannala->precision), mvLappingArea(2,0) ,tvr(nullptr) {
      |         ^~~~~~~~~~~~~~
In file included from /home/beraru/ORB_SLAM3/include/Atlas.h:27,
                 from /home/beraru/ORB_SLAM3/include/LocalMapping.h:24,
                 from /home/beraru/ORB_SLAM3/include/LoopClosing.h:24,
                 from /home/beraru/ORB_SLAM3/include/Optimizer.h:26,
                 from /home/beraru/ORB_SLAM3/src/Optimizer.cc:20:
/home/beraru/ORB_SLAM3/include/CameraModels/KannalaBrandt8.h: In constructor ‘ORB_SLAM3::KannalaBrandt8::KannalaBrandt8(std::vector<float>)’:
/home/beraru/ORB_SLAM3/include/CameraModels/KannalaBrandt8.h:105:21: warning: ‘ORB_SLAM3::KannalaBrandt8::precision’ will be initialized after [-Wreorder]
  105 |         const float precision;
      |                     ^~~~~~~~~
/home/beraru/ORB_SLAM3/include/CameraModels/KannalaBrandt8.h:95:26: warning:   ‘std::vector<int> ORB_SLAM3::KannalaBrandt8::mvLappingArea’ [-Wreorder]
   95 |         std::vector<int> mvLappingArea;
      |                          ^~~~~~~~~~~~~
In file included from /home/beraru/ORB_SLAM3/include/Atlas.h:27,
                 from /home/beraru/ORB_SLAM3/include/LocalMapping.h:24,
                 from /home/beraru/ORB_SLAM3/include/LoopClosing.h:24,
                 from /home/beraru/ORB_SLAM3/include/Optimizer.h:26,
                 from /home/beraru/ORB_SLAM3/src/Optimizer.cc:20:
/home/beraru/ORB_SLAM3/include/CameraModels/KannalaBrandt8.h:53:9: warning:   when initialized here [-Wreorder]
[... cut ...]
/home/beraru/ORB_SLAM3/src/Optimizer.cc: In static member function ‘static int ORB_SLAM3::Optimizer::PoseInertialOptimizationLastKeyFrame(ORB_SLAM3::Frame*, bool)’:
/home/beraru/ORB_SLAM3/src/Optimizer.cc:7706:10: warning: unused variable ‘bOut’ [-Wunused-variable]
 7706 |     bool bOut = false;
      |          ^~~~
make[2]: *** [CMakeFiles/ORB_SLAM3.dir/build.make:102: CMakeFiles/ORB_SLAM3.dir/src/LoopClosing.cc.o] Error 1
make[2]: *** [CMakeFiles/ORB_SLAM3.dir/build.make:232: CMakeFiles/ORB_SLAM3.dir/src/Optimizer.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:390: CMakeFiles/ORB_SLAM3.dir/all] Error 2
make: *** [Makefile:84: all] Error 2

thuliangjz, 2020-08-10:
@belal-ibrahim it seems you need to have `openssl` installed. Try using `apt-get install libssl-dev`. About another error, it seems stl complains about the allocator defined there, in `include/LoopClosing.h:50`, you can see:

~~~c++
typedef map<KeyFrame*,g2o::Sim3,std::less<KeyFrame*>,
        Eigen::aligned_allocator<std::pair<const KeyFrame*, g2o::Sim3> > > KeyFrameAndPose;
~~~

I wonder if it will works by deleting the `const` keyword there, as your compiler argues. However I am not able to reproduce the error on my laptop, so you may as well post your compiler version, like gcc, g++, as well as that of cmake and (possibly) Eigen library :)

belal-ibrahim, 2020-08-10:
libssl-dev is already the newest version (1.1.1f-1ubuntu2)
gcc version 9.3.0 (Ubuntu 9.3.0-10ubuntu2)
cmake version 3.16.3
Eigen version 3.3.4

belal-ibrahim, 2020-08-10:
onfiguring and building Thirdparty/DBoW2 ...
mkdir: cannot create directory ‘build’: File exists
-- Configuring done
-- Generating done
-- Build files have been written to: /home/beraru/ORB_SLAM3/Thirdparty/DBoW2/build
[100%] Built target DBoW2
Configuring and building Thirdparty/g2o ...
mkdir: cannot create directory ‘build’: File exists
-- BUILD TYPE:Release
-- Compiling on Unix
-- Configuring done
-- Generating done
-- Build files have been written to: /home/beraru/ORB_SLAM3/Thirdparty/g2o/build
[100%] Built target g2o
Uncompress vocabulary ...
Configuring and building ORB_SLAM3 ...
mkdir: cannot create directory ‘build’: File exists
Build type: Release
-- Using flag -std=c++11.
OPENCV VERSION:
3.4.0
-- BUILD TYPE:Release
-- Compiling on Unix
-- Configuring done
-- Generating done
-- Build files have been written to: /home/beraru/ORB_SLAM3/build
[ 37%] Built target g2o
make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/libdc1394.so', needed by '../lib/libORB_SLAM3.so'.  Stop.
make[2]: *** Waiting for unfinished jobs....
[ 38%] Building CXX object CMakeFiles/ORB_SLAM3.dir/src/LoopClosing.cc.o
[ 39%] Building CXX object CMakeFiles/ORB_SLAM3.dir/src/Optimizer.cc.o
/home/beraru/ORB_SLAM3/src/Optimizer.cc:5179:5: warning: "/*" within comment [-Wcomment]
 5179 |     /*std::cout << " Construction time: " << t_const << std::endl;
      |      
In file included from /home/beraru/ORB_SLAM3/include/Atlas.h:27,
                 from /home/beraru/ORB_SLAM3/include/LocalMapping.h:24,
                 from /home/beraru/ORB_SLAM3/include/LoopClosing.h:24,
                 from /home/beraru/ORB_SLAM3/src/LoopClosing.cc:20:
/home/beraru/ORB_SLAM3/include/CameraModels/KannalaBrandt8.h: In constructor ‘ORB_SLAM3::KannalaBrandt8::KannalaBrandt8(std::vector<float>)’:
/home/beraru/ORB_SLAM3/include/CameraModels/KannalaBrandt8.h:105:21: warning: ‘ORB_SLAM3::KannalaBrandt8::precision’ will be initialized after [-Wreorder]
  105 |         const float precision;
      |                     ^~~~~~~~~
/home/beraru/ORB_SLAM3/include/CameraModels/KannalaBrandt8.h:95:26: warning:   ‘std::vector<int> ORB_SLAM3::KannalaBrandt8::mvLappingArea’ [-Wreorder]
   95 |         std::vector<int> mvLappingArea;
      |                          ^~~~~~~~~~~~~
In file included from /home/beraru/ORB_SLAM3/include/Atlas.h:27,
                 from /home/beraru/ORB_SLAM3/include/LocalMapping.h:24,
                 from /home/beraru/ORB_SLAM3/include/LoopClosing.h:24,
                 from /home/beraru/ORB_SLAM3/src/LoopClosing.cc:20:
/home/beraru/ORB_SLAM3/include/CameraModels/KannalaBrandt8.h:53:9: warning:   when initialized here [-Wreorder]
   53 |         KannalaBrandt8(const std::vector<float> _vParameters) : GeometricCamera(_vParameters), precision(1e-6), mvLappingArea(2,0) ,tvr(nullptr) {
      |         ^~~~~~~~~~~~~~
In file included from /home/beraru/ORB_SLAM3/include/Atlas.h:27,
                 from /home/beraru/ORB_SLAM3/include/LocalMapping.h:24,
                 from /home/beraru/ORB_SLAM3/include/LoopClosing.h:24,
                 from /home/beraru/ORB_SLAM3/src/LoopClosing.cc:20:
/home/beraru/ORB_SLAM3/include/CameraModels/KannalaBrandt8.h: In constructor ‘ORB_SLAM3::KannalaBrandt8::KannalaBrandt8(std::vector<float>, float)’:
/home/beraru/ORB_SLAM3/include/CameraModels/KannalaBrandt8.h:105:21: warning: ‘ORB_SLAM3::KannalaBrandt8::precision’ will be initialized after [-Wreorder]
  105 |         const float precision;
      |                     ^~~~~~~~~
/home/beraru/ORB_SLAM3/include/CameraModels/KannalaBrandt8.h:95:26: warning:   ‘std::vector<int> ORB_SLAM3::KannalaBrandt8::mvLappingArea’ [-Wreorder]
   95 |         std::vector<int> mvLappingArea;
      |                          ^~~~~~~~~~~~~
/home/beraru/ORB_SLAM3/include/CameraModels/KannalaBrandt8.h:59:9: warning:   when initialized here [-Wreorder]
   59 |         KannalaBrandt8(const std::vector<float> _vParameters, const float _precision) : GeometricCamera(_vParameters),
      |         ^~~~~~~~~~~~~~
/home/beraru/ORB_SLAM3/include/CameraModels/KannalaBrandt8.h: In constructor ‘ORB_SLAM3::KannalaBrandt8::KannalaBrandt8(ORB_SLAM3::KannalaBrandt8*)’:
/home/beraru/ORB_SLAM3/include/CameraModels/KannalaBrandt8.h:105:21: warning: ‘ORB_SLAM3::KannalaBrandt8::precision’ will be initialized after [-Wreorder]
  105 |         const float precision;
      |                     ^~~~~~~~~
[... 1449 lines cut ...]
      |                                 ^~~~
/home/beraru/ORB_SLAM3/src/Optimizer.cc:6801:31: warning: comparison of integer expressions of different signedness: ‘int’ and ‘std::vector<ORB_SLAM3::MapPoint*>::size_type’ {aka ‘long unsigned int’} [-Wsign-compare]
 6801 |                 for(int i=0; i<vpMonoMPsOpt.size(); ++i)
      |                              ~^~~~~~~~~~~~~~~~~~~~
/home/beraru/ORB_SLAM3/src/Optimizer.cc:6821:32: warning: comparison of integer expressions of different signedness: ‘int’ and ‘std::vector<ORB_SLAM3::MapPoint*>::size_type’ {aka ‘long unsigned int’} [-Wsign-compare]
 6821 |                  for(int i=0; i<vpStereoMPsOpt.size(); ++i)
      |                               ~^~~~~~~~~~~~~~~~~~~~~~
/home/beraru/ORB_SLAM3/src/Optimizer.cc:6841:32: warning: comparison of integer expressions of different signedness: ‘int’ and ‘std::vector<ORB_SLAM3::MapPoint*>::size_type’ {aka ‘long unsigned int’} [-Wsign-compare]
 6841 |                  for(int i=0; i<vpMonoMPsBad.size(); ++i)
      |                               ~^~~~~~~~~~~~~~~~~~~~
[... cut ...]
/home/beraru/ORB_SLAM3/src/Optimizer.cc: In static member function ‘static int ORB_SLAM3::Optimizer::PoseInertialOptimizationLastKeyFrame(ORB_SLAM3::Frame*, bool)’:
/home/beraru/ORB_SLAM3/src/Optimizer.cc:7706:10: warning: unused variable ‘bOut’ [-Wunused-variable]
 7706 |     bool bOut = false;
      |          ^~~~
make[2]: *** [CMakeFiles/ORB_SLAM3.dir/build.make:102: CMakeFiles/ORB_SLAM3.dir/src/LoopClosing.cc.o] Error 1
make[2]: *** [CMakeFiles/ORB_SLAM3.dir/build.make:232: CMakeFiles/ORB_SLAM3.dir/src/Optimizer.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:390: CMakeFiles/ORB_SLAM3.dir/all] Error 2
make: *** [Makefile:84: all] Error 2

jjgr3496, 2020-08-11:
Hi @belal-ibrahim, thank you for posting all the output. However I would ask you next time only post the error messages and not all the warnings (either by modifiying the compiler flags at CMakeLists.txt or by only copying the error lines) to facilitate the readability of the message.

Regarding your problem, it looks like a problem when finding the file openssl/md5.h. You should first install in your system the openssl library (https://cloudwafer.com/blog/installing-openssl-on-ubuntu-16-04-18-04/). If after this, errors keep poping up, please copy us the error messages you are getting!

closed 2020-08-14 by jjgr3496