UZ-SLAMLab/ORB_SLAM3 #14: make failed(undefined reference to symbol '_ZN2cv6String10deallocateEv') , can someone help me to solve this problem?

wycjl, 2020-07-27:
:-1: error: CMakeFiles/stereo euroc.dir/Examples/Stereo/stereo euroc.cc.o: undefined reference to symbol '_ZN2cv6String10deallocateEv'

ericzzj1989, 2020-07-28:
It is the same problem with ros in ORB-SLAM2
Add "-lboost_system" in Cmakelists.txt of ros example, like:

set(LIBS 
${OpenCV_LIBS} 
${EIGEN3_LIBS}
${Pangolin_LIBRARIES}
${PROJECT_SOURCE_DIR}/../../../Thirdparty/DBoW2/lib/libDBoW2.so
${PROJECT_SOURCE_DIR}/../../../Thirdparty/g2o/lib/libg2o.so
${PROJECT_SOURCE_DIR}/../../../lib/libORB_SLAM3.so
-lboost_system
)

DreamWaterFound, 2020-07-29:
Besides "boost" problem, you may have a problem with multiple versions of OpenCV. Pay attention to cmake warnings, check cmake variables and link path after executed `cmake ..`. Custom installed OpenCV, ROS, and Anaconda may lead to this error. Without more information about your OS version, GCC/G++ version, OpenCV version and installed path, etc, I am sorry that I have no more detailed suggestions to help you.

Notice: this error is ENTIRELY about your own dependencies, not the author's problem. Give dislike will not help you solve any problems.

jjgr3496, 2020-08-11:
As @DreamWaterFound, you may have duplicated libraries or multiple versions of OpenCV that maybe are messing with the linking step.Please, check wheter CMake is picking the correct path to OpenCV or if the version installed is above the minimum required. I suggest you to set the CMake verbose flag in order to track better the error.

jjgr3496, 2020-08-14:
This bug was solved in #44.

closed 2020-08-14 by jjgr3496