UZ-SLAMLab/ORB_SLAM3 #15: error in build_ros.sh

chluck2008, 2020-07-27:
In file included from /home/t450/orb_slam3_ws/src/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../include/MapPoint.h:23:0,
                 from /home/t450/orb_slam3_ws/src/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../include/FrameDrawer.h:24,
                 from /home/t450/orb_slam3_ws/src/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../include/Viewer.h:23,
                 from /home/t450/orb_slam3_ws/src/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../include/Tracking.h:27,
                 from /home/t450/orb_slam3_ws/src/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../include/System.h:32,
                 from /home/t450/orb_slam3_ws/src/ORB_SLAM3/Examples/ROS/ORB_SLAM3/src/AR/ViewerAR.h:27,
                 from /home/t450/orb_slam3_ws/src/ORB_SLAM3/Examples/ROS/ORB_SLAM3/src/AR/ViewerAR.cc:19:
/home/t450/orb_slam3_ws/src/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../include/KeyFrame.h:32:29: fatal error: GeometricCamera.h: 没有那个文件或目录
compilation terminated.
CMakeFiles/MonoAR.dir/build.make:200: recipe for target 'CMakeFiles/MonoAR.dir/src/AR/ViewerAR.cc.o' failed
make[2]: *** [CMakeFiles/MonoAR.dir/src/AR/ViewerAR.cc.o] Error 1
make[2]: *** 正在等待未完成的任务....
In file included from /home/t450/orb_slam3_ws/src/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../include/MapPoint.h:23:0,
                 from /home/t450/orb_slam3_ws/src/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../include/FrameDrawer.h:24,
                 from /home/t450/orb_slam3_ws/src/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../include/Viewer.h:23,
                 from /home/t450/orb_slam3_ws/src/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../include/Tracking.h:27,
                 from /home/t450/orb_slam3_ws/src/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../include/System.h:32,
                 from /home/t450/orb_slam3_ws/src/ORB_SLAM3/Examples/ROS/ORB_SLAM3/src/AR/ros_mono_ar.cc:31:
/home/t450/orb_slam3_ws/src/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../include/KeyFrame.h:32:29: fatal error: GeometricCamera.h: 没有那个文件或目录
compilation terminated.
In file included from /home/t450/orb_slam3_ws/src/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../include/MapPoint.h:23:0,
                 from /home/t450/orb_slam3_ws/src/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../include/FrameDrawer.h:24,
                 from /home/t450/orb_slam3_ws/src/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../include/Viewer.h:23,
                 from /home/t450/orb_slam3_ws/src/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../include/Tracking.h:27,
                 from /home/t450/orb_slam3_ws/src/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../include/System.h:32,
                 from /home/t450/orb_slam3_ws/src/ORB_SLAM3/Examples/ROS/ORB_SLAM3/src/ros_mono.cc:30:
/home/t450/orb_slam3_ws/src/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../include/KeyFrame.h:32:29: fatal error: GeometricCamera.h: 没有那个文件或目录
compilation terminated.
CMakeFiles/MonoAR.dir/build.make:119: recipe for target 'CMakeFiles/MonoAR.dir/src/AR/ros_mono_ar.cc.o' failed
make[2]: *** [CMakeFiles/MonoAR.dir/src/AR/ros_mono_ar.cc.o] Error 1
CMakeFiles/Makefile2:820: recipe for target 'CMakeFiles/MonoAR.dir/all' failed
make[1]: *** [CMakeFiles/MonoAR.dir/all] Error 2
make[1]: *** 正在等待未完成的任务....
CMakeFiles/Mono.dir/build.make:119: recipe for target 'CMakeFiles/Mono.dir/src/ros_mono.cc.o' failed
make[2]: *** [CMakeFiles/Mono.dir/src/ros_mono.cc.o] Error 1
CMakeFiles/Makefile2:718: recipe for target 'CMakeFiles/Mono.dir/all' failed
make[1]: *** [CMakeFiles/Mono.dir/all] Error 2
In file included from /home/t450/orb_slam3_ws/src/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../include/MapPoint.h:23:0,
                 from /home/t450/orb_slam3_ws/src/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../include/FrameDrawer.h:24,
                 from /home/t450/orb_slam3_ws/src/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../include/Viewer.h:23,
                 from /home/t450/orb_slam3_ws/src/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../include/Tracking.h:27,
                 from /home/t450/orb_slam3_ws/src/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../include/System.h:32,
                 from /home/t450/orb_slam3_ws/src/ORB_SLAM3/Examples/ROS/ORB_SLAM3/src/ros_stereo.cc:33:
/home/t450/orb_slam3_ws/src/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../include/KeyFrame.h:32:29: fatal error: GeometricCamera.h: 没有那个文件或目录
compilation terminated.
CMakeFiles/Stereo.dir/build.make:119: recipe for target 'CMakeFiles/Stereo.dir/src/ros_stereo.cc.o' failed
make[2]: *** [CMakeFiles/Stereo.dir/src/ros_stereo.cc.o] Error 1
CMakeFiles/Makefile2:104: recipe for target 'CMakeFiles/Stereo.dir/all' failed
make[1]: *** [CMakeFiles/Stereo.dir/all] Error 2
In file included from /home/t450/orb_slam3_ws/src/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../include/MapPoint.h:23:0,
                 from /home/t450/orb_slam3_ws/src/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../include/FrameDrawer.h:24,
                 from /home/t450/orb_slam3_ws/src/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../include/Viewer.h:23,
                 from /home/t450/orb_slam3_ws/src/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../include/Tracking.h:27,
                 from /home/t450/orb_slam3_ws/src/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../include/System.h:32,
                 from /home/t450/orb_slam3_ws/src/ORB_SLAM3/Examples/ROS/ORB_SLAM3/src/ros_rgbd.cc:33:
/home/t450/orb_slam3_ws/src/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../include/KeyFrame.h:32:29: fatal error: GeometricCamera.h: 没有那个文件或目录
compilation terminated.
CMakeFiles/RGBD.dir/build.make:119: recipe for target 'CMakeFiles/RGBD.dir/src/ros_rgbd.cc.o' failed
make[2]: *** [CMakeFiles/RGBD.dir/src/ros_rgbd.cc.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/RGBD.dir/all' failed
make[1]: *** [CMakeFiles/RGBD.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

ericzzj1989, 2020-07-28:
You can replace "ORB_SLAM2" by "ORB_SLAM3".

xiefei2929, 2020-07-28:
The author said this is a Beta version and has not come to rewrite the ROS interface, nor provide the visual inertial ROS interface, I added two visual-inertial ROS interfaces Mono_inertial and Stereo_inertial at https://github.com/xiefei2929/ORB_SLAM3-ROS-Interface
you can have a try.

chluck2008, 2020-07-29:
@xiefei2929   @ericzzj1989 thanks

closed 2020-08-10 by richard-elvira