UZ-SLAMLab/ORB_SLAM3 #16: ORB-SLAM3 crash when run with kitti datasets

michaelnguyen11, 2020-07-27:
Hi authors,

Thank you for excellent work and public your code.

I'm a big fan of ORB-SLAM2 and studying ORB-SLAM3. The results from stereo-inertial SLAM + TUM datasets are awesome, obviously better than filter-based approach MSCKF-VIO, which I'm using. But I'm facing with this issues :

When run ORB-SLAM3 with Kitti datasets, the program crashed after new map created.

Backtrace :
`
#0  0x00007f679f40cfe4 in ORB_SLAM3::EdgeSE3ProjectXYZOnlyPose::computeError() ()
    at /ORB_SLAM3/lib/libORB_SLAM3.so
#1  0x00007f679d2eed61 in g2o::SparseOptimizer::computeActiveErrors() ()
    at /ORB_SLAM3/Thirdparty/g2o/lib/libg2o.so
#2  0x00007f679d2f831e in g2o::OptimizationAlgorithmLevenberg::solve(int, bool) ()
    at /ORB_SLAM3/Thirdparty/g2o/lib/libg2o.so
#3  0x00007f679d2f02ad in g2o::SparseOptimizer::optimize(int, bool) ()
    at /ORB_SLAM3/Thirdparty/g2o/lib/libg2o.so
#4  0x00007f679f3d2d29 in ORB_SLAM3::Optimizer::PoseOptimization(ORB_SLAM3::Frame*) ()
    at /ORB_SLAM3/lib/libORB_SLAM3.so
#5  0x00007f679f322a2e in ORB_SLAM3::Tracking::TrackReferenceKeyFrame() () at /ORB_SLAM3/lib/libORB_SLAM3.so
`

I would be nice if you guys or someone help me to overcome this issues. I also commented this issue in https://github.com/UZ-SLAMLab/ORB_SLAM3/issues/3

Many thanks in advance.

Regards,
Michael

michaelnguyen11, 2020-07-27:
Attaching full backtrace file.
[orb_slam3_stereo_kitti.txt](https://github.com/UZ-SLAMLab/ORB_SLAM3/files/4981573/orb_slam3_stereo_kitti.txt)

How to reproduce:
- Run this command : ./stereo_kitti ../../Vocabulary/ORBvoc.txt KITTI00-02.yaml /path/to/kitti_dataset/0/00

yan99033, 2020-07-27:
Hi, check the solution that I posted in #3 and see if it helps.

michaelnguyen11, 2020-07-28:
Thank you very much @yan99033 ,

It work like a charm, cheers.

closed 2020-07-28 by michaelnguyen11