UZ-SLAMLab/ORB_SLAM3 #17: in ros save map?

chluck2008, 2020-07-28:
What can I do to save and load maps under ROS, and realize odometer under ROS

w5543081, 2020-07-29:
same question，How can ORB-SLAM3  save maps ?  the interface of ORB-SLAM2  which is for saving map  is abandoned.

chluck2008, 2020-07-29:
[image]

varunns, 2020-09-02:
There is a piece of commented code, 2 methods : SaveAtlas and LoadAtlas in System.cc, looks like it is the relevant code, I am trying to find out the right place to make the calls to save the map

richard-elvira, 2020-09-02:
The feature to save maps and load in a new session is currently in developing, We are going to publish it in the 0.4 version before of the end of the year.

kwads123, 2020-12-02:
> The feature to save maps and load in a new session is currently in developing, We are going to publish it in the 0.4 version before of the end of the year.

@richard-elvira Do you have any update on when the save/load feature will be added?

rebellojason7, 2020-12-18:
@richard-elvira is there any update with regard to this feature ? Thanks

pavelsamoylenko, 2021-01-21:
@richard-elvira is there any update regard to map saving?

cntzj, 2021-03-24:
@richard-elvira is there any update with regard to this feature ? Thanks

richard-elvira, 2021-03-24:
Hi, sorry for the delay, we had been busy lately.

We are close to submitting a new version of the code with this functionality and a more optimized code to run faster.

neo-knight-td, 2021-04-20:
When would you be able to provide this code ? @richard-elvira

neo-knight-td, 2021-05-04:
Hi all,

Just made the Save/Load Atlas Functionality here https://github.com/UZ-SLAMLab/ORB_SLAM3/pull/310

Hope it helps !

phdsky, 2021-07-29:
Any official update? Looking forward to v4.0...

jdtardos, 2021-12-23:
Version 1.0 now includes map load and saving operations

closed 2021-12-23 by jdtardos