UZ-SLAMLab/ORB_SLAM3 #19: ¿Como puedo ejecutar ORB_SLAM3 en una camara tiempo real sin utilizar dataset?

RafaelAgular, 2020-07-28:


jjgr3496, 2020-08-11:
Hola @RafaelAgular.

Los ejemplos que damos junto con la libreria muestran el funcionamiento de las clases y metodos de la misma y de como se usan. Si quieres usar ORB-SLAM3 con tu propia camara en tiempo real, deberias crear tu propio programa el cual, en primer lugar crea todas las clases necesarias (como por ejemplo, el system de ORB-SLAM3) y posteriormente, un bucle en el que coges la imagen de la camara (el codigo de esta parte depende integramente de la camara que utilices y de como lees sus imagenes) y se la pasas al ORB-SLAM3 para que la procese (de manera identica a como se hace en los ejemplos con datasets).

Un saludo.

varunns, 2020-08-25:
Hi I tried to add the code for reading images from a live cam stream , but the system get stuck at initialiazation. Following is the code:
```
  // Create SLAM system. It initializes all system threads and gets ready to process frames.
  ORB_SLAM3::System SLAM(argv[1],argv[2],ORB_SLAM3::System::MONOCULAR,true);

  cv::VideoCapture cap(id); //id if 4 for my laptop
  if (!cap.isOpened()) 
  {
    std::cerr << "ERROR: Could not open camera" << std::endl;
    return 1;
  }

  int count = 0;
  for(;;)
  {
  	Mat im;
  	cap >> im;

  	//return if frame is not valid
  	if(im.empty())
    {
        cerr << endl << "Failed to load image at: "
             << count << endl;
        return 1;
    }

//    imshow("im", im);
//  	cv::waitKey(20);

    std::chrono::steady_clock::time_point t1 = std::chrono::steady_clock::now();
    double tframe = t1.time_since_epoch().count();

    // Pass the image to the SLAM system
    SLAM.TrackMonocular(im,tframe);

    std::chrono::steady_clock::time_point t2 = std::chrono::steady_clock::now();

    double ttrack= std::chrono::duration_cast<std::chrono::duration<double> >(t2 - t1).count();

    vTimesTrack.push_back(ttrack);

     // Wait to load the next frame
    double T=0;
    if(count>0)
            T = tframe-vTimesTrack[count];

		if(ttrack<T)
            usleep((T-ttrack)*1e6);

    count++;
  	
  }
```

jjgr3496, 2020-09-01:
@varunns  Your code looks fine, however without seeing your images I can not tell whether is problem of ORB-SLAM3 or an issue with your images. Please, note that we are aware of some issues related with monocular initialization and we are currently working on it!

jdtardos, 2021-12-23:
Version 1.0 now includes examples to run ORB-SLAM3 with Intel Realsense cameras

closed 2021-12-23 by jdtardos