UZ-SLAMLab/ORB_SLAM3 #20: Running with my own sequence.

mirellameelo, 2020-07-28:
Hi.

I'm using a stereo camera. I based myself on the EuRoC dataset to form my own sequences. I saved frames, timestamps and I also have the camera calibration parameters. 
Here is the organization of my folders:

```hl
MH01
   |---- mav0
          |---- cam0
                |---- data
                       |---- photos of the left side (timestamp.png)
          |---- cam1
                |---- data
                       |---- photos of the right side (timestamp.png)
   |---- timestamps.txt (timestamp timestamp.png )
   |---- config.yalm
````

Running ./euroc_examples.sh, I have the error:

```hl
.
.
.
Failed to load image at: /path/MH01/mav0/cam0/data/1403636579763555584.png
```

It is looking for the original dataset, so I assumed this is not the appropriate way, right? Can anyone help me with the right command to run ORBSLAM3 with my own sequence? 

Tks in advance.

Halcao, 2020-07-29:
The timestamp file in `./euroc_examples.sh` is located in `./Examples/Monocular/EuRoC_TimeStamps/MH01.txt`. So you need either replace the stamp file argument in `./euroc_examples.sh` with your own file path or place your timestamp file in the path above.

mirellameelo, 2020-07-29:
@Halcao sorry, my mistake, I have changed the ./euroc_examples.sh as well. I just edited my prior comment. :)

Halcao, 2020-07-29:
OK. Did you modify the timestamp file? Could you please post the first 10 lines of the file?

mirellameelo, 2020-07-29:
Sure! 

```hl
1595976430480 1595976430480.png
1595976430834 1595976430834.png
1595976431288 1595976431288.png
1595976431620 1595976431620.png
1595976431952 1595976431952.png
1595976432438 1595976432438.png
1595976432780 1595976432780.png
1595976433071 1595976433071.png
1595976433530 1595976433530.png
1595976433912 1595976433912.png
```

Halcao, 2020-07-29:
@mirellameelo This file is supposed to contain only the first column which is the file name (without extension `.png`) of your image. And the timestamp will be divided by 1e9, so you'd better either append nine 0s to the filename like 1595976430480**000000000**.png or modify the following code in mono_euroc.cc.
```c++
    while(!fTimes.eof())
    {
        string s;
        getline(fTimes,s);
        if(!s.empty())
        {
            stringstream ss;
            ss << s;
            vstrImages.push_back(strImagePath + "/" + ss.str() + ".png");
            double t;
            ss >> t;
            // vTimeStamps.push_back(t/1e9);
           vTimeStamps.push_back(t);
        }
    }
```
If you choose to modify the code, the timestamp in the trajectory file will still be multiplied by 10e9, and you should also modify it.(System.cc:752)
```c++
//             f << setprecision(6) << 1e9*pKF->mTimeStamp  << " " <<  setprecision(9) << twb.at<float>(0) << " " << twb.at<float>(1) << " " << twb.at<float>(2) << " " << q[0] << " " << q[1] << " " << q[2] << " " << q[3] << endl;

            f << setprecision(6) << pKF->mTimeStamp  << " " <<  setprecision(9) << twb.at<float>(0) << " " << twb.at<float>(1) << " " << twb.at<float>(2) << " " << q[0] << " " << q[1] << " " << q[2] << " " << q[3] << endl;
```

mirellameelo, 2020-07-29:
oh no... I just notice I never changed the timestamp path!! But you definitely predicted what would be the next mistake. rsrsrs. Now the SLAM system starts but it only loads the first image... I'm trying to figure it out. Tks for your patience! :)

closed 2020-07-29 by mirellameelo