UZ-SLAMLab/ORB_SLAM3 #21: Error when running with mono dataset

gxytcrc, 2020-07-29:
Hey
I am running with my own monocular datasets (TUM form). The datasets can be run successfully in ORB-SLAM2. But in ORB-SLAM3 it will meet trouble:
```
./Examples/Monocular/mono_tum Vocabulary/ORBvoc.txt Examples/Monocular/TUM3.yaml ceiling_1_enhance 

ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Monocular

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name: 

Camera Parameters: 
- fx: 535.4
- fy: 539.2
- cx: 320.1
- cy: 247.6
- bf: 0
- k1: 0
- k2: 0
- p1: 0
- p2: 0
- fps: 30
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 1000
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

-------
Start processing sequence ...
Images in the sequence: 1592
Segmentation fault (core dumped)
```
Is any one knows how to fix this problem? Thank you

anakin0620, 2020-07-29:
#3 Has the solution.

gxytcrc, 2020-07-29:
> #3 Has the solution.

The problem solved! Thank you !

closed 2020-07-29 by gxytcrc