UZ-SLAMLab/ORB_SLAM3 #23: Issue when the loop closing is processed

gxytcrc, 2020-07-29:
Hey!
I am running with my own monocular datasets (TUM form). The first issue I have reported in #21 is solved, Thank you! But I have met a new problem, when the system detect the loop, I meet "Segmentation fault (core dump)" again, the issue is:

```
./Examples/Monocular/mono_tum Vocabulary/ORBvoc.txt Examples/Monocular/TUM4.yaml ceiling_1_enhance 

ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Monocular

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name: 

Camera Parameters: 
- fx: 726.287
- fy: 726.287
- cx: 354.65
- cy: 186.466
- bf: 0
- k1: 0
- k2: 0
- p1: 0
- p2: 0
- fps: 10
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 1000
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

-------
Start processing sequence ...
Images in the sequence: 1592

First KF:0; Map init KF:0
New Map created with 243 points
Point distribution in KeyFrame: left-> 243 --- right-> 0
BoW: 165 matches between 1189 points with coarse Sim3
terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(3.4.10) /home/gxytcrc/opencv/modules/imgproc/src/color.cpp:182: error: (-215:Assertion failed) !_src.empty() in function 'cvtColor'

Segmentation fault (core dumped)
```

I find the issue is in line 4001 of Optimizer.cc
```
    cv::Mat img1 = cv::imread(pKF1->mNameFile, CV_LOAD_IMAGE_UNCHANGED);
    cv::cvtColor(img1, img1, CV_GRAY2BGR);

```
The img1 returns empty value, and this cause the system crash. 

is anyone met the same issue? or do anyone has the solution?

liangsongLee, 2020-08-05:
I have the same problem, have you solved this?

shubhendu1907, 2020-08-08:
I have encountered the same problem too. I think that CV_LOAD_IMAGE_UNCHANGED is outdated in some Opencv 3.x.x versions. I set it to -1 and tried again but still got the same error. Did you find any solution?

closed 2020-08-10 by richard-elvira