UZ-SLAMLab/ORB_SLAM3 #27: Low numbers of matched features when running in IMU_STEREO mode with KITTI raw data

biendltb, 2020-07-30:
Hi,

Thanks for this amazing work.

I've tried to config and run the code with KITTI raw data (unsynced & unrectified data) in which the IMU data is available (it is not available in KITTI evaluation dataset). However, I faced the issue of low numbers of matched features when running the code. Please refer to the log below:
```
...
mnFirstFrameId = 101
mnInitialFrameId = 99
72 Frames had been set to lost
last KF is empty!
not IMU meas
last KF is empty!
First KF:31; Map init KF:30
New Map created with 128 points
Matches Inliners: 48 | Matching ratio: 48/1010
Fail to track local map!
IMU is not or recently initialized. Reseting active map...
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 105
mnInitialFrameId = 103
74 Frames had been set to lost
last KF is empty!
not IMU meas
last KF is empty!
First KF:32; Map init KF:31
New Map created with 133 points
Matches Inliners: 43 | Matching ratio: 43/1005
Fail to track local map!
IMU is not or recently initialized. Reseting active map...
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 108
mnInitialFrameId = 106
76 Frames had been set to lost
last KF is empty!
not IMU meas
last KF is empty!
First KF:33; Map init KF:32
New Map created with 121 points
Matches Inliners: 43 | Matching ratio: 43/1006
Fail to track local map!
IMU is not or recently initialized. Reseting active map...
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 111
mnInitialFrameId = 109
78 Frames had been set to lost
last KF is empty!
not IMU meas

Saving trajectory to f_kitti_0001.txt ...
```

The number of features is lower than 50 (which is set in the code as the threshold for a successfully tracking). This made the program repeatedly failed in the tracking stage and the tracking map kept resetting all the time.

One of the sources for the issue that I could think of is the wrong parameters in my configuration file. The camera and IMU calibration were carefully calculated.

Please let me know if you spot any issue in the code.

The code could be found at https://github.com/biendltb/ORB_SLAM3/tree/FEAT/kitti_raw_imu_stereo

Thank you and best wishes,

Bien

ccamposm, 2020-08-14:
Hi @biendltb.
You should add stereo rectification to your stereo_inertial_kitti.cc file. Check provided stereo_inertial_euroc.cc for a guideline.

biendltb, 2020-08-15:
Hi @ccamposm,

Thank you very much for your time to review my code. That's a silly mistake since I cloned and modified it from the `stereo_inertial_tum_vi.cc` without realising that their distortion model differs from KITTI.
I updated the file by your suggestion and lower the FAST threshold to get more feature detected but the number of matches is still pretty low and leads to the fail of tracking.

You could see my code in the same repo.

I truly appreciate your time and effort.

Best wishes,

Bien

ericzzj1989, 2020-09-09:
Hi @biendltb,

How do you set the IMU noise parameters in your repo https://github.com/biendltb/ORB_SLAM3/tree/FEAT/kitti_raw_imu_stereo ?
I truly appreciate your time and effort.

Best,
Zhenjun

biendltb, 2020-09-09:
Hi @ericzzj1989,

You could find the section of setting the IMU parameters in the YAML configure file. For example, I use KITTI in my case, the config is at https://github.com/biendltb/ORB_SLAM3/blob/13822ef9227108adc4f019c70722cfc74ee61012/Examples/Stereo-Inertial/KITTI_26_09.yaml#L46.

Hope it help.

Bien

ericzzj1989, 2020-09-10:
Hi @biendltb,

Thank you for your prompt reply. I'd like to ask what you refer to when setting these IMU noise parameters.
Thank you for your confirmation and looking forward to your reply. 

Best,
Zhenjun

closed 2020-12-18 by jdtardos