UZ-SLAMLab/ORB_SLAM3 #28: IMU initalize is too late.

HwangDaeHyun, 2020-07-30:
We are testing with Intel's T-265 sensor and VI sensor.

Overall these sensors seem to work fine with ORB SLAM3.

The screenshot below is the result of the sensor experiment.

[image]

[image]

**By the way, when we tested with the T-265 sensor, the initialization is always late with the message'IMU is not initialized. Reset Active Map' (below photo)**

[image]

We have experimented with both the stereo IMU and mono IMU versions of your algorithm. And the results of both are the same. (late initialize with T-265)

Why is initialization so late? How to quickly complete initialization?

biendltb, 2020-07-30:
I think that the issue that I posted in #27 is the same as yours. It is not about the process of IMU initialization but there is probably something wrong with the calib parameters or by other reasons that I haven't figured out yet. This makes the number of matches inliers lower than 50 (refer to this [line](https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/master/src/Tracking.cc#L2051)).

I could see the number of points for initializing the map in your case is relatively low. Testing on TUM VI shows me that a new map created in the first frame has 201 points. Since ORB features are matched by using the transformation obtained from IMU, wrong camera calibration matrixes or IMU-camera transformation matrix could lead to the low number of tracked points.

You should re-check your calib parameters and please let me know if you found something else.

zhoumingchaoslam, 2020-07-31:
Hi , biendltb.
I have the same problem as HwangDaeHyun 。
I use mono IMU interface as follow

`using namespace std;

class ImageGrabber
{
public:
    ImageGrabber(ORB_SLAM3::System* pSLAM):mpSLAM(pSLAM){}

    void GrabMonoImu(const sensor_msgs::ImageConstPtr& msg , const sensor_msgs::ImuConstPtr& imu_msg);

    vector<ORB_SLAM3::IMU::Point> vImuMeas;

    ORB_SLAM3::System* mpSLAM;
};

int main(int argc, char **argv)
{
    ros::init(argc, argv, "Mono_inertial");
    ros::start();
    if(argc != 3)
    {
        cerr << endl << "Usage: rosrun ORB_SLAM2 Mono_inertial path_to_vocabulary path_to_settings" << endl;        
        ros::shutdown();
        return 1;
    }    

    // Create SLAM system. It initializes all system threads and gets ready to process frames.
    ORB_SLAM3::System SLAM(argv[1],argv[2],ORB_SLAM3::System::IMU_MONOCULAR,true);
    ImageGrabber igb(&SLAM);

    ros::NodeHandle nodeHandler;

    // message_filters::Subscriber<sensor_msgs::Image> img_sub(nodeHandler, "/mynteye/right/image_raw", 1);
    // message_filters::Subscriber<sensor_msgs::Imu> imu_sub(nodeHandler, "/mynteye/imu/data_raw", 1);

    // message_filters::Subscriber<sensor_msgs::Image> img_sub(nodeHandler, "/camera/image_right", 1);
    // message_filters::Subscriber<sensor_msgs::Imu> imu_sub(nodeHandler, "/imu/data", 1);

    message_filters::Subscriber<sensor_msgs::Image> img_sub(nodeHandler, "/cam0/image_raw", 1);
    message_filters::Subscriber<sensor_msgs::Imu> imu_sub(nodeHandler, "/imu0", 1);
    
    typedef message_filters::sync_policies::ApproximateTime<sensor_msgs::Image, sensor_msgs::Imu> sync_pol;
    message_filters::Synchronizer<sync_pol> sync(sync_pol(5), img_sub, imu_sub);
    // sync.registerCallback(boost::bind(&ImageGrabber::GrabMonoImu,&igb,_1,_2));
    sync.registerCallback(boost::bind(&ImageGrabber::GrabMonoImu,&igb,_1,_2));

    ros::spin();

    // Stop all threads
    SLAM.Shutdown();

    // Save camera trajectory
    SLAM.SaveKeyFrameTrajectoryTUM("KeyFrameTrajectory.txt");

    ros::shutdown();

    return 0;
}

void ImageGrabber::GrabMonoImu(const sensor_msgs::ImageConstPtr& msg , const sensor_msgs::ImuConstPtr& imu_msg)
{   
    // Copy the ros image message to cv::Mat.
    cv_bridge::CvImageConstPtr cv_ptr;
    try
    {
        cv_ptr = cv_bridge::toCvShare(msg);
    }
    catch (cv_bridge::Exception& e)
    {
        ROS_ERROR("cv_bridge exception: %s", e.what());
        return;
    }
        float t = imu_msg->header.stamp.toSec();
        float dx = imu_msg->linear_acceleration.x;
        float dy = imu_msg->linear_acceleration.y;
        float dz = imu_msg->linear_acceleration.z;
        float rx = imu_msg->angular_velocity.x;
        float ry = imu_msg->angular_velocity.y;
        float rz = imu_msg->angular_velocity.z;
        vImuMeas.push_back(ORB_SLAM3::IMU::Point(dx, dy, dz, rx, ry, rz, t));

    mpSLAM->TrackMonocular(cv_ptr->image,cv_ptr->header.stamp.toSec(),vImuMeas);
    vImuMeas.clear();
}`

Hope you can help，thanks。

biendltb, 2020-08-05:
@zhoumingchaoslam Hi, I'm not sure how it related but I guess the issue could be just lie in the model we use in the configuration file since it processes differently for each type of the camera model. For examples, this issue occurs if I use the `PinHole` model for the TUM dataset with the same config (the author uses `KannalaBrandt8` originally and it works fine).

@HwangDaeHyun @zhoumingchaoslam could use please share your config file (`.yaml`) here?

ccamposm, 2020-08-10:
Hi @HwangDaeHyun.
From your terminal screenshot, it seems that map is being initialized with a low number of 3D points (28 and 38 points) and it is immediately lost. This might be caused by a bad calibration. Are intrinsic camera parameters accurately calibrated?

HwangDaeHyun, 2020-08-11:
Hi @ccamposm  , @biendltb 

Thank you for your reply.

First , We use './your stereo_inertial_tum_vi' , and the below script is a our parameter setting result.

`%YAML:1.0

#--------------------------------------------------------------------------------------------
# Camera Parameters. Adjust them!
#--------------------------------------------------------------------------------------------
Camera.type: "KannalaBrandt8"

# Left Camera calibration and distortion parameters (OpenCV)
Camera.fx: 286.323211669922
Camera.fy: 286.406188964844
Camera.cx: 419.292694091797
Camera.cy: 398.536987304688

# Equidistant distortion 0.0034823894022493434, 0.0007150348452162257, -0.0020532361418706202, 0.00020293673591811182
Camera.k1: -0.00618791813030839
Camera.k2: 0.0397853218019009
Camera.k3: -0.03747383877635
Camera.k4: 0.00621245102956891

# Right Camera calibration and distortion parameters (OpenCV)
Camera2.fx: 285.523101806641
Camera2.fy: 285.560607910156
Camera2.cx: 422.4619140625
Camera2.cy: 401.262603759766

# Kannala-Brandt distortion parameters
Camera2.k1: -0.00953370705246925
Camera2.k2: 0.0469451397657394
Camera2.k3: -0.0442381091415882
Camera2.k4: 0.00841759797185659
    
# Transformation matrix from right camera to left camera
Tlr: !!opencv-matrix
  rows: 3
  cols: 4
  dt: f
  data: [0.999969,-0.000418366,0.00782156,0.0643667802214622,0.000462181,0.999984,-0.00560085,0.000242494963458739,-0.00781909,0.00560429,0.999954,-0.000372416659956798]

# Lapping area between images
# Lapping.left: 0
# Lapping.right: 511

# Camera resolution
Camera.width: 848
Camera.height: 800

# Camera frames per second 
Camera.fps: 30.0

# Color order of the images (0: BGR, 1: RGB. It is ignored if images are grayscale)
Camera.RGB: 1

# Close/Far threshold. Baseline times.
ThDepth: 40
Camera.bf: 19.3079

# Transformation from body-frame (imu) to left camera
Tbc: !!opencv-matrix
   rows: 4
   cols: 4
   dt: f
   data: [-0.999918,-0.00828281,0.00981762,0.0106991194188595,0.00826257,-0.999964,-0.00210108,-8.84094624780118e-05,0.00983466,-0.00201979,0.99995,-0.00010523090895731,0.0, 0.0, 0.0, 1.0]

# IMU noise (Use those from VINS-mono)
IMU.NoiseGyro: 0.02268927090279
IMU.NoiseAcc: 0.08182447828798
IMU.GyroWalk: 0.007071067804794407
IMU.AccWalk: 0.09999999873700
IMU.Frequency: 200

# thFarPoints: 20.0

#--------------------------------------------------------------------------------------------
# ORB Parameters
#--------------------------------------------------------------------------------------------

# ORB Extractor: Number of features per image
ORBextractor.nFeatures: 1000 # Tested with 1250

# ORB Extractor: Scale factor between levels in the scale pyramid 	
ORBextractor.scaleFactor: 1.2

# ORB Extractor: Number of levels in the scale pyramid	
ORBextractor.nLevels: 8

# ORB Extractor: Fast threshold
# Image is divided in a grid. At each cell FAST are extracted imposing a minimum response.
# Firstly we impose iniThFAST. If no corners are detected we impose a lower value minThFAST
# You can lower these values if your images have low contrast			
ORBextractor.iniThFAST: 20 # 20
ORBextractor.minThFAST: 7 # 7

#--------------------------------------------------------------------------------------------
# Viewer Parameters
#--------------------------------------------------------------------------------------------
Viewer.KeyFrameSize: 0.05
Viewer.KeyFrameLineWidth: 1
Viewer.GraphLineWidth: 0.9
Viewer.PointSize: 2
Viewer.CameraSize: 0.08
Viewer.CameraLineWidth: 3
Viewer.ViewpointX: 0
Viewer.ViewpointY: -0.7
Viewer.ViewpointZ: -3.5
Viewer.ViewpointF: 500
`
and we found all calibration parameter from t265 interface.(the command 'rs-enumrate-device -c' )

the result is below code.

HwangDaeHyun, 2020-08-13:
Hello @ccamposm
Thanks for your answer.

I fixed this problem.

In my case, IMU parameter was wrong.

I ran ORB-SLAM3 using the T-265's default built-in, external parameters. You can extract these parameters using the command line'rs-enumerate-devices -c'.

And using the motion capture system and the rpg_trajectory_evaluation (https://github.com/uzh-rpg/rpg_trajectory_evaluation) tool, we evaluated the VIO algorithm as shown below.

! [down_trajectory_side] (https://user-images.githubusercontent.com/16742591/90117450-e5958800-dd91-11ea-83e4-5caad0c6ba37.png)

ORB-SLAM3 showed the best performance in RMSE. 

thank you.

closed 2020-08-13 by HwangDaeHyun

ccamposm, 2020-08-13:
Hi @HwangDaeHyun, We are happy to hear that!
Thanks for sharing your results.

PatZhuang, 2020-08-26:
> Hello @ccamposm
> Thanks for your answer.
>
> I fixed this problem.
>
> In my case, IMU parameter was wrong.
>
> I ran ORB-SLAM3 using the T-265's default built-in, external parameters. You can extract these [...]

I came to exact same issue, could you please explain what exactly is "IMU parameters"?
Are you referring to the IMU settings in yaml file that ORB_SLAM3 uses for "PATH_TO_SETTINGS_FILE" or the calibration settings of acceleration and gyro of the imu itself? 
And how could I resolve this issue?

zdegrh, 2020-09-04:
> Hello @ccamposm
> Thanks for your answer.
>
> I fixed this problem.
>
> In my case, IMU parameter was wrong.
>
> I ran ORB-SLAM3 using the T-265's default built-in, external parameters. You can extract these [...]

I used all the IMU peremeter from "rs-enumerate-devices -c", but it gets wrong track result.

it is my transformation
Transformation from body-frame (imu) to camera
Tbc: !!opencv-matrix
   rows: 4
   cols: 4
   dt: f
   data: [ -0.999972,         0.00193326,       0.00728582 ,  0.0107000041753054,
           -0.00192006,      -0.999996,         0.001819, 1.2732925824821e-11,
           0.00728931,       0.00180496,       0.999972, -2.91038304567337e-11,
          0.0, 0.0, 0.0, 1.0]
# IMU noise (Use those from VINS-mono)
IMU.NoiseGyro: 0.000005148030141 # rad/s^0.5 
IMU.NoiseAcc: 0.000066952452471 # m/s^1.5
IMU.GyroWalk: 0.000000499999999 # rad/s^1.5
IMU.AccWalk:  0.000099999997474 # m/s^2.5
IMU.Frequency: 200

is the peremeter correct?
how is your peremeter?
thanks!

HwangDaeHyun, 2020-09-04:
I think you should take the square root of the IMU data you used.

The data you used are variance, so the values ​​look very small.

The Realsense command outputs Varaince. when you running  'rs-enumerate-devices -c".

zdegrh, 2020-09-04:
> I think you should take the square root of the IMU data you used.
> 
> The data you used are variance, so the values ​​look very small.
> 
> When you run the 'rs-enumerate-devices -c' , outputs are varaince of the noise and bias.

thanks,
I have tried just now, the result is still wrong,
could you please paste the yaml file?

HwangDaeHyun, 2020-09-04:
`%YAML:1.0

#--------------------------------------------------------------------------------------------
# Camera Parameters. Adjust them!
#--------------------------------------------------------------------------------------------
Camera.type: "KannalaBrandt8"

# Left Camera calibration and distortion parameters (OpenCV)
Camera.fx: 284.024810791016
Camera.fy: 283.993194580078
Camera.cx: 420.571411132812
Camera.cy: 392.527893066406

# Equidistant distortion 0.0034823894022493434, 0.0007150348452162257, -0.0020532361418706202, 0.00020293673591811182
Camera.k1: -0.00133633601944894
Camera.k2: 0.033676739782095
Camera.k3: -0.0316411294043064
Camera.k4: 0.00447790184989572

# Right Camera calibration and distortion parameters (OpenCV)
Camera2.fx: 284.504608154297
Camera2.fy: 284.571014404297
Camera2.cx: 422.380615234375
Camera2.cy: 400.060607910156

# Kannala-Brandt distortion parameters
Camera2.k1: -0.00288506899960339
Camera2.k2: 0.0386773310601711
Camera2.k3: -0.0366080291569233
Camera2.k4: 0.00597896287217736
    
# Transformation matrix from right camera to left camera
Tlr: !!opencv-matrix
  rows: 3
  cols: 4
  dt: f
  data: [0.999962,-0.00160437,0.00852221,0.0641284808516502,0.00164596,0.999987,-0.0048758,-5.3733241657028e-05,-0.00851427,0.00488964,0.999952,-0.000366911059245467 ]

# Lapping area between images
# Lapping.left: 0
# Lapping.right: 511

# Camera resolution
Camera.width: 848
Camera.height: 800

# Camera frames per second 
Camera.fps: 30.0

# Color order of the images (0: BGR, 1: RGB. It is ignored if images are grayscale)
Camera.RGB: 1

# Close/Far threshold. Baseline times.
ThDepth: 40
Camera.bf: 19.3079

# Transformation from body-frame (imu) to left camera
Tbc: !!opencv-matrix
   rows: 4
   cols: 4
   dt: f
   data: [-0.999957,0.00290471,0.00883124,0.0107000023126602,-0.00294505,-0.999985,-0.00455868,-1.09139364212751e-11,0.00881787,-0.00458449,0.999951,-2.91038304567337e-11,0,0,0,1]

# IMU noise (Use those from VINS-mono)
IMU.NoiseGyro: 0.0023
IMU.NoiseAcc: 0.0082
IMU.GyroWalk: 0.00071
IMU.AccWalk: 0.01
IMU.Frequency: 200

# thFarPoints: 20.0

#--------------------------------------------------------------------------------------------
# ORB Parameters
#--------------------------------------------------------------------------------------------

# ORB Extractor: Number of features per image
ORBextractor.nFeatures: 1000 # Tested with 1250

# ORB Extractor: Scale factor between levels in the scale pyramid 	
ORBextractor.scaleFactor: 1.2

# ORB Extractor: Number of levels in the scale pyramid	
ORBextractor.nLevels: 8

# ORB Extractor: Fast threshold
# Image is divided in a grid. At each cell FAST are extracted imposing a minimum response.
# Firstly we impose iniThFAST. If no corners are detected we impose a lower value minThFAST
# You can lower these values if your images have low contrast			
ORBextractor.iniThFAST: 20 # 20
ORBextractor.minThFAST: 7 # 7

#--------------------------------------------------------------------------------------------
# Viewer Parameters
#--------------------------------------------------------------------------------------------
Viewer.KeyFrameSize: 0.05
Viewer.KeyFrameLineWidth: 1
Viewer.GraphLineWidth: 0.9
Viewer.PointSize: 2
Viewer.CameraSize: 0.08
Viewer.CameraLineWidth: 3
Viewer.ViewpointX: 0
Viewer.ViewpointY: -0.7
Viewer.ViewpointZ: -3.5
Viewer.ViewpointF: 500`

Here. I don't know these parameters are right for your sensor.
but. in my case doing well.

When the result is still wrong,  you should be times 10 or more number to your IMU parameter .

zdegrh, 2020-09-04:
I checked your yaml file with mine, they are all most the same,
any way, thanks!

> https://gist.github.com/HwangDaeHyun/47bf45c880c3e2fbff1ccfafe7d41f77
> 
> Here. I don't know these parameters are right for your sensor.
> but. in my case doing well.
> 
> When the result is still wrong, you should be times 10 or more number to your IMU parameter .

seungsunkim, 2021-05-03:
@HwangDaeHyun 
Hello. I have a question about the value of the yaml file.
I'm using RealSense D455 and I'm using ORB SLAM3's Monocular-inertial.
I'm asking because I don't know the meaning of the Tbc value in the yaml file. What I understood is the alignment value from the sensor to imu, is that right?
Also, when rs-enumerate-devices.exe-c is performed to determine the value, Matrix produces a 3x3 matrix, which Yaml requires a 4x4 matrix. The price is different, what should I do?
This is a picture of the result that I turned around.

[image]

The picture below shows the Tbc value of the yaml file.

[image]