UZ-SLAMLab/ORB_SLAM3 #30: opencv4 & cv.h

opentld, 2020-07-31:
Opencv/cv.h file was cancelled in opencv4, I used opencv2/opencv.hpp instead, but in PnPsolver.h and PnPsolver.cc, cvMat is used for svd calculation. Because opencv2 cancels the definitions of cvMat and cvSVD, So if opencv4 is used, how to solve these problems?
I think you are using opencv3.2. There should be the same problem. How did you solve it?

opentld, 2020-07-31:
I found the solutions. Close question.

closed 2020-07-31 by opentld

shubhendu1907, 2020-08-02:
How did you solve the problem? I am using OpenCV 4.2.0. I used this: https://github.com/Windfisch/ORB_SLAM2 to make it Opencv4 compatible but still keep getting Segmentation Fault error. I also changed the yaml file format for KITTI dataset.

OAkyildiz, 2020-10-30:
> I found the solutions. Close question.

@opentld we would appreciate it if you share the solution you founds, so the users with similar problems can benefit from it.

antithing, 2021-08-10:
I had the same issue, seems to be solved by adding:

#include <opencv2/core/core_c.h>