UZ-SLAMLab/ORB_SLAM3 #33: "Segmentation fault (core dump)" is reported when executing TUM_VI dataset

shanpenghui, 2020-07-31:
When I run the dataset-room4_512_16 dataset, I encounter the problem:

```
Segmentation fault      (core dumped)
```

The whole message is:
```
sph@sph-System-Product-Name:~/Documents/ORB_SLAM3_Fixed$ ./tum_vi.sh 
Launching Room 4 with Monocular-Inertial sensor
num_seq = 1
file name: dataset-room4_512_monoi
Loading images for sequence 0.../home/sph/Downloads/dataset-room4_512_16/mav0/cam0/data
Examples/Monocular-Inertial/TUM_TimeStamps/dataset-room4_512.txt
LOADED!
Loading IMU for sequence 0...LOADED!
vstrImageFilenames[seq].size() = 2228

-------

ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Monocular-Inertial

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name: dataset-room4_512_monoi

Camera Parameters: 
- fx: 190.97846984863281
- fy: 190.97331237792969
- cx: 254.93170166015625
- cy: 256.89743041992188
- bf: 0
- k1: 0
- k2: 0
- p1: 0
- p2: 0
- k3: -0.0020532361231744289
- fps: 20
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 1500
- Scale Levels: 8
- Scale Factor: 1.2000000476837158
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

Left camera to Imu Transform (Tbc): 
[-0.99952501, 0.0075019184, -0.029890131, 0.045574836;
 0.029615344, -0.03439736, -0.99896932, -0.071161799;
 -0.008522328, -0.99938005, 0.034158852, -0.044681255;
 0, 0, 0, 1]

IMU frequency: 200 Hz
IMU gyro noise: 0.00015999999595806003 rad/s/sqrt(Hz)
IMU gyro walk: 2.2000000171829015e-05 rad/s^2/sqrt(Hz)
IMU accelerometer noise: 0.00279999990016222 m/s^2/sqrt(Hz)
IMU accelerometer walk: 0.00085999997099861503 m/s^3/sqrt(Hz)
init extractor
init extractor
init extractor
init extractor
init extractor
init extractor
init extractor
init extractor
init extractor
init extractor
init extractor
init extractor
init extractor
init extractor
init extractor
init extractor
init extractor
init extractor
init extractor
init extractor
init extractor
init extractor
init extractor
init extractor
init extractor
init extractor
init extractor
init extractor
init extractor
init extractor
init extractor
init extractor
init extractor
init extractor
init extractor
init extractor
init extractor
init extractor
init extractor
init extractor
init extractor
init extractor
init extractor
init extractor
init extractor
init extractor
init extractor
init extractor
init extractor
init extractor
init extractor
init extractor
init extractor
init extractor
init extractor
init extractor
init extractor
init extractor
init extractor
First KF:0; Map init KF:0
New Map created with 386 points
start inserting MPs
./tum_vi.sh: line 17: 10817 Segmentation fault      (core dumped) ./Examples/Monocular-Inertial/mono_inertial_tum_vi Vocabulary/ORBvoc.txt Examples/Monocular-Inertial/TUM_512.yaml "$pathDatasetTUM_VI"/dataset-room4_512_16/mav0/cam0/data Examples/Monocular-Inertial/TUM_TimeStamps/dataset-room4_512.txt Examples/Monocular-Inertial/TUM_IMU/dataset-room4_512.txt dataset-room4_512_monoi
```

My tum_vi.sh file is:
```
#!/bin/bash
pathDatasetTUM_VI='/home/sph/Downloads' #Example, it is necesary to change it by the dataset path
#------------------------------------
# Monocular-Inertial Examples
echo "Launching Room 4 with Monocular-Inertial sensor"
./Examples/Monocular-Inertial/mono_inertial_tum_vi Vocabulary/ORBvoc.txt Examples/Monocular-Inertial/TUM_512.yaml "$pathDatasetTUM_VI"/dataset-room4_512_16/mav0/cam0/data Examples/Monocular-Inertial/TUM_TimeStamps/dataset-room4_512.txt Examples/Monocular-Inertial/TUM_IMU/dataset-room4_512.txt dataset-room4_512_monoi
```

I use gdb to find where is dumped, I found that:
```
warning: Unexpected size of section `.reg-xstate/8693' in core file.
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Core was generated by `./Examples/Monocular-Inertial/mono_inertial_tum_vi Vocabulary/ORBvoc.txt Exampl'.
Program terminated with signal SIGSEGV, Segmentation fault.

warning: Unexpected size of section `.reg-xstate/8693' in core file.
#0  0x00007fe0aa1bbd8d in Eigen::DenseStorage<double, 4, 4, 1, 0>::DenseStorage (other=..., this=0x42780000ffffffff)
    at /usr/include/eigen3/Eigen/src/Core/DenseStorage.h:194
194	    DenseStorage(const DenseStorage& other) : m_data(other.m_data) {
[Current thread is 1 (Thread 0x7fe0aa69ce80 (LWP 8693))]
```
Can someone help? Thank you very very very very much ~~~

anakin0620, 2020-08-01:
#3

shanpenghui, 2020-08-01:
> #3

Seems not to work, becauser there is camera type in the TUM_VI_512.yaml

anakin0620, 2020-08-09:
@shanpenghui  Could you post TUM_VI_512.yaml here?

jjgr3496, 2020-08-11:
Hi @shanpenghui, it seems like the calibration file you are providing to the program is not correct. From the output you posted I cannot see any of the Kannala-Brandt distortion parameters. Could please, try the following?:
-Download the latest commit of ORB-SLAM3 and check if you error persists.
-Post here the calibration file (.yaml) you are using.

Thank you.

closed 2020-10-05 by jdtardos