UZ-SLAMLab/ORB_SLAM3 #34: While executing  TUM2 and TUM1 dataset in RGB-D mode,it is interrupted with "Segmentation fault (core dump) "

aiyouaiyouhehehe, 2020-07-31:
~/src/ORB_SLAM3$ ./Examples/RGB-D/rgbd_tum Vocabulary/ORBvoc.txt Examples/RGB-D/TUM1.yaml /home/zd/src/ORB_SLAM3/data/rgbd_dataset_freiburg1_room /home/zd/src/ORB_SLAM3/Examples/RGB-D/associations.txt

ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: RGB-D

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name: 

Camera Parameters: 
- fx: 517.306
- fy: 516.469
- cx: 318.643
- cy: 255.314
- bf: 40
- k1: 0.262383
- k2: -0.953104
- p1: -0.005358
- p2: 0.002628
- k3: 1.16331
- fps: 30
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 1000
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

Depth Threshold (Close/Far Points): 3.09294

-------
Start processing sequence ...
Images in the sequence: 1352

First KF:0; Map init KF:0
New Map created with 852 points
BoW: 248 matches between 683 points with coarse Sim3
terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(4.3.0) /home/zd/app/opencv-4.3.0/modules/imgproc/src/color.cpp:182: error: (-215:Assertion failed) !_src.empty() in function 'cvtColor'

Segmentation fault (core dump)

When  executing tum3, it is all right. I don't know why.

mxpfeng, 2020-08-04:
I have the same problem. 
My environmental information:
OS: Ubuntu 18.04 LTS
Sensor:realsense d435i

aiyouaiyouhehehe, 2020-08-10:
> 
> 
> I have the same problem.
> My environmental information:
> OS: Ubuntu 18.04 LTS
> Sensor:realsense d435i

Did you try the tum1 and tum2 in RGB-D mode？

jjgr3496, 2020-08-11:
Hi all. We have solved some bugs related with RGBD cameras in the latest commits. Could you please download the latest commit and check if you error keeps poping up?

Regards.

aiyouaiyouhehehe, 2020-08-11:
> 
> 
> Hi all. We have solved some bugs related with RGBD cameras in the latest commits. Could you please download the latest commit and check if you error keeps poping up?
> 
> Regards.

Thanks very much！It works！

closed 2020-08-11 by aiyouaiyouhehehe