UZ-SLAMLab/ORB_SLAM3 #36: using webcam error

opentld, 2020-08-01:
when I using webcam as MONOCULAR camera, error occurs:
[image]
it seems that the code collapses at the Tracking.cc 
[image]

does anyone know the reasons?

opentld, 2020-08-03:
`#include<iostream>
#include<algorithm>
#include<fstream>
#include<chrono>

#include<opencv2/core/core.hpp>

#include<System.h>

#include<time.h>

using namespace std;

// From http://www.euclideanspace.com/maths/geometry/rotations/conversions/matrixToQuaternion/

int main(int argc, char **argv)
{
	string vocabPath = "ORBvoc.txt";
	string settingsPath = "webcam.yaml";
	if (argc == 1)
	{

	}
	else if (argc == 2)
	{
		vocabPath = argv[1];
	}
	else if (argc == 3)
	{
		vocabPath = argv[1];
		settingsPath = argv[2];
	}
    else
    {
        cerr << endl << "Usage: mono_webcam.exe path_to_vocabulary path_to_settings" << endl;
        return 1;
    }

    // Create SLAM system. It initializes all system threads and gets ready to process frames.
    ORB_SLAM3::System SLAM(vocabPath, settingsPath,ORB_SLAM3::System::MONOCULAR,true);

    cout << endl << "-------" << endl;
    cout << "Start processing sequence ..." << endl;

	cv::VideoCapture cap(0);

	// From http://stackoverflow.com/questions/19555121/how-to-get-current-timestamp-in-milliseconds-since-1970-just-the-way-java-gets
	__int64 now = std::chrono::duration_cast<std::chrono::milliseconds>(std::chrono::system_clock::now().time_since_epoch()).count();

	// Main loop
	cv::Mat im;
	cv::Mat Tcw;
    while (true)
    {
		cap.read(im);

		if (im.empty())
			continue;

		__int64 curNow = std::chrono::duration_cast<std::chrono::milliseconds>(std::chrono::system_clock::now().time_since_epoch()).count();

		// Pass the image to the SLAM system
		Tcw = SLAM.TrackMonocular(im, curNow / 1000.0);

		/* This can write each image with its position to a file if you want
		if (!Tcw.empty())
		{
			cv::Mat Rwc = Tcw.rowRange(0, 3).colRange(0, 3).t();
			cv::Mat twc = -Rwc*Tcw.rowRange(0, 3).col(3);
			std::ostringstream stream;
			//stream << "imgs/" << Rwc.at<float>(0, 0) << " " << Rwc.at<float>(0, 1) << " " << Rwc.at<float>(0, 2) << " " << twc.at<float>(0) << " " <<
			//	Rwc.at<float>(1, 0) << " " << Rwc.at<float>(1, 1) << " " << Rwc.at<float>(1, 2) << " " << twc.at<float>(1) << " " <<
				//Rwc.at<float>(2, 0) << " " << Rwc.at<float>(2, 1) << " " << Rwc.at<float>(2, 2) << " " << twc.at<float>(2) << ".jpg";
			stream << "imgs/" << curNow << ".jpg";
			string fileName = stream.str();
			cv::imwrite(fileName, im);
		}
		*/

		// This will make a third window with the color images, you need to click on this then press any key to quit
		cv::imshow("Image", im);

		if (cv::waitKey(1) >= 0)
			break;
    }

    // Stop all threads
    SLAM.Shutdown();
	cap.release();

    // Save camera trajectory
    SLAM.SaveKeyFrameTrajectoryTUM("KeyFrameTrajectory.txt");

    return 0;
}
`

here are my codes, why goes wrong? @richard-elvira @jmm-montiel @jjgr3496 @jdtardos

jjgr3496, 2020-08-03:
Hi,
Could you provide us the calibration file you are using along the error message you are getting?

opentld, 2020-08-04:
%YAML:1.0

#--------------------------------------------------------------------------------------------
# Camera Parameters. Adjust them!
#--------------------------------------------------------------------------------------------

# Camera calibration and distortion parameters (OpenCV) 
Camera.fx: 458.654
Camera.fy: 457.296
Camera.cx: 367.215
Camera.cy: 248.375

Camera.k1: -0.28340811
Camera.k2: 0.07395907
Camera.p1: 0.00019359
Camera.p2: 1.76187114e-05

# Camera frames per second 
Camera.fps: 20.0

# Color order of the images (0: BGR, 1: RGB. It is ignored if images are grayscale)
Camera.RGB: 1

#--------------------------------------------------------------------------------------------
# ORB Parameters
#--------------------------------------------------------------------------------------------

# ORB Extractor: Number of features per image
ORBextractor.nFeatures: 1000

# ORB Extractor: Scale factor between levels in the scale pyramid 	
ORBextractor.scaleFactor: 1.2

# ORB Extractor: Number of levels in the scale pyramid	
ORBextractor.nLevels: 8

# ORB Extractor: Fast threshold
# Image is divided in a grid. At each cell FAST are extracted imposing a minimum response.
# Firstly we impose iniThFAST. If no corners are detected we impose a lower value minThFAST
# You can lower these values if your images have low contrast			
ORBextractor.iniThFAST: 20
ORBextractor.minThFAST: 7

#--------------------------------------------------------------------------------------------
# Viewer Parameters
#---------------------------------------------------------------------------------------------
Viewer.KeyFrameSize: 0.05
Viewer.KeyFrameLineWidth: 1
Viewer.GraphLineWidth: 0.9
Viewer.PointSize:2
Viewer.CameraSize: 0.08
Viewer.CameraLineWidth: 3
Viewer.ViewpointX: 0
Viewer.ViewpointY: -0.7
Viewer.ViewpointZ: -1.8
Viewer.ViewpointF: 500

-----------------------------------------------------------------------------------------------
did you mean the yml file?  
There is no error message output, the program jumps directly out of the "WAITING FOR IMAGES" window.
I try to track the error, found that the problem may be in Tracking.cc, when getting the mCurrentFrame, the Frame function failed to execute correctly, it exited directly, and did not run into the Frame function.
@jjgr3496

jjgr3496, 2020-08-04:
Thank you for posting the calibration file (the .yaml file). The error you are getting is likely caused because you need to add one of the following lines to your calibration file, depending of wich kind of camera you are using:

`Camera.type: "PinHole" #For a pinhole camera`

or

`Camera.type: "KannalaBrandt8" #For a fisheye camera using the KannalaBrand camera model`

closed 2020-08-05 by jjgr3496

opentld, 2020-08-06:
it works! thank you! @jjgr3496