UZ-SLAMLab/ORB_SLAM3 #37: How to run in real time with stereo/mono camera + imu ?

Shame-fight, 2020-08-03:
Thank you for your work. 1. Is this program currently only able to run data sets and cannot be run in real time with the camera (rgbd/stereo/mono)? If possible, which programs need to be modified.
2. Do you have a plan to complete the function of saving the map.
3.Whether the map can be used for navigation, as far as I know, sparse maps cannot be used for navigation and obstacle avoidance.
Looking forward to receiving your reply~

xiefei2929, 2020-08-03:
I committed two visual-inertial ROS interfaces Mono_inertial and Stereo_inertial at    https://github.com/xiefei2929/ORB_SLAM3-ROS-Interface    which can be run in real time with the camera stereo+imu/mono+imu
I also modified RGBD-inertial mode and committed  its ROS interface base on ORB-SLAM3 which can be run in real time with the camera RGBD+imu, the codes at    https://github.com/xiefei2929/ORB_SLAM3-RGBD-Inertial

Shame-fight, 2020-08-04:
> I committed two visual-inertial ROS interfaces Mono_inertial and Stereo_inertial at https://github.com/xiefei2929/ORB_SLAM3-ROS-Interface which can be run in real time with the camera stereo+imu/mono+imu.
> I also modified RGBD-inertial mode and committed its ROS interface base on ORB-SLAM3 which can be run in real time with the camera RGBD+imu such as Realsense D435i, the codes at https://github.com/xiefei2929/ORB_SLAM3-RGBD-Inertial
> 
> In addition, only after you calibrate your equipment accurately can it work normally. Inaccurate calibration will raise the times of VIBA iterations and cause stuck.

Hi,thank you. I tried to compile ORB_SLAM3-ROS-Interface with the command'./build.sh', but an error occurred: The file Examples/Monocular-Inertial/mono_inertial_tum_vi.cc, etc. is missing. I have a doubt: it is to compile your ROS directly -interface or replace your ros package with /Examples/ROS/ORB_SLAM3 of ORB_SLAM3?

xiefei2929, 2020-08-04:
Just delete the Examples which aren't there in CMakelist. I updated Cmakelist, now it is fixed

Shame-fight, 2020-08-04:
> Just delete the Examples which aren't there in CMakelist. I updated Cmakelist, now it is fixed
好的.我运行了程序,但一直在reset,没有形成轨迹.我不确定是不是没有标定小觅相机导致的.我修改了imu和image订阅的话题,没有修改EuRoC.yaml.打印的信息如下
`First KF:1124; Map init KF:1123
New Map created with 116 points
Fail to track local map!
IMU is not or recently initialized. Reseting active map...
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 4205
mnInitialFrameId = 4203
2948 Frames had been set to lost
last KF is empty!
not IMU meas
last KF is empty!`

Shame-fight, 2020-08-04:
> Just delete the Examples which aren't there in CMakelist. I updated Cmakelist, now it is fixed

好的.我运行了程序,但一直在reset,没有形成轨迹.我不确定是不是没有标定小觅相机导致的.我修改了imu和image订阅的话题,没有修改EuRoC.yaml.打印的信息如下
First KF:1124; Map init KF:1123 
New Map created with 116 points 
Fail to track local map! 
IMU is not or recently initialized. Reseting active map... 
LM: Active map reset recieved 
LM: Active map reset, waiting... 
LM: Reseting current map in Local Mapping... 
LM: End reseting Local Mapping... 
LM: Reset free the mutex 
LM: Active map reset, Done!!! 
mnFirstFrameId = 4205 
mnInitialFrameId = 4203 2948 
Frames had been set to lost last KF is empty! 
not IMU meas 
last KF is empty!

xiefei2929, 2020-08-05:
必须要标定的

478161648, 2020-10-04:
> 必须要标定的

老哥儿，知道这三个参数是啥吗？看到一个问答里说，好的imu就填低一点 差得imu就填高一点，自己估摸着填吗？
IMU.NoiseGyro: 1.7e-04 # 1.6968e-04
IMU.NoiseAcc: 2.0e-03 # 2.0000e-3
IMU.GyroWalk: 1.9393e-05
IMU.AccWalk: 3.e-03 # 3.0000e-3

zthcool, 2020-10-22:
> > 必须要标定的
> 
> 老哥儿，知道这三个参数是啥吗？看到一个问答里说，好的imu就填低一点 差得imu就填高一点，自己估摸着填吗？
> IMU.NoiseGyro: 1.7e-04 # 1.6968e-04
> IMU.NoiseAcc: 2.0e-03 # 2.0000e-3
> IMU.GyroWalk: 1.9393e-05
> IMU.AccWalk: 3.e-03 # 3.0000e-3
这个也是标定出来的，kalibr标定工具箱

jdtardos, 2021-12-23:
Version 1.0 now includes  examples to run ORB-SLAM3 with Intel Realsense cameras

closed 2021-12-23 by jdtardos