UZ-SLAMLab/ORB_SLAM3 #38: ORB_SLAM3  ROS ERROR！！！！

fox-wangguodong, 2020-08-04:
fox@YangTianM6201c-00:~/catkin_kinect$ rosrun ORB_SLAM3  RGBD ORBvoc.txt  Asus_d435i.yaml 

ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: RGB-D

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name: 

Camera Parameters: 
- fx: 614.859
- fy: 614.964
- cx: 325.856
- cy: 228.775
- bf: 30.7429
- k1: 0
- k2: 0
- p1: 0
- p2: 0
- fps: 30
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 1000
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

Depth Threshold (Close/Far Points): 2.5
First KF:0; Map init KF:0
New Map created with 289 points
段错误 (核心已转储)

fox-wangguodong, 2020-08-04:
message_filters::Subscriber<sensor_msgs::Image> rgb_sub(nh, "/camera/color/image_raw", 1);
    message_filters::Subscriber<sensor_msgs::Image> depth_sub(nh, "/camera/depth/image_rect_raw", 1);

that is my realsense d435i topic

anakin0620, 2020-08-05:
#3 Common error. You could find solution there.

xiefei2929, 2020-08-05:
I modified RGBD-inertial mode and committed its ROS interface base on ORB-SLAM3 which can be run in real time with the camera RGBD+imu such as Realsense D435i, the codes at https://github.com/xiefei2929/ORB_SLAM3-RGBD-Inertial

fox-wangguodong, 2020-08-05:
> I modified RGBD-inertial mode and committed its ROS interface base on ORB-SLAM3 which can be run in real time with the camera RGBD+imu such as Realsense D435i, the codes at https://github.com/xiefei2929/ORB_SLAM3-RGBD-Inertial

I refer to your code and encountered the same problem ： 
fox@YangTianM6201c-00:~/ORB_SLAM3-RGBD-Inertial$ rosrun ORB_SLAM3 RGBD_inertial ORBvoc.txt Asus_d435i.yaml 

ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: RGBD-Inertial

Loading ORB Vocabulary. This could take a while...
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
已放弃 (核心已转储)

xiefei2929, 2020-08-05:
You used wrong command, my vocabulary file is ORBvoc.bin
And please make sure your Asus_d435i.yaml has the same form as the Myd435i.yaml which I committed

fox-wangguodong, 2020-08-06:
> You used wrong command, my vocabulary file is ORBvoc.bin
> And please make sure your Asus_d435i.yaml has the same form as the Myd435i.yaml which I committed

fox@YangTianM6201c-00:~$ rostopic list 

/camera/accel/imu_info
/camera/accel/sample
/camera/color/camera_info
/camera/color/image_raw
/camera/color/image_raw/compressed
/camera/color/image_raw/compressed/parameter_descriptions
/camera/color/image_raw/compressed/parameter_updates
/camera/color/image_raw/compressedDepth
/camera/color/image_raw/compressedDepth/parameter_descriptions
/camera/color/image_raw/compressedDepth/parameter_updates
/camera/color/image_raw/theora
/camera/color/image_raw/theora/parameter_descriptions
/camera/color/image_raw/theora/parameter_updates
/camera/depth/camera_info
/camera/depth/image_rect_raw
/camera/depth/image_rect_raw/compressed
/camera/depth/image_rect_raw/compressed/parameter_descriptions
/camera/depth/image_rect_raw/compressed/parameter_updates
/camera/depth/image_rect_raw/compressedDepth
/camera/depth/image_rect_raw/compressedDepth/parameter_descriptions
/camera/depth/image_rect_raw/compressedDepth/parameter_updates
/camera/depth/image_rect_raw/theora
/camera/depth/image_rect_raw/theora/parameter_descriptions
/camera/depth/image_rect_raw/theora/parameter_updates
/camera/extrinsics/depth_to_color
/camera/extrinsics/depth_to_infra1
/camera/extrinsics/depth_to_infra2
/camera/gyro/imu_info
/camera/gyro/sample
/camera/infra1/camera_info
/camera/infra1/image_rect_raw
/camera/infra1/image_rect_raw/compressed
/camera/infra1/image_rect_raw/compressed/parameter_descriptions
/camera/infra1/image_rect_raw/compressed/parameter_updates
/camera/infra1/image_rect_raw/compressedDepth
/camera/infra1/image_rect_raw/compressedDepth/parameter_descriptions
/camera/infra1/image_rect_raw/compressedDepth/parameter_updates
/camera/infra1/image_rect_raw/theora
/camera/infra1/image_rect_raw/theora/parameter_descriptions
/camera/infra1/image_rect_raw/theora/parameter_updates
/camera/infra2/camera_info
/camera/infra2/image_rect_raw
/camera/infra2/image_rect_raw/compressed
/camera/infra2/image_rect_raw/compressed/parameter_descriptions
/camera/infra2/image_rect_raw/compressed/parameter_updates
/camera/infra2/image_rect_raw/compressedDepth
/camera/infra2/image_rect_raw/compressedDepth/parameter_descriptions
/camera/infra2/image_rect_raw/compressedDepth/parameter_updates
/camera/infra2/image_rect_raw/theora
/camera/infra2/image_rect_raw/theora/parameter_descriptions
/camera/infra2/image_rect_raw/theora/parameter_updates
/camera/motion_module/parameter_descriptions
/camera/motion_module/parameter_updates
/camera/realsense2_camera_manager/bond
/camera/rgb_camera/auto_exposure_roi/parameter_descriptions
/camera/rgb_camera/auto_exposure_roi/parameter_updates
/camera/rgb_camera/parameter_descriptions
/camera/rgb_camera/parameter_updates
/camera/stereo_module/auto_exposure_roi/parameter_descriptions
/camera/stereo_module/auto_exposure_roi/parameter_updates
/camera/stereo_module/parameter_descriptions
/camera/stereo_module/parameter_updates
/diagnostics
/rosout
/rosout_agg
/tf
/tf_static

How to fill in topic below ？

    ros::Subscriber sub_imu = n.subscribe("/camera/imu", 2000, imu_callback, ros::TransportHints().tcpNoDelay());
    ros::Subscriber sub_img = n.subscribe("/camera/color/image_raw", 100, img_callback);
    ros::Subscriber sub_img_depth = n.subscribe("/camera/depth/image_rect_raw", 100, img_depth_callback);

xiefei2929, 2020-08-06:
You don't have topic "/camera/imu".
Open your d435i launch file such as rs_camera.launch 
find this line and set default="copy"
<arg name="unite_imu_method"          default="copy"/>
then you will get the topic "/camera/imu"

fox-wangguodong, 2020-08-06:
> You don't have topic "/camera/imu".
> Open your d435i launch file such as rs_camera.launch
> find this line and set default="copy"
> 
> arg name="unite_imu_method" default="copy"
> 
> then you will get the topic "/camera/imu"

fox@YangTianM6201c-00:~/ORB_SLAM3-RGBD-Inertial$ rosrun ORB_SLAM3 RGBD_inertial Vocabulary/ORBvoc.bin Myd435i.yaml 

ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: RGBD-Inertial

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name: 

Camera Parameters: 
- fx: 611.738
- fy: 610.111
- cx: 313.172
- cy: 252.256
- bf: 40
- k1: 0
- k2: 0
- p1: 0
- p2: 0
- fps: 30
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 1000
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

Depth Threshold (Close/Far Points): 2.6155

Left camera to Imu Transform (Tbc): 
[0.00087480003, -0.0047406, 0.99998838, 1.30441e-05;
 -0.99996811, -0.0079460004, 0.00083709997, 0.0149225;
 0.0079418998, -0.9999572, -0.0047474001, 0.00031683501;
 0, 0, 0, 1]

IMU frequency: 500 Hz
IMU gyro noise: 0.01 rad/s/sqrt(Hz)
IMU gyro walk: 2e-05 rad/s^2/sqrt(Hz)
IMU accelerometer noise: 0.1 m/s^2/sqrt(Hz)
IMU accelerometer walk: 0.0002 m/s^3/sqrt(Hz)
[ WARN] [1596709517.198844258]: waiting for image and imu...

xiefei2929, 2020-08-06:
It seems running well, just haven't subscribed the right topic. You can use Rviz to check wether there are messages on these topics.

fox-wangguodong, 2020-08-06:
> It seems running well, just haven't subscribed the right topic. You can use Rviz to check wether there are messages on these topics.

i try to run orb-slam2 to test topic，and topic is well

fox-wangguodong, 2020-08-06:
> > It seems running well, just haven't subscribed the right topic. You can use Rviz to check wether there are messages on these topics.
> 
> i try to run orb-slam2 to test topic，and topic is well

sometime it print “throw img”

xiefei2929, 2020-08-06:
which  means that your depth frame and rgb frame do not arrive at the same time
You should enable the timestamp synchronization

arg name="enable_sync"               default="true"/

fox-wangguodong, 2020-08-06:
> which means that your depth frame and rgb frame do not arrive at the same time
> You should enable the timestamp synchronization
> 
> arg name="enable_sync" default="true"/

ok，i will to it tomorrow

xiefei2929, 2020-08-06:
By the way, your device need accurate calibration, for example using https://github.com/ethz-asl/kalibr

closed 2020-08-10 by richard-elvira