UZ-SLAMLab/ORB_SLAM3 #39: Did stereo-inertial provide an online calibration for extrinsic parameters and time?

qiaozhijian, 2020-08-04:


xiefei2929, 2020-08-05:
Seems not. 
It relies heavily on calibration accuracy.

jdtardos, 2020-08-06:
No, you should perform off-line calibration, for example using https://github.com/ethz-asl/kalibr

qiaozhijian, 2020-08-06:
I tested it on my robot and I made sure that my timestamp and external calibration are accurate (calibrated by kalibr), but the system crashes because of a bad BA everytime

jdtardos, 2020-08-06:
Could you please provide more information about the crash?

qiaozhijian, 2020-08-06:
yes! I'm willing! it is as follows:

`last KF is empty!
9.4599269222115634
not enough acceleration
last KF is empty!
First KF:0; Map init KF:0
New Map created with 477 points
IMU in Map 0 is initialized
start VIBA 1
end VIBA 1
Not enough motion for initializing. Reseting...
TRACK: Reset map because local mapper set the bad imu flag 
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 159
mnInitialFrameId = 0
164 Frames had been set to lost
last KF is empty!
not IMU meas
last KF is empty!
9.3207136995208799
not enough acceleration
last KF is empty!
First KF:26; Map init KF:0
New Map created with 203 points`

Vins-fusion and open-vins can achieve good results on the same data set. If needed, I can provide my dataset and parameter file.

qiaozhijian, 2020-08-06:
- last KF is empty!
9.4698022330497551
not enough acceleration
last KF is empty!
9.4720551031521705
not enough acceleration
last KF is empty!
9.4578314637517362
not enough acceleration
last KF is empty!
9.4599269222115634
not enough acceleration
last KF is empty!
First KF:0; Map init KF:0
New Map created with 477 points
IMU in Map 0 is initialized
start VIBA 1
end VIBA 1
Not enough motion for initializing. Reseting...
TRACK: Reset map because local mapper set the bad imu flag 
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 159
mnInitialFrameId = 0
164 Frames had been set to lost
last KF is empty!
not IMU meas
last KF is empty!
9.3207136995208799
not enough acceleration
last KF is empty!
First KF:26; Map init KF:0
New Map created with 203 points

xiefei2929, 2020-08-06:
VINS has great online calibration capability, you cannot make sure your  parameters are  accurate.

qiaozhijian, 2020-08-06:
> VINS has great online calibration capability, you cannot make sure your parameters are accurate.

maybe. But I see the result of VINS online calibration is not  accurate, so I don't use online calibration when I use VINS. but I use in open-vins

ccamposm, 2020-08-06:
It seems there is not enough motion for accurately initializing inertial parameters (Maybe camera remains steady for some time just after initializing?). However, since you are using a stereo-inertial setup, you could try commenting from line 138 to 145 at LocalMapping.cc:
https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/master/src/LocalMapping.cc#L138
This check is specially important for monocular-inertial case, where scale also needs to be estimated, but not for stereo-inertial one.

qiaozhijian, 2020-08-06:
Thank you for your suggestion. I modified it according to your suggestion. My system did run for a while longer than before (although the effect was not as good as STEREO), but after a while the robot began to "fall into the ground", although my robot only moved on the ground.

qiaozhijian, 2020-08-06:
> It seems there is not enough motion for accurately initializing inertial parameters (Maybe camera remains steady for some time just after initializing?). However, since you are using a [...]

Thank you for your suggestion. I modified it according to your suggestion. My system did run for a while longer than before (although the effect was not as good as STEREO), but after a while the robot began to "fall into the ground", although my robot only moved on the ground.

qiaozhijian, 2020-08-06:
> It seems there is not enough motion for accurately initializing inertial parameters (Maybe camera remains steady for some time just after initializing?). However, since you are using a [...]

My robot does walk for a while and stops for a while, and it only has two modes of motion: straight and rotating respectively.

ccamposm, 2020-08-06:
Thank you for the update.
Inertial parameters estimation requires enough accelerometer and gyroscope excitation, which is achieved by performing aggressive motion. In addition, having gravity vector always pointing at the same direction (i.e ground robot) usually leads to bad results. Is that your case?
You could also check values for estimated biases, they should be close to zero.

qiaozhijian, 2020-08-06:
Yes, it's my case. And you are right: bias of ACC_Z is too big

qiaozhijian, 2020-08-06:
> Thank you for the update.
> Inertial parameters estimation requires enough accelerometer and gyroscope excitation, which is achieved by performing aggressive motion. In addition, having gravity vector [...]

Yes, it's my case. And you are right: bias of ACC_Z is too big

ccamposm, 2020-08-06:
You could try increasing prior information for accelerometer bias to force it to remain close to zero (increase input argument priorA at https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/master/src/LocalMapping.cc#L167)
or even fixing acc bias to zero during initialization since it is quite likely to converge to the right value during posterior Local BAs.

closed 2020-08-13 by jdtardos