UZ-SLAMLab/ORB_SLAM3 #40: make failed

zhaishilin, 2020-08-05:
"CMakeFiles/ORB_SLAM3.dir/build.make:101: recipe for target 'CMakeFiles/ORB_SLAM3.dir/src/LoopClosing.cc.o' failed
make[2]: *** [CMakeFiles/ORB_SLAM3.dir/src/LoopClosing.cc.o] Error 1
CMakeFiles/Makefile2:183: recipe for target 'CMakeFiles/ORB_SLAM3.dir/all' failed"

Dose anyone encountered this problem? Thank you very much for help

ns15417, 2020-08-06:
you didn't state you problem clearly, you should post the exact error, not here the filename

belal-ibrahim, 2020-08-09:
|                   ^
/home/beraru/ORB_SLAM3/src/Optimizer.cc: In static member function ‘static int ORB_SLAM3::Optimizer::PoseInertialOptimizationLastKeyFrame(ORB_SLAM3::Frame*, bool)’:
/home/beraru/ORB_SLAM3/src/Optimizer.cc:7706:10: warning: unused variable ‘bOut’ [-Wunused-variable]
 7706 |     bool bOut = false;
      |          ^~~~
/home/beraru/ORB_SLAM3/src/LocalMapping.cc: In member function ‘void ORB_SLAM3::LocalMapping::KeyFrameCulling()’:
/home/beraru/ORB_SLAM3/src/LocalMapping.cc:1057:48: warning: ‘last_ID’ may be used uninitialized in this function [-Wmaybe-uninitialized]
 1057 |                     if((bInitImu && (pKF->mnId<last_ID) && t<3.) || (t<0.5))
      |                                                ^~~~~~~
make[2]: *** [CMakeFiles/ORB_SLAM3.dir/build.make:102: CMakeFiles/ORB_SLAM3.dir/src/LoopClosing.cc.o] Error 1
/home/beraru/ORB_SLAM3/src/Tracking.cc: In member function ‘void ORB_SLAM3::Tracking::Track()’:
/home/beraru/ORB_SLAM3/src/Tracking.cc:1155:13: warning: ‘bOK’ may be used uninitialized in this function [-Wmaybe-uninitialized]
 1155 |             if(bOK)
      |             ^~
make[2]: *** [CMakeFiles/ORB_SLAM3.dir/build.make:232: CMakeFiles/ORB_SLAM3.dir/src/Optimizer.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:390: CMakeFiles/ORB_SLAM3.dir/all] Error 2
make: *** [Makefile:84: all] Error 2

jjgr3496, 2020-08-11:
Hi @zhaishilin, unfortunately you didn't provide enough information about your error. Could you please post the full output when you try to build the library?

closed 2020-08-18 by jjgr3496