UZ-SLAMLab/ORB_SLAM3 #46: building issue

belal-ibrahim, 2020-08-09:
I am trying to build it using ./build.sh and i got this type of error:
note that i am using ubuntu 20.04 
/home/beraru/ORB_SLAM3/src/Optimizer.cc: In static member function ‘static void ORB_SLAM3::Optimizer::MergeInertialBA(ORB_SLAM3::KeyFrame*, ORB_SLAM3::KeyFrame*, bool*, ORB_SLAM3::Map*, ORB_SLAM3::LoopClosing::KeyFrameAndPose&)’:
/home/beraru/ORB_SLAM3/src/Optimizer.cc:7015:19: warning: suggest explicit braces to avoid ambiguous ‘else’ [-Wdangling-else]
 7015 |                 if(!pMP->isBad())
      |                   ^
/home/beraru/ORB_SLAM3/src/Optimizer.cc: In static member function ‘static int ORB_SLAM3::Optimizer::PoseInertialOptimizationLastKeyFrame(ORB_SLAM3::Frame*, bool)’:
/home/beraru/ORB_SLAM3/src/Optimizer.cc:7706:10: warning: unused variable ‘bOut’ [-Wunused-variable]
 7706 |     bool bOut = false;
      |          ^~~~
/home/beraru/ORB_SLAM3/src/LocalMapping.cc: In member function ‘void ORB_SLAM3::LocalMapping::KeyFrameCulling()’:
/home/beraru/ORB_SLAM3/src/LocalMapping.cc:1057:48: warning: ‘last_ID’ may be used uninitialized in this function [-Wmaybe-uninitialized]
 1057 |                     if((bInitImu && (pKF->mnId<last_ID) && t<3.) || (t<0.5))
      |                                                ^~~~~~~
make[2]: *** [CMakeFiles/ORB_SLAM3.dir/build.make:102: CMakeFiles/ORB_SLAM3.dir/src/LoopClosing.cc.o] Error 1
/home/beraru/ORB_SLAM3/src/Tracking.cc: In member function ‘void ORB_SLAM3::Tracking::Track()’:
/home/beraru/ORB_SLAM3/src/Tracking.cc:1155:13: warning: ‘bOK’ may be used uninitialized in this function [-Wmaybe-uninitialized]
 1155 |             if(bOK)
      |             ^~
make[2]: *** [CMakeFiles/ORB_SLAM3.dir/build.make:232: CMakeFiles/ORB_SLAM3.dir/src/Optimizer.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:390: CMakeFiles/ORB_SLAM3.dir/all] Error 2
make: *** [Makefile:84: all] Error 2

ns15417, 2020-08-11:
Did you post all your error info? I only saw the whole message of all kinds of warnings which can be neglected. For Error 1, it only mentions LoopCLosing.cc.o, are there more details of the log

belal-ibrahim, 2020-08-11:
ORB_SLAM3/src/System.cc:26:10: fatal error: openssl/md5.h: No such file or directory
26 | #include <openssl/md5.h>
| ^~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/ORB_SLAM3.dir/build.make:63: CMakeFiles/ORB_SLAM3.dir/src/System.cc.o] Error 1
/home/beraru/ORB_SLAM3/src/Optimizer.cc:5179:5: warning: "/" within comment [-Wcomment]
5179 | /std::cout << " Construction time: " << t_const << std::endl;
|
In file included from /home/beraru/ORB_SLAM3/include/Atlas.h:27,
from /home/beraru/ORB_SLAM3/include/LocalMapping.h:24,
from /home/beraru/ORB_SLAM3/include/LoopClosing.h:24,
from /home/beraru/ORB_SLAM3/src/LoopClosing.cc:20:
/home/beraru/ORB_SLAM3/include/CameraModels/KannalaBrandt8.h: In constructor ‘ORB_SLAM3::KannalaBrandt8::KannalaBrandt8(std::vector)’:
/home/beraru/ORB_SLAM3/include/CameraModels/KannalaBrandt8.h:105:21: warning: ‘ORB_SLAM3::KannalaBrandt8::precision’ will be initialized after [-Wreorder]
105 | const float precision;
| ^~~~~~~~~
/home/beraru/ORB_SLAM3/include/CameraModels/KannalaBrandt8.h:95:26: warning: ‘std::vector ORB_SLAM3::KannalaBrandt8::mvLappingArea’ [-Wreorder]
95 | std::vector mvLappingArea;
| ^~~~~~~~~~~~~
In file included from /home/beraru/ORB_SLAM3/include/Atlas.h:27,
from /home/beraru/ORB_SLAM3/include/LocalMapping.h:24,
from /home/beraru/ORB_SLAM3/include/LoopClosing.h:24,
from /home/beraru/ORB_SLAM3/src/LoopClosing.cc:20:
/home/beraru/ORB_SLAM3/include/CameraModels/KannalaBrandt8.h:53:9: warning: when initialized here [-Wreorder]
53 | KannalaBrandt8(const std::vector _vParameters) : GeometricCamera(_vParameters), precision(1e-6), mvLappingArea(2,0) ,tvr(nullptr) {
| ^~~~~~~~~~~~~~
In file included from /home/beraru/ORB_SLAM3/include/Atlas.h:27,
from /home/beraru/ORB_SLAM3/include/LocalMapping.h:24,
from /home/beraru/ORB_SLAM3/include/LoopClosing.h:24,
from /home/beraru/ORB_SLAM3/src/LoopClosing.cc:20:
/home/beraru/ORB_SLAM3/include/CameraModels/KannalaBrandt8.h: In constructor ‘ORB_SLAM3::KannalaBrandt8::KannalaBrandt8(std::vector, float)’:
/home/beraru/ORB_SLAM3/include/CameraModels/KannalaBrandt8.h:105:21: warning: ‘ORB_SLAM3::KannalaBrandt8::precision’ will be initialized after [-Wreorder]
105 | const float precision;
| ^~~~~~~~~
/home/beraru/ORB_SLAM3/include/CameraModels/KannalaBrandt8.h:95:26: warning: ‘std::vector ORB_SLAM3::KannalaBrandt8::mvLappingArea’ [-Wreorder]
95 | std::vector mvLappingArea;
| ^~~~~~~~~~~~~
/home/beraru/ORB_SLAM3/include/CameraModels/KannalaBrandt8.h:59:9: warning: when initialized here [-Wreorder]
59 | KannalaBrandt8(const std::vector _vParameters, const float _precision) : GeometricCamera(_vParameters),
| ^~~~~~~~~~~~~~
/home/beraru/ORB_SLAM3/include/CameraModels/KannalaBrandt8.h: In constructor ‘ORB_SLAM3::KannalaBrandt8::KannalaBrandt8(ORB_SLAM3::KannalaBrandt8)’:
/home/beraru/ORB_SLAM3/include/CameraModels/KannalaBrandt8.h:105:21: warning: ‘ORB_SLAM3::KannalaBrandt8::precision’ will be initialized after [-Wreorder]
105 | const float precision;
| ^~~~~~~~~
/home/beraru/ORB_SLAM3/include/CameraModels/KannalaBrandt8.h:95:26: warning: ‘std::vector ORB_SLAM3::KannalaBrandt8::mvLappingArea’ [-Wreorder]
95 | std::vector mvLappingArea;
| ^~~~~~~~~~~~~
/home/beraru/ORB_SLAM3/include/CameraModels/KannalaBrandt8.h:65:9: warning: when initialized here [-Wreorder]
65 | KannalaBrandt8(KannalaBrandt8 pKannala) : GeometricCamera(pKannala->mvParameters), precision(pKannala->precision), mvLappingArea(2,0) ,tvr(nullptr) {
| ^~~~~~~~~~~~~~
In file included from /home/beraru/ORB_SLAM3/include/Atlas.h:27,
from /home/beraru/ORB_SLAM3/include/LocalMapping.h:24,
from /home/beraru/ORB_SLAM3/include/LoopClosing.h:24,
from /home/beraru/ORB_SLAM3/include/Optimizer.h:26,
from /home/beraru/ORB_SLAM3/src/Optimizer.cc:20:
/home/beraru/ORB_SLAM3/include/CameraModels/KannalaBrandt8.h: In constructor ‘ORB_SLAM3::KannalaBrandt8::KannalaBrandt8(std::vector)’:
/home/beraru/ORB_SLAM3/include/CameraModels/KannalaBrandt8.h:105:21: warning: ‘ORB_SLAM3::KannalaBrandt8::precision’ will be initialized after [-Wreorder]
105 | const float precision;
| ^~~~~~~~~
/home/beraru/ORB_SLAM3/include/CameraModels/KannalaBrandt8.h:95:26: warning: ‘std::vector ORB_SLAM3::KannalaBrandt8::mvLappingArea’ [-Wreorder]
95 | std::vector mvLappingArea;
| ^~~~~~~~~~~~~
In file included from /home/beraru/ORB_SLAM3/include/Atlas.h:27,
from /home/beraru/ORB_SLAM3/include/LocalMapping.h:24,
from /home/beraru/ORB_SLAM3/include/LoopClosing.h:24,
from /home/beraru/ORB_SLAM3/include/Optimizer.h:26,
from /home/beraru/ORB_SLAM3/src/Optimizer.cc:20:
/home/beraru/ORB_SLAM3/include/CameraModels/KannalaBrandt8.h:53:9: warning: when initialized here [-Wreorder]
53 | KannalaBrandt8(const std::vector _vParameters) : GeometricCamera(_vParameters), precision(1e-6), mvLappingArea(2,0) ,tvr(nullptr) {
| ^~~~~~~~~~~~~~
[... 1423 lines cut ...]
| ^~~~
/home/beraru/ORB_SLAM3/src/Optimizer.cc:6801:31: warning: comparison of integer expressions of different signedness: ‘int’ and ‘std::vector<ORB_SLAM3::MapPoint*>::size_type’ {aka ‘long unsigned int’} [-Wsign-compare]
6801 | for(int i=0; i<vpMonoMPsOpt.size(); ++i)
| ~^~~~~~~~~~~~~~~~~~~~
/home/beraru/ORB_SLAM3/src/Optimizer.cc:6821:32: warning: comparison of integer expressions of different signedness: ‘int’ and ‘std::vector<ORB_SLAM3::MapPoint*>::size_type’ {aka ‘long unsigned int’} [-Wsign-compare]
6821 | for(int i=0; i<vpStereoMPsOpt.size(); ++i)
| ~^~~~~~~~~~~~~~~~~~~~~~
[... cut ...]
/home/beraru/ORB_SLAM3/src/Optimizer.cc: In static member function ‘static int ORB_SLAM3::Optimizer::PoseInertialOptimizationLastKeyFrame(ORB_SLAM3::Frame*, bool)’:
/home/beraru/ORB_SLAM3/src/Optimizer.cc:7706:10: warning: unused variable ‘bOut’ [-Wunused-variable]
7706 | bool bOut = false;
| ^~~~
make[2]: *** [CMakeFiles/ORB_SLAM3.dir/build.make:102: CMakeFiles/ORB_SLAM3.dir/src/LoopClosing.cc.o] Error 1
make[2]: *** [CMakeFiles/ORB_SLAM3.dir/build.make:232: CMakeFiles/ORB_SLAM3.dir/src/Optimizer.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:390: CMakeFiles/ORB_SLAM3.dir/all] Error 2
make: *** [Makefile:84: all] Error 2

jjgr3496, 2020-08-11:
Hi @belal-ibrahim, I would ask you next time only post the error messages and not all the warnings (either by modifiying the compiler flags at CMakeLists.txt or by only copying the error lines) to facilitate the readability of the message.

Regarding your problem, it looks like a problem when finding the file openssl/md5.h. You should first install in your system the openssl library (https://cloudwafer.com/blog/installing-openssl-on-ubuntu-16-04-18-04/). If after this, errors keep poping up, please copy us the error messages you are getting! 

Regards

ns15417, 2020-08-11:
According to the log, you donot include the file openssl/md5.h which belong to the lib crypto which should be linked in CMakelists.txt. SO you should check whether crypto is installed or rightly installed. OR you can delete the line `#include openssl/md5.h`  in System.cc cuz the code used md5 is noted in System.cc. you can check the function System::CalculateCheckSum() to have a look. it has never been used.

belal-ibrahim, 2020-08-11:
I have installed openssl as mentioned by @jjgr3496 and it solved openssl error. However, i still got some errors
for @ns15417  i have commented openssl in system.cc as  a final trial and that's the only error i got:

/home/beraru/ORB_SLAM3/src/Optimizer.cc:2388:37:   required from here
/usr/include/c++/9/bits/stl_map.h:122:71: error: static assertion failed: std::map must have the same value_type as its allocator
  122 |       static_assert(is_same<typename _Alloc::value_type, value_type>::value,

make[2]: *** [CMakeFiles/ORB_SLAM3.dir/build.make:102: CMakeFiles/ORB_SLAM3.dir/src/LoopClosing.cc.o] Error 1
make[2]: *** [CMakeFiles/ORB_SLAM3.dir/build.make:232: CMakeFiles/ORB_SLAM3.dir/src/Optimizer.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:390: CMakeFiles/ORB_SLAM3.dir/all] Error 2
make: *** [Makefile:84: all] Error 2

jjgr3496, 2020-08-11:
I am not able to reproduce your last error. I have the feeling that is related with your compiler, could you please provide us which compiler and version are you using? (you can find this information in the files generated by CMake, more precisely, at `ORB_SLAM3/cmake-build-debug/CMakeFiles/<your_cmake_version>/CMakeCXXCompiler.cmake`).

In our experiments, we built ORB-SLAM3 with the following:
`set(CMAKE_CXX_COMPILER "/usr/bin/c++")
set(CMAKE_CXX_COMPILER_ARG1 "")
set(CMAKE_CXX_COMPILER_ID "GNU")
set(CMAKE_CXX_COMPILER_VERSION "7.5.0")
set(CMAKE_CXX_COMPILER_VERSION_INTERNAL "")
set(CMAKE_CXX_COMPILER_WRAPPER "")
set(CMAKE_CXX_STANDARD_COMPUTED_DEFAULT "14")`

In any case, the error seems to be related with the data structures used in a for loop. Since I cannot reproduce the error, I cannot guarantee the following solves the error but you can give it a try: swap lines 2388, 2389 and 2390 at Optimizer.cc from:
` for(set<KeyFrame*>::const_iterator sit=spConnections.begin(), send=spConnections.end(); sit!=send; sit++)
        {
            const long unsigned int nIDj = (*sit)->mnId;`

to:
`for(auto pKf : spConnections)
{
	const long unsigned int nIDj = pKf->mnId;`

belal-ibrahim, 2020-08-11:
@jjgr3496   I **am trying to build with:**
```
set(CMAKE_CXX_COMPILER "/usr/bin/c++")
set(CMAKE_CXX_COMPILER_ARG1 "")
set(CMAKE_CXX_COMPILER_ID "GNU")
set(CMAKE_CXX_COMPILER_VERSION "9.3.0")
set(CMAKE_CXX_COMPILER_VERSION_INTERNAL "")
set(CMAKE_CXX_COMPILER_WRAPPER "")
set(CMAKE_CXX_STANDARD_COMPUTED_DEFAULT "14")
```

**### for the Optimizer.cc i have changed the code aforementioned but unfortunately i am still getting the same problem** 
```
69%] Linking CXX shared library ../../../Thirdparty/g2o/lib/libg2o.so
[ 69%] Built target g2o
make[2]: *** [CMakeFiles/ORB_SLAM3.dir/build.make:102: CMakeFiles/ORB_SLAM3.dir/src/LoopClosing.cc.o] Error 1
/home/beraru/ORB_SLAM3/src/Tracking.cc: In member function ‘void ORB_SLAM3::Tracking::Track()’:
/home/beraru/ORB_SLAM3/src/Tracking.cc:1155:13: warning: ‘bOK’ may be used uninitialized in this function [-Wmaybe-uninitialized]
 1155 |             if(bOK)
      |             ^~
make[2]: *** [CMakeFiles/ORB_SLAM3.dir/build.make:232: CMakeFiles/ORB_SLAM3.dir/src/Optimizer.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:390: CMakeFiles/ORB_SLAM3.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
```

jjgr3496, 2020-08-11:
I would ask you to try downgrading your compiler version to 7.5.0 and check it the error persists. Anyways I will try to upgrade my compiler in order to replicate the error!

belal-ibrahim, 2020-08-11:
`@jjgr3496  **I did downgrade the` compiler to 7.5.0** 
I got this error `/home/beraru/ORB_SLAM3/src/Optimizer.cc:2434:79: error: ‘sit’ was not declared in this scope
             if((nIDi!=pCurKF->mnId || nIDj!=pLoopKF->mnId) && pKF->GetWeight(*sit)<minFeat)
                                                                               ^~~
/home/beraru/ORB_SLAM3/src/Optimizer.cc:2434:79: note: suggested alternative: ‘sin’
             if((nIDi!=pCurKF->mnId || nIDj!=pLoopKF->mnId) && pKF->GetWeight(*sit)<minFeat)
`
`make[2]: *** [CMakeFiles/ORB_SLAM3.dir/build.make:232: CMakeFiles/ORB_SLAM3.dir/src/Optimizer.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:390: CMakeFiles/ORB_SLAM3.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
`

belal-ibrahim, 2020-08-11:
**what i have made is replace back  ** 
`for(set<KeyFrame*>::const_iterator sit=spConnections.begin(), send=spConnections.end(); sit!=send; sit++) { const long unsigned int nIDj = (*sit)->mnId;`
 
I got this output without any **explicit** error! i got this message instead 
```
make[1]: *** [CMakeFiles/Makefile2:390: CMakeFiles/ORB_SLAM3.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
```

belal-ibrahim, 2020-08-11:
**Here are my Cmake  sets** 
`et(CMAKE_CXX_COMPILER "/usr/bin/c++")
set(CMAKE_CXX_COMPILER_ARG1 "")
set(CMAKE_CXX_COMPILER_ID "GNU")
set(CMAKE_CXX_COMPILER_VERSION "7.5.0")
set(CMAKE_CXX_COMPILER_VERSION_INTERNAL "")
set(CMAKE_CXX_COMPILER_WRAPPER "")
set(CMAKE_CXX_STANDARD_COMPUTED_DEFAULT "14")
set(CMAKE_CXX_COMPILE_FEATURES "cxx_std_98;cxx_template_template_parameters;cxx_std_11;cxx_alias_templates;cxx_alignas;cxx_alignof;cxx_attributes;cxx_auto_type;cxx_constexpr;cxx_decltype;cxx_decltype_incomplete_return_types;cxx_default_function_template_args;cxx_defaulted_functions;cxx_defaulted_move_initializers;cxx_delegating_constructors;cxx_deleted_functions;cxx_enum_forward_declarations;cxx_explicit_conversions;cxx_extended_friend_declarations;cxx_extern_templates;cxx_final;cxx_func_identifier;cxx_generalized_initializers;cxx_inheriting_constructors;cxx_inline_namespaces;cxx_lambdas;cxx_local_type_template_args;cxx_long_long_type;cxx_noexcept;cxx_nonstatic_member_init;cxx_nullptr;cxx_override;cxx_range_for;cxx_raw_string_literals;cxx_reference_qualified_functions;cxx_right_angle_brackets;cxx_rvalue_references;cxx_sizeof_member;cxx_static_assert;cxx_strong_enums;cxx_thread_local;cxx_trailing_return_types;cxx_unicode_literals;cxx_uniform_initialization;cxx_unrestricted_unions;cxx_user_literals;cxx_variadic_macros;cxx_variadic_templates;cxx_std_14;cxx_aggregate_default_initializers;cxx_attribute_deprecated;cxx_binary_literals;cxx_contextual_conversions;cxx_decltype_auto;cxx_digit_separators;cxx_generic_lambdas;cxx_lambda_init_captures;cxx_relaxed_constexpr;cxx_return_type_deduction;cxx_variable_templates;cxx_std_17")
set(CMAKE_CXX98_COMPILE_FEATURES "cxx_std_98;cxx_template_template_parameters")
set(CMAKE_CXX11_COMPILE_FEATURES "cxx_std_11;cxx_alias_templates;cxx_alignas;cxx_alignof;cxx_attributes;cxx_auto_type;cxx_constexpr;cxx_decltype;cxx_decltype_incomplete_return_types;cxx_default_function_template_args;cxx_defaulted_functions;cxx_defaulted_move_initializers;cxx_delegating_constructors;cxx_deleted_functions;cxx_enum_forward_declarations;cxx_explicit_conversions;cxx_extended_friend_declarations;cxx_extern_templates;cxx_final;cxx_func_identifier;cxx_generalized_initializers;cxx_inheriting_constructors;cxx_inline_namespaces;cxx_lambdas;cxx_local_type_template_args;cxx_long_long_type;cxx_noexcept;cxx_nonstatic_member_init;cxx_nullptr;cxx_override;cxx_range_for;cxx_raw_string_literals;cxx_reference_qualified_functions;cxx_right_angle_brackets;cxx_rvalue_references;cxx_sizeof_member;cxx_static_assert;cxx_strong_enums;cxx_thread_local;cxx_trailing_return_types;cxx_unicode_literals;cxx_uniform_initialization;cxx_unrestricted_unions;cxx_user_literals;cxx_variadic_macros;cxx_variadic_templates")
set(CMAKE_CXX14_COMPILE_FEATURES "cxx_std_14;cxx_aggregate_default_initializers;cxx_attribute_deprecated;cxx_binary_literals;cxx_contextual_conversions;cxx_decltype_auto;cxx_digit_separators;cxx_generic_lambdas;cxx_lambda_init_captures;cxx_relaxed_constexpr;cxx_return_type_deduction;cxx_variable_templates")
set(CMAKE_CXX17_COMPILE_FEATURES "cxx_std_17")
set(CMAKE_CXX20_COMPILE_FEATURES "")`

jjgr3496, 2020-08-12:
> **what i have made is to replace back**
> `for(set<KeyFrame*>::const_iterator sit=spConnections.begin(), send=spConnections.end(); sit!=send; sit++) { const long unsigned int nIDj = (*sit)->mnId;`
> 
> I got this output without any **explicit** error! i got this message instead
> 
> ```
> make[1]: *** [CMakeFiles/Makefile2:390: CMakeFiles/ORB_SLAM3.dir/all] Error 2
> make: *** [Makefile:84: all] Error 2
> ```

Could you please dump all the output when you build the system to a file and upload here? You can do it as follows:

`./build.sh &> log.txt`

belal-ibrahim, 2020-08-12:
Here it's 
[log.txt](https://github.com/UZ-SLAMLab/ORB_SLAM3/files/5061895/log.txt)

jjgr3496, 2020-08-12:
It looks like you are missing libdc1394 library. Please, install it an make a fresh build of ORB-SLAM3.

belal-ibrahim, 2020-08-12:
[log.txt](https://github.com/UZ-SLAMLab/ORB_SLAM3/files/5061941/log.txt)

jjgr3496, 2020-08-12:
You are still getting the same error (here is the important line of the log: `make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/libdc1394.so', needed by '../lib/libORB_SLAM3.so'.  Stop.`)

belal-ibrahim, 2020-08-18:
[log.txt](https://github.com/UZ-SLAMLab/ORB_SLAM3/files/5090015/log.txt)
 **I have opened the directory searching for the package but as i am using ubuntu 20.04 LTS I found the files with other nomination as per image attached: **
[image]
**what i have did for, i have changed the code line to _libIlmImf.so.6_  in all files as per screenshot **
[image]

huanglilong, 2020-09-15:
fix this build error using this change (ubuntu 20.04, g++9.3.0)
[image]

belal-ibrahim, 2020-09-16:
@huanglilong **I have edited the file Loopclosing.h** as you mentioned and i got this type of errors in the file attached
[log.txt](https://github.com/UZ-SLAMLab/ORB_SLAM3/files/5229457/log.txt)

domenigoni, 2020-12-02:
i have exactly the same errors ...