UZ-SLAMLab/ORB_SLAM3 #47: Running Stereo with fisheye camera failed

ns15417, 2020-08-10:
Hi,did anyone runs the stereo camera without using IMU, i got my dataset by two cameras looking upwards and located as left and right. 
I finished to run my dataset by each single camera but failed when running stereo. The problem is listed below:
1. KeyFrames are too few and tracking always gets lost and thus go to start a new map;
2. I changed the conditions in NeedNewKeyFrame() from 
`bNeedToInsertClose = (nTrackedClose<100) && (nNonTrackedClose>70)` to
`bNeedToInsertClose = (nTrackedClose < nTotalClosedPoints*0.8) && (nNonTrackedClose>nTotalClosedPoints*0.7)`.
and nTotalClosedPoints is all the close points whose depth is between 0 and mThDepth.
After changed this line, i can get a whole map or sometimes two submaps.
But it seems that the trajectory is not in **a horizon plane** while my cameras stay at the same level in the dataset.

ns15417, 2020-08-10:
i got a problem when uploading the images, anyone interested can have a look at this link.
[https://raw.githubusercontent.com/ns15417/ORB_SLAM3/master/Result/sideview.png](url)
THanks for your time.

ccamposm, 2020-08-10:
Hi @ns15417, there shouldn't be any problem if calibration is fine. Maybe double check Tlr transformation (see https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/master/Examples/Stereo/TUM_512.yaml#L41).
Are these two cameras seeing a common region? Which is the overlap between both of them?

ns15417, 2020-08-10:
Hi @ccamposm , thanks for your reply. 
I already checked my Tlr and i think it is right in the calibration file. The real distance between my two cameras is 0.1m  in the x axis, camera coordinate. 
The common region for these two cameras are quiet big, maybe 80% because the Fov of the cameras are 180; 
 i upload the images of left and right cameras at this link:
[https://github.com/ns15417/ORB_SLAM3/tree/master/Result](url)

ccamposm, 2020-08-13:
Thank you for sharing those images. Reviewing them I can see that the number of stereo matches is quite low (60-75 points), while images have a lot of texture. If you successfully run this sequence for both cameras separately intrinsic parameters should be ok, and problem only may exist in Tlr calibration. How did you calibrate this transformation?
Checking calibration file from you fork (https://github.com/ns15417/ORB_SLAM3/blob/master/Examples/Stereo/180_stereo_kalibr.yaml) term Tlr(0,3) should be positive, since right camera is toward +x direction wrt left camera. Maybe you are providing Trl instead of Tlr. Try inverting that matrix. In addition, your Camera.bf should be more like 21.718 (fx*b)
Why is right image colored?

jdtardos, 2020-08-14:
Looking at your images, I think your ceiling has a lot of apparent contours, that are bad for SLAM. Try putting your camera in horizontal position.

closed 2020-10-05 by jdtardos