UZ-SLAMLab/ORB_SLAM3 #49: Failed to initialize monocular slam

Chatoyant19, 2020-08-11:
Hi,
When I run monocular ORB-SLAM on my camera, it always fails to be initialized successfully. I found that most of the time because the value of nsimilar is equal to 2 so that cannot reconstruct fundamental matrix.
I'm sure I have correct camera intrinsic. And besides, is there any other parameters needs to modify?
Can you tell me what should I do?
Thank you very much.

liangxiao05, 2020-08-13:
Hi,have you solved this?
I met the same problem，always shown "TRYING TO INITIALIZE"，and no keypoints extracted were shown in the image.

jdtardos, 2020-08-14:
If it's a  monocular camera, translate the camera, don't perform pure rotations.
If it still fails to initialize, please, post images of your initialization failures.

Chatoyant19, 2020-08-15:
@jdtardos 
Thank you for the answer.
yes, it's a monocular camera, but the reason for the initialization failure is not pure rotations. The trajectory path I collected is similar to kitti sequence 03.
I tried change my image encoding format from BAYER_RGGB8 to BGR8 and other format，it can be initialized Occasionally. But most of the time it's not useful.
I output some values in initialization part and found that the numbers of matched keypoints are sufficient. The reason is the ReconstructH and ReconstructF, they always return false So that it can't be initialized. Maybe I should modify which parameters to fix the problem.

jdtardos, 2020-08-15:
You could try to increase 0.7 to 0.9 in this code:
    if(nGood1>0.7*maxGood)
        nsimilar++;
    if(nGood2>0.7*maxGood)
        nsimilar++;
    if(nGood3>0.7*maxGood)
        nsimilar++;
    if(nGood4>0.7*maxGood)
        nsimilar++;
Please, let us know if this improves your initialization.

closed 2020-10-19 by jdtardos

sandeepnmenon, 2021-11-29:
I am facing the same issue for my camera. Monocular camera is not getting initialise because `ReconstructF` always returns false. I increased the ratio to 0.9, but still the same. I see the values for the nGood and maxGood I get are
```
nMinGood: 539 
maxGood: 599 nGood1: 0 nGood2: 599 nGood3: 0 nGood4 599
nMinGood: 447 - 447.30000000000001
maxGood: 497 nGood1: 0 nGood2: 497 nGood3: 0 nGood4 497
nMinGood: 447 - 447.30000000000001
maxGood: 497 nGood1: 0 nGood2: 497 nGood3: 0 nGood4 497
nMinGood: 348 - 348.30000000000001
maxGood: 387 nGood1: 387 nGood2: 0 nGood3: 386 nGood4 0
```

Also, if I comment out the nsimilar check, then the parallax checks fail. All the parallaxes returned from CheckRT are less than 0.5 and faile to meet the minParallax=1 criteria.

What could be the issue with my data.

Sample data
[image]
[image]