UZ-SLAMLab/ORB_SLAM3 #50: ORB-SLAM3, make error

xiaoyuamw, 2020-08-11:
Hi:
when i made  the ORB-SLAM3, the following errors have occurred!
Anyone woho can answer my problems? Thank you !
******
/home/xy/ORB-SLAM3/ORB_SLAM3/src/LocalMapping.cc:1455:12: warning: unused variable ‘t_inertial_only’ [-Wunused-variable]
     double t_inertial_only = std::chrono::duration_cast<std::chrono::duration<double> >(t1 - t0).count();
            ^
CMakeFiles/ORB_SLAM3.dir/build.make:110: recipe for target 'CMakeFiles/ORB_SLAM3.dir/src/LocalMapping.cc.o' failed
make[2]: *** [CMakeFiles/ORB_SLAM3.dir/src/LocalMapping.cc.o] Error 1
CMakeFiles/Makefile2:400: recipe for target 'CMakeFiles/ORB_SLAM3.dir/all' failed
make[1]: *** [CMakeFiles/ORB_SLAM3.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
******

jjgr3496, 2020-08-11:
Hi @xiaoyuamw, I can guess the error comes from the LocalMapping file however you don't provide the error messages. Could you please post the full output you are getting when building ORB-SLAM3?

Regards.

xiaoyuamw, 2020-08-11:
These are the  full output of error messages you wanted, i really hope that they are usefull for you. Thank you again!

In file included from /usr/local/include/eigen3/Eigen/src/SparseCore/SparseMatrixBase.h:152:0,
                 from /usr/local/include/eigen3/Eigen/SparseCore:37,
                 from /usr/local/include/eigen3/Eigen/Sparse:17,
                 from /home/xy/ORB-SLAM3/ORB_SLAM3/Thirdparty/g2o/g2o/solvers/linear_solver_eigen.h:30,
                 from /home/xy/ORB-SLAM3/ORB_SLAM3/include/Optimizer.h:36,
                 from /home/xy/ORB-SLAM3/ORB_SLAM3/src/LocalMapping.cc:23:
/usr/local/include/eigen3/Eigen/src/SparseCore/../plugins/MatrixCwiseUnaryOps.h: At global scope:
/usr/local/include/eigen3/Eigen/src/SparseCore/../plugins/MatrixCwiseUnaryOps.h:77:32: warning: ‘template<class _Operation> class std::binder1st’ is deprecated [-Wdeprecated-declarations]
 inline const CwiseUnaryOp<std::binder1st<std::equal_to<Scalar> >, const Derived>
                                ^
In file included from /usr/include/c++/5/bits/stl_function.h:1128:0,
                 from /usr/include/c++/5/string:48,
                 from /usr/include/c++/5/bits/locale_classes.h:40,
                 from /usr/include/c++/5/bits/ios_base.h:41,
                 from /usr/include/c++/5/ios:42,
                 from /usr/include/c++/5/ostream:38,
                 from /usr/include/c++/5/iostream:39,
                 from /home/xy/ORB-SLAM3/ORB_SLAM3/Thirdparty/DBoW2/DBoW2/BowVector.h:13,
                 from /home/xy/ORB-SLAM3/ORB_SLAM3/include/Frame.h:27,
                 from /home/xy/ORB-SLAM3/ORB_SLAM3/include/MapPoint.h:24,
                 from /home/xy/ORB-SLAM3/ORB_SLAM3/include/KeyFrame.h:23,
                 from /home/xy/ORB-SLAM3/ORB_SLAM3/include/LocalMapping.h:23,
                 from /home/xy/ORB-SLAM3/ORB_SLAM3/src/LocalMapping.cc:20:
/usr/include/c++/5/backward/binders.h:108:11: note: declared here
     class binder1st
           ^
In file included from /usr/local/include/eigen3/Eigen/src/SparseCore/SparseMatrixBase.h:152:0,
                 from /usr/local/include/eigen3/Eigen/SparseCore:37,
                 from /usr/local/include/eigen3/Eigen/Sparse:17,
                 from /home/xy/ORB-SLAM3/ORB_SLAM3/Thirdparty/g2o/g2o/solvers/linear_solver_eigen.h:30,
                 from /home/xy/ORB-SLAM3/ORB_SLAM3/include/Optimizer.h:36,
                 from /home/xy/ORB-SLAM3/ORB_SLAM3/src/LocalMapping.cc:23:
/usr/local/include/eigen3/Eigen/src/SparseCore/../plugins/MatrixCwiseUnaryOps.h: In member function ‘const Eigen::CwiseUnaryOp<std::binder1st<std::equal_to<typename Eigen::internal::traits<T>::Scalar> >, const Derived> Eigen::SparseMatrixBase<Derived>::cwiseEqual(const Scalar&) const’:
/usr/local/include/eigen3/Eigen/src/SparseCore/../plugins/MatrixCwiseUnaryOps.h:80:28: warning: ‘template<class _Operation> class std::binder1st’ is deprecated [-Wdeprecated-declarations]
   return CwiseUnaryOp<std::binder1st<std::equal_to<Scalar> >,const Derived>
                            ^
In file included from /usr/include/c++/5/bits/stl_function.h:1128:0,
                 from /usr/include/c++/5/string:48,
                 from /usr/include/c++/5/bits/locale_classes.h:40,
                 from /usr/include/c++/5/bits/ios_base.h:41,
                 from /usr/include/c++/5/ios:42,
                 from /usr/include/c++/5/ostream:38,
                 from /usr/include/c++/5/iostream:39,
                 from /home/xy/ORB-SLAM3/ORB_SLAM3/Thirdparty/DBoW2/DBoW2/BowVector.h:13,
                 from /home/xy/ORB-SLAM3/ORB_SLAM3/include/Frame.h:27,
                 from /home/xy/ORB-SLAM3/ORB_SLAM3/include/MapPoint.h:24,
                 from /home/xy/ORB-SLAM3/ORB_SLAM3/include/KeyFrame.h:23,
                 from /home/xy/ORB-SLAM3/ORB_SLAM3/include/LocalMapping.h:23,
                 from /home/xy/ORB-SLAM3/ORB_SLAM3/src/LocalMapping.cc:20:
/usr/include/c++/5/backward/binders.h:108:11: note: declared here
     class binder1st
           ^
In file included from /home/xy/ORB-SLAM3/ORB_SLAM3/src/LocalMapping.cc:20:0:
/home/xy/ORB-SLAM3/ORB_SLAM3/include/LocalMapping.h: In constructor ‘ORB_SLAM3::LocalMapping::LocalMapping(ORB_SLAM3::System*, ORB_SLAM3::Atlas*, float, bool, const string&)’:
/home/xy/ORB-SLAM3/ORB_SLAM3/include/LocalMapping.h:169:10: warning: ‘ORB_SLAM3::LocalMapping::bInitializing’ will be initialized after [-Wreorder]
     bool bInitializing;
          ^
/home/xy/ORB-SLAM3/ORB_SLAM3/include/LocalMapping.h:156:10: warning:   ‘bool ORB_SLAM3::LocalMapping::mbAbortBA’ [-Wreorder]
     bool mbAbortBA;
          ^
/home/xy/ORB-SLAM3/ORB_SLAM3/src/LocalMapping.cc:32:1: warning:   when initialized here [-Wreorder]
 LocalMapping::LocalMapping(System* pSys, Atlas *pAtlas, const float bMonocular, bool bInertial, const string &_strSeqName):
 ^
In file included from /home/xy/ORB-SLAM3/ORB_SLAM3/src/LocalMapping.cc:20:0:
/home/xy/ORB-SLAM3/ORB_SLAM3/include/LocalMapping.h:163:10: warning: ‘ORB_SLAM3::LocalMapping::mbAcceptKeyFrames’ will be initialized after [-Wreorder]
     bool mbAcceptKeyFrames;
          ^
/home/xy/ORB-SLAM3/ORB_SLAM3/include/LocalMapping.h:91:10: warning:   ‘bool ORB_SLAM3::LocalMapping::mbNewInit’ [-Wreorder]
[... 149 lines cut ...]
     while(spMergeConnectedKFs.size() < numTemporalKFs && nNumTries < nMaxTries)
                                      ^
[... cut ...]
CMakeFiles/ORB_SLAM3.dir/build.make:662: recipe for target 'CMakeFiles/ORB_SLAM3.dir/src/MLPnPsolver.cpp.o' failed
make[2]: *** [CMakeFiles/ORB_SLAM3.dir/src/MLPnPsolver.cpp.o] Error 1
CMakeFiles/ORB_SLAM3.dir/build.make:134: recipe for target 'CMakeFiles/ORB_SLAM3.dir/src/LoopClosing.cc.o' failed
make[2]: *** [CMakeFiles/ORB_SLAM3.dir/src/LoopClosing.cc.o] Error 1
CMakeFiles/Makefile2:400: recipe for target 'CMakeFiles/ORB_SLAM3.dir/all' failed
make[1]: *** [CMakeFiles/ORB_SLAM3.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
xy@hp:~/ORB-SLAM3/ORB_SLAM3/build$

jjgr3496, 2020-08-11:
Hi @xiaoyuamw, I cannot still see the error. I believe you are not copying all the ouput (maybe because your terminal removes old messages). Could you try to redirect all the output to a logfile and upload it? You can do it as it follows:

`./build.sh &> log.txt`

xiaoyuamw, 2020-08-12:
Hi @jjgr3496, thank you fou your answer. I have made again by your method, this time you can see the full uotput of error messages. Thank you again!
[log.txt](https://github.com/UZ-SLAMLab/ORB_SLAM3/files/5060050/log.txt)

jjgr3496, 2020-08-12:
Hi @xiaoyuamw, thanks for updating the log. I have finally been able to spot the errors:

- At `Map.h`, in line 140, you should put the name space before list and string (the line should end up looking like this: `void printReprojectionError(std::list<KeyFrame*> &lpLocalWindowKFs, KeyFrame* mpCurrentKF, std::string &name, std::string &name_folder);`). This error seems to pop up depending of the version of your compiler.

- At `MLPnPsolver.cpp`, we call the method `Eigen::SparseMatrix<double>.setIdentity()`, however your compiler seems like cannot find it. Please, check your Eigen version (remember we require a version grater than 3.1.1).

- Finally, the last error looks like a problem when finding the file openssl/md5.h. You should first install in your system the openssl library (https://cloudwafer.com/blog/installing-openssl-on-ubuntu-16-04-18-04/). 

These are all the errors I can find from your log. If you keep finding some issues building, paste me again the log!

Regards.

xiaoyuamw, 2020-08-13:
Hi @jjgr3496, i have met such an error this time, maybe you can help me to solve it. Thank you again!

[ 71%] Linking CXX shared library ../lib/libORB_SLAM3.so
/usr/bin/ld: 找不到 -lboost_serialization
collect2: error: ld returned 1 exit status
CMakeFiles/ORB_SLAM3.dir/build.make:805: recipe for target '../lib/libORB_SLAM3.so' failed
make[2]: *** [../lib/libORB_SLAM3.so] Error 1
CMakeFiles/Makefile2:400: recipe for target 'CMakeFiles/ORB_SLAM3.dir/all' failed
make[1]: *** [CMakeFiles/ORB_SLAM3.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

jjgr3496, 2020-08-13:
Hi @xiaoyuamw, could you please upload  again the build log file with the new error?

xiaoyuamw, 2020-08-13:
Hi @jjgr3496, i have uploaded the full output of the build log file, maybe you can find out where the errors happened. Thank you again!

jjgr3496, 2020-08-13:
Hi again @xiaoyuamw, after reading carefully the log file, I finally found the error (I had to translate from Chinese to English to discover the error :P). It seems that you don have installed the `lboost_serialization` so the compilation fails. Please, try to install that library and compile again.

xiaoyuamw, 2020-08-14:
Hi @jjgr3496, i have run the ORB-SLAM3 successfully, and i will  take a further study of this greatful project made by you and your team, thanks very much again!

closed 2020-08-14 by ccamposm

Kanishk598, 2021-06-27:
@jjgr3496 Hello Juan, I admire your work in SLAM and have used your ORB-SLAM3 package in my research earlier.
Earlier before when I compiled the repository, I solved the small errors myself, but I don't know how to solve errors this time. Can you please check my log file and help me identify what can be done?
Thanks
[This](https://github.com/Kanishk598/ORB_SLAM3/blob/kanishk/error_log.txt) is the log file of `./build.sh`

Kanishk598, 2021-06-29:
Update:
I successfully compiled the project. The error was coming from a broken OpenSSL version that I installed apart from the version that comes with Ubuntu 18.04. 
Thank you

rickkcir, 2022-03-10:
> Hi again @xiaoyuamw, after reading carefully the log file, I finally found the error (I had to translate from Chinese to English to discover the error :P). It seems that you don have installed the [...]

Thank you. I meet the same issue. Through installed the boost, I successfully compiled the project. How to installed the boost? You could watch my [blog](https://blog.csdn.net/xiaoren886/article/details/123413342?spm=1001.2014.3001.5501).