UZ-SLAMLab/ORB_SLAM3 #51: Running ORB_SLAM3 on gazebo failed

hustchenli, 2020-08-11:
I'm trying to run ORB_SLAM3 on gazebo. The program can run and show the figure that taken by virtual camera. But it can not extract feature points and estimate camera pose.
[image]
Here is the information of rostopic /camera/rgb/camera_info:
[image]
And yaml file:
```
%YAML:1.0

#--------------------------------------------------------------------------------------------
# Camera Parameters. Adjust them!
#--------------------------------------------------------------------------------------------
Camera.type: "PinHole"
# Camera calibration and distortion parameters (OpenCV) 
Camera.fx: 277.19135641132203
Camera.fy: 277.19135641132203
Camera.cx: 160.5
Camera.cy: 120.5

Camera.k1: 0
Camera.k2: 0
Camera.p1: 0
Camera.p2: 0

Camera.width: 320 
Camera.height: 240 

# Camera frames per second 
Camera.fps: 30.0

# IR projector baseline times fx (aprox.)
Camera.bf: 13.8595

# Color order of the images (0: BGR, 1: RGB. It is ignored if images are grayscale)
Camera.RGB: 1

# Close/Far threshold. Baseline times.
ThDepth: 50.0

# Deptmap values factor 
DepthMapFactor: 1000.0

#--------------------------------------------------------------------------------------------
# ORB Parameters
#--------------------------------------------------------------------------------------------

# ORB Extractor: Number of features per image
ORBextractor.nFeatures: 1000

# ORB Extractor: Scale factor between levels in the scale pyramid 	
ORBextractor.scaleFactor: 1.2

# ORB Extractor: Number of levels in the scale pyramid	
ORBextractor.nLevels: 8

# ORB Extractor: Fast threshold
# Image is divided in a grid. At each cell FAST are extracted imposing a minimum response.
# Firstly we impose iniThFAST. If no corners are detected we impose a lower value minThFAST
# You can lower these values if your images have low contrast			
ORBextractor.iniThFAST: 20
ORBextractor.minThFAST: 7

#--------------------------------------------------------------------------------------------
# Viewer Parameters
#--------------------------------------------------------------------------------------------
Viewer.KeyFrameSize: 0.05
Viewer.KeyFrameLineWidth: 1
Viewer.GraphLineWidth: 0.9
Viewer.PointSize:2
Viewer.CameraSize: 0.08
Viewer.CameraLineWidth: 3
Viewer.ViewpointX: 0
Viewer.ViewpointY: -0.7
Viewer.ViewpointZ: -1.8
Viewer.ViewpointF: 500
```
And I got this: 
`Framebuffer with requested attributes not available. Using available framebuffer. You may see visual artifacts.`
[image]
Does anyone have any idea? Thanks

jjgr3496, 2020-08-14:
Hi @hustchenli, could you please share with us some image example of your images? If your all your images are like the one showing in your image, it is pretty impossible to extract ORB features.

hustchenli, 2020-08-14:
@jjgr3496 Hi, here are some more examples of my images: 
[image]
[image]
My gazebo model is just like this: 
[image]

jdtardos, 2020-08-14:
ORB-SLAM3 is a feature-based SLAM system and needs texture in the environment to find features. It cannot work in an empty environment with plain walls and floor. Add textured objects to tour simulation.

closed 2020-08-14 by jdtardos

hustchenli, 2020-08-14:
> ORB-SLAM3 is a feature-based SLAM system and needs texture in the environment to find features. It cannot work in an empty environment with plain walls and floor. Add textured objects to tour simulation.

Thanks for your explanation.