UZ-SLAMLab/ORB_SLAM3 #54: Failed to load image at: /Datasets/MH01/......Segmentation fault (core dumped)

squaresoft2015, 2020-08-11:
I tried to run ./euroc_examples.sh in the root folder and set the path to the dataset actually uncompressed. But the program output a error message:
```
Failed to load image at: /Datasets/MH01/mav0/cam0/data/1403636579763555584.png
./euroc_examples.sh: line 7: 12029 Segmentation fault      (core dumped) ./Examples/Monocular/mono_euroc ./Vocabulary/ORBvoc.txt ./Examples/Monocular/EuRoC.yaml "$pathDatasetEuroc"/MH01 ./Examples/Monocular/EuRoC_TimeStamps/MH01.txt dataset-MH01_mono
Launching MH02 with Monocular sensor
num_seq = 1
```
The file name contained in the timestamp file does not match with the real file name in the corresponding dataset folder. Is this the real reason for such error ?

Any help will be appreciate.

jjgr3496, 2020-08-12:
Hi @squaresoft2015, yeah, it looks like either the image is missing or have another name. Please check you are providing the correct path to the dataset. If the error persist, you can try to download again the dataset (it may be that the first time you downloaded the dataset the file was corrupted during the download).

SongWang2017, 2020-08-12:
Hi @squaresoft2015 When I unzip the ASL files, e.g. MH_01_easy.zip, by default it's unzipped into a folder named 'mav0'.
then I append the unzip command with a '-d MH01' flag. and assign 'pathDatasetEuroc' in the  example.sh as the parent folder of MH01. Problem solved.

squaresoft2015, 2020-08-13:
@SongWang2017 @jjgr3496  Thank you so much for solving the problem. Now the program start and run smoothly. But when I execute ./euroc_eval_example.sh, it prompts the following error message:
```
Evaluation of Vicon Room 2 trajectory with Stereo sensor
  File "evaluation/evaluate_ate_scale.py", line 81
    dots += numpy.dot(data_zerocentered[:,column].transpose(),rotmodel[:,column])
                                                                                ^
TabError: inconsistent use of tabs and spaces in indentation
```
Does anyone know the reason? I think it is related to ATE computation. Thanks!

jjgr3496, 2020-08-13:
It is a simple error in one of the python scripts. Please, check this pull request to solve it: #59

closed 2020-08-13 by jdtardos

iiiicce8010, 2021-09-05:
> @SongWang2017 @jjgr3496 Thank you so much for solving the problem. Now the program start and run smoothly. But when I execute ./euroc_eval_example.sh, it prompts the following error message:
>
> ``` [...]

so could you tell me how to solve this core dump? I met the same problem, download again?