UZ-SLAMLab/ORB_SLAM3 #57: Can not correct the loop?

liangsongLee, 2020-08-13:
Hi, thanks for your great work!
When I test orbslam3 using my own fisheye camera in MONOCULOR mode, the  system can run successfully. But it can not correct the loop. In other words, it often fails. However  the system is not broken, continue to execute. I guess the system does not detect the loop, or can not correct the loop. What should I do?

ccamposm, 2020-08-13:
Hi @liangsongLee.
If both candidate images for loop-closing have a similar point of view (ORB descriptors are likely to be similar) and there exist enough map points, there shouldn't be any problem. Is that the case?

liangsongLee, 2020-08-13:
@ccamposm Hi, I find the problem is here: [https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/adc40628f01682b2fe4d143270daa4fe6d4ada8d/src/LoopClosing.cc#L687](url), bConverge is always false, so it can not do the following process in loop closing. The reason for this is here: [https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/adc40628f01682b2fe4d143270daa4fe6d4ada8d/src/Sim3Solver.cc#L277](url) `mnInliersi>mRansacMinInliers`is false. so bConverge is false. However, when I use pinhole camera, this situation will not happen.Is it caused by the fisheye camera?

liangsongLee, 2020-08-13:
@ccamposm Hi, I find the problem is here: https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/adc40628f01682b2fe4d143270daa4fe6d4ada8d/src/LoopClosing.cc#L687, bConverge is always false, so it can not do the following process in loop closing. The reason for this is here: https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/adc40628f01682b2fe4d143270daa4fe6d4ada8d/src/Sim3Solver.cc#L277 mnInliersi>mRansacMinInliersis false. so bConverge is false. However, when I use pinhole camera, this situation will not happen.Is it caused by the fisheye camera?

liangsongLee, 2020-08-13:
@ccamposm Hi, I find the problem is here:
 https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/adc40628f01682b2fe4d143270daa4fe6d4ada8d/src/LoopClosing.cc#L687
 bConverge is always false, so it can not do the following process in loop closing. 
The reason for this is here: 
https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/adc40628f01682b2fe4d143270daa4fe6d4ada8d/src/Sim3Solver.cc#L277 
`mnInliersi>mRansacMinInliers`is false. so bConverge is false. However, when I use pinhole camera, this situation will not happen.Is it caused by the fisheye camera?

ccamposm, 2020-08-14:
You should check your calibration for fisheye, since loopclosing is similarly performed for any camera model.

closed 2020-08-14 by ccamposm

ouyangandy, 2020-08-21:
@liangsongLee 请问您解决了这个问题吗？我使用廉价的单目鱼眼相机，进行标定后也是这个结果，始终无法检测到闭环，看到了您的分析，但不知道该如何处理。

liangsongLee, 2020-08-21:
> You should check your calibration for fisheye, since loopclosing is similarly performed for any camera model. Reopen if problem persists.

Hi, I used TUM VI dataset with loops, using their published calibration params of fisheye, but it can not detect the loop either. So I guess it is not the problem of my calibration.

liangsongLee, 2020-08-21:
> You should check your calibration for fisheye, since loopclosing is similarly performed for any camera model. Reopen if problem persists.

Hi, I tested TUM VI open datasets with loops, using their published calibration params of fisheye, but it can not detect the loop either. So I guess it is not the problem of my calibration.

liangsongLee, 2020-08-21:
> @liangsongLee 请问您解决了这个问题吗？我使用廉价的单目鱼眼相机，进行标定后也是这个结果，始终无法检测到闭环，看到了您的分析，但不知道该如何处理。

没有解决，我试了tum鱼眼公开数据集，用了他们公开的鱼眼标定参数，也不能闭环，针孔相机有时候也不能闭环。