UZ-SLAMLab/ORB_SLAM3 #58: "terminate called after throwing an instance of 'cv::Exception'" occurs after "BoW: xxx matches between xxx points with coarse Sim3"

qliao001, 2020-08-13:
I have tried the Monocular version with customized streaming sequences, which worked well with ORB-SLAM2. However, with ORB-SLAM3, for some sequences, whenever "BoW: xxx matches between xxx points with coarse Sim3" occurs, the system throws a cv exception as below:

> BoW: 276 matches between 635 points with coarse Sim3
terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(4.4.0-pre) /home/qi/opencv_build/opencv/modules/imgproc/src/color.cpp:182: error: (-215:Assertion failed) !_src.empty() in function 'cvtColor'
Aborted (core dumped)

I can confirm that the file exists and is not corrupted. By printing the loaded image, we can see that the exception may happen at different frame ids of the same sequence and the image was load successfully in mono_euroc.cc. I assume it might be a cvtColor exception in LoopClosing.cc? Please find in below **two error printings** when running the **same** sequence:

[image]

[image]

ccamposm, 2020-08-13:
Hi @qliao001, are you working with the latest version of ORB-SLAM3? This error should have been solved

qliao001, 2020-08-13:
> Hi @qliao001, are you working with the latest version of ORB-SLAM3? This error should have been solved

Right! updated to the latest version, problem solved!! Thank you so much!!

closed 2020-08-13 by qliao001