UZ-SLAMLab/ORB_SLAM3 #63: Changing brief descriptor from 256 bits to 128 bits, faced problem

XinyuWang2, 2020-08-16:
I am trying to let ORBSLAM3 use descriptor of 128 bits rather than 256 bits hopping to save some time. 

ORBextractor.cc:
for (int i = 0; i < 16; ++i, pattern += 16) // edited ori: 32
static int bit_pattern_31_[128*4] = // edited, ori: 256
const int npoints = 256; // edited, ori:512
descriptors = Mat::zeros((int)keypoints.size(), 16, CV_8UC1); // edited, ori:32
_descriptors.create(nkeypoints, 16, CV_8U); // edited, ori:32
Mat desc = cv::Mat(nkeypointsLevel, 16, CV_8U); // edited: ori:32

ORBmatcher.cc
int bestDist=128; // edited, ori:256
for(int i=0; i<4; i++, pa++, pb++)//edited, ori:i<8
const int ORBmatcher::TH_HIGH = 60;
const int ORBmatcher::TH_LOW = 30;

Then I get this 
[image]

Features points showing in the frame indicates tracking is still good. But it seems that ORBSLAM3 can get to know the rotation of the camera well but can't get the right translation (thus the camera always stays at the same place.) I don't know where else I need to change also. Thanks.

jdtardos, 2020-08-16:
It's not that easy, you will need also to modify the library Thirdparty/DBoW2/DBoW2/  and the vocabulary provided with ORB-SLAM3, because they use 256 bites (32 bytes).  
It's difficult to tell how much a shorter descriptor will degrade matching performance. Good luck.

closed 2020-08-16 by jdtardos

XinyuWang2, 2020-08-16:
Oh, Thanks a lot! I will try that.