UZ-SLAMLab/ORB_SLAM3 #65: Trajectory alignment problem

ericzzj1989, 2020-08-17:
First, thanks for your excellent work.

Now I use EuRoc Monocular-Inertial dataset to generate the frame trajectory (Twb) and evaluate this trajectory based on the pipeline (rpg_trajectory_evaluation: https://github.com/uzh-rpg/rpg_trajectory_evaluation). However, the trajectory is not aligned with the groundtruth data provided by EuRoc. Here are the trajectory results:

[image]
[image]

On the other hand, using your evaluation tool provided by this repo is correctly aligned.
[image]

Is there any extra coordinate transformation when using rpg_trajectory_evaluation?

Many thanks.

richard-elvira, 2020-08-17:
Could you give us the reported results in the file absolute_error_stadistics in traj_est folder?

ericzzj1989, 2020-08-17:
@richard-elvira Many thanks for your kind reply.
No problem. Here are the results in the file absolute_err_statistics in traj_est folder:
rot:
  max: 125.46642048627574
  mean: 123.0160995676472
  median: 123.21334015410343
  min: 121.72575838724455
  num_samples: 3632
  rmse: 123.01690802845008
  std: 0.44599107937585536
scale:
  max: 66.85431138114315
  mean: 13.60133231800874
  median: 8.158057887538861
  min: 0.006628639468053699
  num_samples: 3632
  rmse: 19.444485401028647
  std: 13.895746532155355
trans:
  max: 7.822996243370112
  mean: 3.981793528698914
  median: 4.279504419529851
  min: 1.2279128190801212
  num_samples: 3632
  rmse: 4.242375020916088
  std: 1.4639214845421802

PS: the align_type is posyaw.

PPS: I think that taking gravity magnitude and direction (aligned with Z axis) into account from the Inertial-only MAP estimation beginning maybe cause this phenomenon since the gravity direction maybe no longer aligned with the Z axis as the optimization progresses. On the other hand, the results of VINS-Mono are correctly aligned with groundtruth using rpg_trajectory_evaluation tool since the gravity direction is aligned with Z axis in the end in VINS-Mono's initialization pipeline. 

Many thanks.

ericzzj1989, 2020-08-18:
@richard-elvira Could you give me some advice about which coordinate system this work entirely follow? NED or ENU?

jdtardos, 2020-10-05:
There is ALWAYS an aligning transformation between a SLAM system and a ground truth system. In ORB_SLAM3, the map is built with respect to the reference the first camera pose, z axis pointing forward. The ground truth in Euroc is provided with respect to a world reference that has z pointing up. The evaluation scripts find the optimal alineation.

closed 2020-10-05 by jdtardos