UZ-SLAMLab/ORB_SLAM3 #66: real-time running issue

wdxybhb, 2020-08-17:
Hey guys,

I've changed the code in stereo_inertial_euroc.cc to make the code running for real-time on my current MYNT D stereo camera. But I found the running speed on my laptop is pretty slow say 2-4 fps and I can tell the obvious latency between 2 continuous frames. And once I move more aggressively, the code would lose tracking and generate following info

Fail to track local map!
IMU is not or recently initialized. Reseting active map...
LM: Active map reset recieved
LM: Active map reset, waiting...
FAIL LOCAL-INERTIAL BA!!!!
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 594
mnInitialFrameId = 8503
0 Frames set to lost
not IMU meas
not enough acceleration
not enough acceleration
First KF:496; Map init KF:483
New Map created with 738 points

wdxybhb, 2020-08-17:
I can't figure out where is wrong with my usage. I am pretty sure I had a good calibration for the stereo camera and IMU as well, I also switched to Stereo mode and that was slow too!

jjgr3496, 2020-08-17:
Could you please share us you pc specifications and image resolution? From our test we know that ORB-SLAM3 works at 20 fps for images with a resolution of 512x512 in a modern desktop computer,

wdxybhb, 2020-08-17:
Yeah sure. Actually my PC would run VINS-FUSION normally with loop-closure and other functions.
Spec should be as follows:
Intel® Core™ i7-6700HQ CPU @ 2.60GHz × 8 
GeForce GTX 960M/PCIe/SSE2
8GB RAM
And the resolution is 640*480, I also downsize the frame rate from 30 to 15 hz.
So I suspect there must be something wrong with my usage as per the speed I run the datesets from ASL which worked normally.
Would you please help me to figure out what configuration could violate the code running or which part goes wrong with the frame tracking as it tends to lose tracking once the movement went faster.

Do appreciate for your reply.

jjgr3496, 2020-08-18:
Hi @wdxybhb, how is the format of your dataset? Are you using ros? I am trying to figure out what can be causing the issue. You could also try to measure the execution time of the main methods of the library in order to track which piece of code is consuming most part of time.

wdxybhb, 2020-08-19:
@jjgr3496 Hi mate, thank you so much for your reply!

Actually, I am using a mynt stereo camera directly as my real-time data source, and I did not ros but implement realtime running based on C++ via changing stereo_inertial_euroc.cc directly.
I will add more timestamps in the code to monitor which is the key issue.
Do appreciate your help and hope you guys stay well!

msadowski, 2020-09-23:
I think I'm observing a similar issue running Stereo_Inertial node with ROS and using Realsense D435i in real time. If the camera is not moving I get "not enough acceleration" printed a lot and keyframes are lost. Here is a gif: https://imgur.com/a/LYQlRcb

leo-wkd, 2020-10-21:
> I think I'm observing a similar issue running Stereo_Inertial node with ROS and using Realsense D435i in real time. If the camera is not moving I get "not enough acceleration" printed a lot and keyframes are lost. Here is a gif: https://imgur.com/a/LYQlRcb

same camera, same problem

zyl1234, 2020-12-17:
> @jjgr3496 Hi mate, thank you so much for your reply!
>
> Actually, I am using a mynt stereo camera directly as my real-time data source, and I did not ros but implement realtime running based on C++ [...]

 hello,did you solve this problem?

wzxzibo, 2021-01-17:
> I think I'm observing a similar issue running Stereo_Inertial node with ROS and using Realsense D435i in real time. If the camera is not moving I get "not enough acceleration" printed a lot and keyframes are lost. Here is a gif: https://imgur.com/a/LYQlRcb

hello, I think I have the same problem with you , did you find the solution?

yhy130531, 2021-01-22:
I also encountered the same problem with the T265 camera. This is caused by the inconsistency between the IMU's accelerometer update frequency and the gyroscope update frequency. My accelerometer update frequency is 62 Hz, and the gyroscope update frequency is 200 Hz. Currently, do you have a solution?

yuzhou42, 2021-06-08:
> I think I'm observing a similar issue running Stereo_Inertial node with ROS and using Realsense D435i in real time. If the camera is not moving I get "not enough acceleration" printed a lot and keyframes are lost. Here is a gif: https://imgur.com/a/LYQlRcb

Same issue with ZED2. Have you solved it?

RoZhong, 2021-12-01:
> > 我想我正在观察一个类似的问题，即使用 ROS 运行 Stereo_Inertial 节点并实时使用 Realsense D435i。如果相机没有移动，我会打印出很多“加速度不足”并且关键帧丢失。这是一个 gif：[https](https://imgur.com/a/LYQlRcb) : [//imgur.com/a/LYQlRcb](https://imgur.com/a/LYQlRcb)
> 
> ZED2 也有同样的问题。你解决了吗？

Have you solved this problem? Now I want to run with zed2, but the following problem will appear
[image]
.