UZ-SLAMLab/ORB_SLAM3 #67: [Mono-Inertial] ERROR: Frame with a timestamp older than previous frame detected!

surfii3z, 2020-08-17:
Hi,

thank you for sharing such a wonderful work.

I have been trying to use the Mono_Inertial node, and found out that it always lost the track with this complain

> ERROR: Frame with a timestamp older than previous frame detected!

Could you suggest what could be the potential reasons?

_________________________________________________________-

The calibration file is as the following

YAML:1.0

#--------------------------------------------------------------------------------------------
# Camera Parameters. Adjust them!
#--------------------------------------------------------------------------------------------
Camera.type: "PinHole"
# Camera calibration and distortion parameters (OpenCV) 
Camera.fx: 929.562627
Camera.fy: 928.604856
Camera.cx: 487.474037
Camera.cy: 363.165223

Camera.k1: -0.016272
Camera.k2: 0.093492
Camera.p1: 0.000093
Camera.p2: 0.002999
Camera.k3: 0.000000
    
# Camera resolution
Camera.width: 960
Camera.height: 720

# Camera frames per second 
Camera.fps: 30.0

# Color order of the images (0: BGR, 1: RGB. It is ignored if images are grayscale)
Camera.RGB: 1

# Transformation from body-frame (imu) to camera
Tbc: !!opencv-matrix
   rows: 4
   cols: 4
   dt: f
   data: [0.000624762189693373, 0.9987476589604951, 0.05002722651855723, -0.00027498954959967003,
          -0.20874315768809598, -0.048794909164563804, 0.9767524512167998, 0.00021872887165989068,
          0.977970298010431, -0.011053079234071894, 0.20845125485071164, -8.459932530849928e-05,
          0.0, 0.0, 0.0, 1.0]
### 

IMU.NoiseGyro: 0.000187 # rad/s^0.5 
IMU.NoiseAcc: 0.00186 # m/s^1.5
IMU.GyroWalk: 2.66e-05 # rad/s^1.5
IMU.AccWalk: 0.000433 # m/s^2.5
IMU.Frequency: 15

jjgr3496, 2020-08-17:
It looks like a problem with the order you are getting the images. That error occurs when your current image has a timestamp older than another image that has already been processed by ORB-SLAM. Please check how you capture your images and their respective timestamps.

surfii3z, 2020-08-17:
> It looks like a problem with the order you are getting the images. That error occurs when your current image has a timestamp older than another image that has already been processed by ORB-SLAM. Please check how you capture your images and their respective timestamps.

Thank you for your promptly reply. I will check that and comeback with the feedback.

However though, the exact same system works fine with Mono (without Inertial).

surfii3z, 2020-08-17:
@jjgr3496 So I tried to print out 
`mLastFrame.mTimeStamp
 mCurrentFrame.mTimeStamp` right in the [if statement](https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/8ac600afe0c7f3f8a916d8cc0e16c7bda36027e5/src/Tracking.cc#L868) that check the condition.

It seems like mLastFrame.mTimeStamp complies with the timestamp in the ROS topic (1.59767e+09). However, mCurrentFrame.mTimeStamp shows 0 most of the time or some random value like 0.736123, 0.969909 occasionally.

jjgr3496, 2020-08-18:
@surfii3z  I've been trying to reproduce your error running ORB-SLAM3 ros mono-inertial with EuRoC dataset sequences and I am not able to reproduce you error. Everything seems to work fine.

surfii3z, 2020-08-18:
@jjgr3496  Thank you for your feedback. I have no error with Realsense D435i too. I think it's my system problem. I will close this issue.

closed 2020-08-18 by surfii3z

snowface3, 2020-10-04:
I am seeing the same error when playing back from a ROSBAG

Input sensor was set to: Monocular-Inertial

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name: 

Camera Parameters: 
- Camera: Pinhole
- fx: 608.349
- fy: 639.635
- cx: 613.776
- cy: 361.71
- k1: -0.211149
- k2: 0.0352097
- p1: 5.79281e-05
- p2: -0.000810598
- fps: 20
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 1000
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

Left camera to Imu Transform (Tbc): 
[0.70710701, 0, -0.70710701, -0.26163;
 0, 1, 0, -0.039999999;
 0.70710701, 0, 0.70710701, -0.0919239;
 0, 0, 0, 1]

IMU frequency: 200 Hz
IMU gyro noise: 0.00017 rad/s/sqrt(Hz)
IMU gyro walk: 1.9393e-05 rad/s^2/sqrt(Hz)
IMU accelerometer noise: 0.002 m/s^2/sqrt(Hz)
IMU accelerometer walk: 0.003 m/s^3/sqrt(Hz)
Gtk-Message: 02:43:39.246: Failed to load module "canberra-gtk-module"
Gtk-Message: 02:43:39.247: Failed to load module "canberra-gtk-module"
ERROR: Frame with a timestamp older than previous frame detected!
Creation of new map with id: 1
Exits current map 
Saved map with ID: 0
Creation of new map with last KF id: 0
ERROR: Frame with a timestamp older than previous frame detected!
Creation of new map with id: 2
Exits current map 
Saved map with ID: 1
Creation of new map with last KF id: 0
ERROR: Frame with a timestamp older than previous frame detected!
Creation of new map with id: 3
Exits current map 
Saved map with ID: 2
Creation of new map with last KF id: 0
ERROR: Frame with a timestamp older than previous frame detected!
Creation of new map with id: 4
Exits current map 
Saved map with ID: 3
Creation of new map with last KF id: 0
id last: 51    id curr: 52
Timestamp jump detected, before IMU initialization. Reseting...
SYSTEM-> Reseting active map in monocular case
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 0
mnInitialFrameId = 50
0 Frames set to lost
ERROR: Frame with a timestamp older than previous frame detected!
Creation of new map with id: 5
Exits current map 
Saved map with ID: 4
Creation of new map with last KF id: 0
id last: 57    id curr: 58
Timestamp jump detected, before IMU initialization. Reseting...
SYSTEM-> Reseting active map in monocular case
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 0
mnInitialFrameId = 57
0 Frames set to lost
ERROR: Frame with a timestamp older than previous frame detected!
Creation of new map with id: 6
Exits current map 
Saved map with ID: 5
Creation of new map with last KF id: 0
ERROR: Frame with a timestamp older than previous frame detected!
Creation of new map with id: 7
Exits current map 
Saved map with ID: 6
Creation of new map with last KF id: 0
ERROR: Frame with a timestamp older than previous frame detected!
Creation of new map with id: 8
Exits current map 
Saved map with ID: 7
Creation of new map with last KF id: 0
^Cterminate called without an active exception

zthcool, 2020-10-16:
I got the same problem,Have you solved it?

snowface3, 2020-10-16:
nope

zthcool, 2020-10-17:
> nope

Is it the problem of receiving frequency between image and IMU?

rdehaggart1, 2020-12-04:
> I am seeing the same error when playing back from a ROSBAG
>
> Input sensor was set to: Monocular-Inertial
>
> Loading ORB Vocabulary. This could take a while...
> Vocabulary loaded!
>
> Creation of new map [...]

I'm having the same issue with (pretty much) identical output as listed here... I'm running a .bag file that has been processed fine by some other VI-SLAM algorithms. I have checked the frame timestamping and all seems to be in order... If anyone has found the resolution to this it would be much appreciated!

KopanevPavel, 2021-02-18:
Hello) Having a similar error on KAIST dataset (using ROS). Data itself is Ok

mosimp, 2021-02-18:
I see this error appear when my IMU node is shut down, while trying to run Mono-Inertial.  
If my IMU and camera are active, the error disappears (though I get a 'IMU is not or recently initialized.  Reseting active app...' in its place).

Luke-Byrne-uni, 2021-02-22:
I'm also getting this error, when running the mono inertial node, using images and imu data published from a robot in gazebo simulator. Not sure what exactly the problem is.

spk48, 2021-03-02:
I'm also getting this error, then I try ORB-SLAM2, and I get no error.

NickPerezCarletonUniversity, 2021-09-29:
@snowface3 I noticed you have an error with canberra-gtk-module. The highest voted answer in this thread solved the error:

https://askubuntu.com/questions/342202/failed-to-load-module-canberra-gtk-module-but-already-installed