UZ-SLAMLab/ORB_SLAM3 #69: Looping "IMU is not or recently initialized. Reseting active map... "

qliao001, 2020-08-18:
Hi many thanks for the great contribution! I have been working with monocular and multi-session version and all went well. Now I try to introduce the corresponding IMU sequence, but the system kept looping "IMU is not or recently initialized. Reseting active map... ". Could you please give some advice how can I tackle the problem? or is there any specific trick of collecting data for better IMU initialization?

Here is a screenshot, you can see that the system was basically keep trying to initialize the IMU:

> IMU in Map 0 is initialized
PIOLKF: NOT ENOUGH EDGES
Fail to track local map!
IMU is not or recently initialized. Reseting active map...
SYSTEM-> Reseting active map in monocular case

[image]

Thanks a lot!

yike-chen, 2020-09-01:
I have the same problem. Can anyone explain it?

yike-chen, 2020-09-04:
The reason may be that [ORB-SLAM3](https://github.com/UZ-SLAMLab/ORB_SLAM3/issues/94) only support Global Shutter cameras.

yike-chen, 2020-09-04:
What is the camera type？

qliao001, 2020-09-08:
> What is the camera type？

It is an Android phone embedded camera and IMU. The streaming and IMU measurements are sent by client (cell phone) and processed in the server (laptop). But I don't think it is the problem of camera, since I could run the Monocular version, but not the Monocular-Inertial version.

npcdna, 2020-10-15:
yes,i meet same question. I use camera-imu parameters which calibrated by kalibr, these parameters can work well in vins, but always reset in orbslam3. when i change imu noise parameters, orbslam3 always arise segment error.

mccoychang, 2020-11-02:
I also encountered this problem, is there a solution

ShreyasManjunath, 2020-11-11:
I find similar error like - 

> New Map created with 285 points
> Point distribution in KeyFrame: left-> 285 --- right-> 0
> build optimization graph
> start optimization
> end optimization
> update Keyframes velocities and biases
> IMU in Map 2 is initialized
> Fail to track local map!
> IMU is not or recently initialized. Reseting active map...
> SYSTEM-> Reseting active map in monocular case
> LM: Active map reset recieved
> LM: Active map reset, waiting...
> FAIL LOCAL-INERTIAL BA!!!!
> LM: Reseting current map in Local Mapping...
> LM: End reseting Local Mapping...
> LM: Reset free the mutex
> LM: Active map reset, Done!!!

This loops continously. Any one found solution?

Ehsanmb, 2020-11-26:
I have the same issue! looking for solution

MiaoMiaoMeng, 2021-01-27:
> It is an Android phone embedded camera and IMU. The streaming and IMU measurements are sent by client (cell phone) and processed in the server (laptop). But I don't think it is the problem of [...]

Hi, @qliao001, 
I met the same question, did you figure it out?
I did a lot of work on camera-IMU calibrate, but it doesn't work.
So I try to find the local-Inertial BA to change the threshold, but I didn't finish it yet.
Do you have some idea?

MiaoMiaoMeng, 2021-02-02:
Hi guys, @Ehsanmb @ShreyasManjunath @mccoychang 

Today I find out how to fix this problem.
If you use those APP in Android (CameraImu & MarsLogger) or other application pls make sure the coordinate is same.
For me I use MarsLogger to catch info form Cam and IMU and creat bag for calibrate, and then use CameraImu to do the real time orbslam.
The reason why my local-Inertial BA function was failed is that those IMU‘s coordinate is totally different. 

So in this situation, you can rebuild your application to let them have a unified coordinate system.

pinkfloydfan, 2021-04-14:
I've been encountering this, and believe it or not - I've 'mitigated' it by changing the noise/random walk values in the yaml file to ridiculously small numbers, like 10e-8 small. I have no idea why this works.

iamchoking, 2021-07-08:
I have tried the suggestion of both @MiaoMiaoMeng  and @pinkfloydfan (along with everything else on the web). But neither seem to work. I am currently using:

ZED2 (@15fps publish rate, 1280x720 resolution)
ROS `melodic`, real-time
`opencv-3-2-0`

topics: (published with ZED ros wrapper [zed_wrapper])
`/zed2/zed_node/imu/data_raw`
`/zed2/zed_node/left/image_rect_color`
`/zed2/zed_node/left/image_rect_color`

 with imu settings calibrated from kalibr. I hope a solution to this is found soon!

*This is my first post ever on any Issue ever (i am this determined for an answer!). So if you need more info, please tell me

pinkfloydfan, 2021-07-08:
> I have tried the suggestion of both @MiaoMiaoMeng and @pinkfloydfan (along with everything else on the web). But neither seem to work. I am currently using:
>
> ZED2 (@15fps publish rate, 1280x720 [...]

@iamchoking sanity check: are the image frames from the camera timestamped? I was able to get ORB SLAM 3 working on an ESP32-CAM at about 12FPS, and an IMU from a drone flight controller. My workaround was setting the variances to extremely low values. DM me your configuration yaml file and I can have a look.

iamchoking, 2021-07-12:
@pinkfloydfan

bboysil, 2021-09-27:
Hi, I've been struggling with this error for a fewdays now. Did anyone find a solution?
I was providing camera stream and IMU data from an Android, then from an iPhone.
I set the IMU gyro and acc axis to the "correct" direction by playing back rosbags with recordings that work and watched carefully when the device rolled, pitched, yawed, how their IMU responded.
I also tried setting the IMU walk and noise to higher and lower values, same error.
Any thoughts?

I'm guessing I would really need to "calibrate" the IMU together with the camera. Can something like this help: https://github.com/ethz-asl/kalibr/wiki/camera-imu-calibration
One problem is I can only send IMU data at about 60 Hz. (I tried sending it at higher rates but I would duplicate/interpolate the data).

maghauke, 2021-11-18:
@qliao001  I believe the following can be causing this error:
- You have to calibrate your imu and camera and have an accurate transformation between camera and IMU
- Do you have enough motion in all directions, take a look at using ORBSLAM3 with e.g. EuRoC dataset: MH_01_easy.bag, seems like motion initially is important for the initialization of ORBSLAM3 to work
- also the IMU noise must be tuned for best performance, not sure how to find the best parameters here, but maybe that can be done in the calibration?

Fisine, 2022-10-31:
我也遇到了同样的问题，我多次标定，并且给了足够的运动，但还是循环失败，我找不出问题所在，有人解决这个问题吗

Codeing-ZYFan, 2023-10-28:
I'm also having this issue, trying to recalibrate the IMU using [allan_variance_ros](https://github.com/ori-drs/allan_variance_ros/) mentioned in [A summary of mono-inertial initialization issues, and suggestions for others](https://github.com/UZ-SLAMLab/ORB_SLAM3/issues/736) and then adjust the IMU noise parameters:
```
# IMU noise (Use those from VINS-mono)
IMU.NoiseGyro: 1e-3 # rad/s^0.5
IMU.NoiseAcc: 1e-2   # m/s^1.5
IMU.GyroWalk: 1e-7 # rad/s^1.5
IMU.AccWalk: 1e-7 # m/s^2.5
IMU.Frequency: 200
```
Then things got better.

MaZhaoYong, 2024-05-16:
> 我也遇到了同样的问题，我多次标定，并且给了足够的运动，但还是循环失败，我找不出问题所在，有人解决这个问题吗

hello，我也遇到了同样的问题，请问您解决了么，麻烦不吝赐教了

wast0n, 2024-11-13:
> > 我也遇到了同样的问题，我多次标定，并且给了足够的运动，但还是循环失败，我找不出问题所在，有人解决这个问题吗
> 
> hello，我也遇到了同样的问题，请问您解决了么，麻烦不吝赐教了

IMU的ros节点QOS设置为best_effort。“stereo-inertial node did not work because of QOS setting in IMU subscriber.“ [https://github.com/jnskkmhr/orbslam3/tree/main](url)