UZ-SLAMLab/ORB_SLAM3 #72: Memory leak in Tracker.cc?

choffmann, 2020-08-19:
Following line (and probably more) leaks memory, right?

https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/8ac600afe0c7f3f8a916d8cc0e16c7bda36027e5/src/Tracking.cc#L671

You point to new data but don't free the existing.

Also, it does not use any sort of data management (e.g. smart pointer or something). Why don't you use unique or even shared pointer here?

HongmouZhang, 2020-08-20:
I agree, I think such problem exists everywhere and tones of variables are used without initiation, most of them come from ORBSLAM2.....

choffmann, 2020-08-20:
After writing, I noticed as well, that there are much more cases where memory leaks and yeah, even variables which are declared and never used or used but not initialized, as you mentioned, etc. ....

But the code is beta, so it is expected to be a little messy. 

This issue was more like a reminder, that there are memory leaks, which are worse (in my opinion) than the other issues mentioned as well. And maybe a solution, to use smart pointer, which are available since c++11 or earlier if you consider boost.

I'm not sure if it makes sense to fork and fix a couple of issues and create a PR, as I think they are still working on it heavily.

791136190, 2020-11-26:
> Hi all, @choffmann is RIGHT, there are many memory leaks (mostly of them inherited from ORB-SLAM2). As you guys already noted, the code is still in an early version and we are now working in the most requested features and solving major issues. We will eventually clean up the code and solve any memory leak and dead code. However, we greatly appreciate your comments regarding the code for future improvements!

I have the same problem with the key frame, is it resolved in synchronization?

edavalosanaya, 2023-07-24:
Hello, 

After testing the most recent version of ORBSLAM3, I still identify a memory leak issue. If anyone has suggestions, please let me know. I started looking into this [GitHub repo](https://github.com/fishmarch/ORB-SLAM3-Remove-Bad) that tries to address this. I am trying to update my own [repo that provides Python bindings](https://github.com/edavalosanaya/pyorbslam).

AndrewBloom, 2024-02-10:
I'm running this on android and, looking at the profiler, there are memory leaks that consume 0.1 Gb a second...