UZ-SLAMLab/ORB_SLAM3 #73: The trajectories format of mono_kitti and stereo_kitti are not the same

squaresoft2015, 2020-08-19:
The Monocular program of ORB_SLAM3 generates TUM format trajectory (8 digits per row) and the stereo one generates KITTI format (12 digits per row). How can I plot them in the same coordinate with evo? evo needs to specify the trajectory format before loading the pose data.

richard-elvira, 2020-08-21:
ORBSLAM3 only reports the trajectory at frame level, that means we only obtain the pose when an image exists. In the other hand, EuRoc provides a GT with other measure devices, in Machine Hall uses Leica and in Vicon Room an OptiTrack. For that reason the amount of poses is not the same.

squaresoft2015, 2020-08-21:
> ORBSLAM3 only reports the trajectory at frame level, that means we only obtain the pose when an image exists. In the other hand, EuRoc provides a GT with other measure devices, in Machine Hall uses Leica and in Vicon Room an OptiTrack. For that reason the amount of poses is not the same.

Thank you so much! I have converted the groundtruh to TUM format. But when I plot the two trajectories in the same coordinate with rescale and alignment option, the evo program promopt: [ERROR] found no matching timestamps between reference and f_dataset-MH01_mono.txt with max. time diff 0.01 (s) and time offset 0.0 (s)
How can I align them since the time stamp may not be in the corresponding position?

richard-elvira, 2020-08-21:
Check the timestamp units, perhaps one of the files has the timestamps in seconds and the other in nanoseconds.

254843, 2021-01-19:
Hey! I also met the problem of real pose and estimated pose timestamp. Have you solved it? @squaresoft2015

dinarkino, 2021-03-17:
> Hey! I also met the problem of real pose and estimated pose timestamp. Have you solved it? @squaresoft2015

Hey, I solved the problem by using an advice from [here](https://github.com/MichaelGrupp/evo/issues/12). You should do something like that with an estimated pose with the help of evo package or manually.

```
from evo.tools import file_interface

traj = file_interface.read_tum_trajectory_file("vins_V102.txt")

traj.timestamps = traj.timestamps / 1e9

file_interface.write_tum_trajectory_file("vins_V102_clean.txt", traj)
```

jdtardos, 2021-12-23:
Pleas, check version v1.0

closed 2021-12-23 by jdtardos