UZ-SLAMLab/ORB_SLAM3 #75: Error while running Stereo Intertial Node

zainmehdi, 2020-08-20:
Hi,
I am trying to run Stereo Inertial Node in live mode with camera and imu feed coming from T265 and D435i. I have made sure that the topics are being published. But it always crashes with the following error.

OpenCV Error: Assertion failed (0 <= _rowRange.start && _rowRange.start <= _rowRange.end && _rowRange.end <= m.rows) in Mat, file /build/opencv-L2vuMj/opencv-3.2.0+dfsg/modules/core/src/matrix.cpp, line 483
terminate called after throwing an instance of 'cv::Exception'
  what():  /build/opencv-L2vuMj/opencv-3.2.0+dfsg/modules/core/src/matrix.cpp:483: error: (-215) 0 <= _rowRange.start && _rowRange.start <= _rowRange.end && _rowRange.end <= m.rows in function Mat

Can you please provide some insight into this issue ?

Thanks

jjgr3496, 2020-08-20:
Hi @zainmehdi, could you please try to find in which part of the code is that assertion being thrown? Also if possible, could you share with us the calibration file you are using?

zainmehdi, 2020-08-20:
Hi @jjgr3496 thanks for the reply. I am sharing the calibration files I tried meanwhile I look for the part which throws the assertion.

For T265:

Camera.type: "PinHole"

Camera calibration and distortion parameters (OpenCV) 
Camera.fx: 285.3514099121094
Camera.fy: 285.52569580078127
Camera.cx: 421.63848876953127
Camera.cy: 392.61981201171877

Camera.k1: -0.003909272141754627
Camera.k2: 0.03879940137267113
Camera.p1: -0.0367293618619442
Camera.p2: 0.005974404979497194

Camera.width: 848
Camera.height: 800

Camera frames per second 
Camera.fps: 30.0

IR projector baseline times fx (aprox.)
Camera.bf: 40.0

Color order of the images (0: BGR, 1: RGB. It is ignored if images are grayscale)
Camera.RGB: 1

Close/Far threshold. Baseline times.
ThDepth: 40.0

Deptmap values factor
DepthMapFactor: 1.0

ORB Parameters

ORB Extractor: Number of features per image
ORBextractor.nFeatures: 500 #1000

ORB Extractor: Scale factor between levels in the scale pyramid 	
ORBextractor.scaleFactor: 1.2

ORB Extractor: Number of levels in the scale pyramid	
ORBextractor.nLevels: 5 #8

ORB Extractor: Fast threshold

Image is divided in a grid. At each cell FAST are extracted imposing a minimum response.
Firstly we impose iniThFAST. If no corners are detected we impose a lower value minThFAST
You can lower these values if your images have low contrast			
ORBextractor.iniThFAST: 10 #20
ORBextractor.minThFAST: 7

Viewer Parameters

Viewer.KeyFrameSize: 0.05
Viewer.KeyFrameLineWidth: 1
Viewer.GraphLineWidth: 0.9
Viewer.PointSize:2
Viewer.CameraSize: 0.08
Viewer.CameraLineWidth: 3
Viewer.ViewpointX: 0

zainmehdi, 2020-08-21:
I found the issue. I didnt include IMU extrinsic and bias parameters. That was causing the crash. After putting in IMU parameters it started working. The results however were not good which I am sure has something to do with calibration.

closed 2020-09-01 by jjgr3496

leo-wkd, 2020-10-21:
Hi, could you please share your calibration file of T265 or D435i? I calibred both of these two cameras but found something wrong with the transformation Tbc. And after I ran the program, it would initialize almost every second because of tracking lost.
Thank you

zainmehdi, 2020-10-21:
@leo-wkd I posted my T265 calibration file above for D435 its been a while and I need to look for it. I will surely  post it here if I am able to find it.

leo-wkd, 2020-10-21:
@zainmehdi Thank you for your reply. The calibration file above missed the IMU-related information. Could you please email to me the final-edition configuration file? My email address is  leo97wang@gmail.com
Thank you.