UZ-SLAMLab/ORB_SLAM3 #76: Relocalization not possible

fpczk, 2020-08-20:
When I pass two sequences (monocular pinhole camera only) to ORB-SLAM3 one after another, ORB-SLAM3 creates a new map for each sequence and merges them when it recognizes common places. This happens even if the 2nd sequence starts at a place that was already visited in the 1st sequence. This happens even if both sequences are exactly the same.

ORB-SLAM2, on the other hand, would try to relocalize in the map of the 1st sequence, and then re-use that map directly. Can this behaviour be achieved with ORB-SLAM3? Right now, I'm still using ORB-SLAM2 for some cases, because the way ORB-SLAM3 handles these breaking points between sequences has two major disadvantages:

- Initialization must be done every time a new sequence starts, or more generally, every time the track is lost
- Merging does not always work as well as expected. When I pass my 2 sequences into ORB-SLAM2, it can easily relocalize at the beginning of the 2nd seq., and extend the map of the first seq. But in ORB-SLAM3, I often end up with 2 non-merged maps.

When I try to force ORB-SLAM3 to relocalize instead of starting a new map (by modification of the `Tracking::Track()` function), relocalization fails. I came across  [this comment](https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/adc40628f01682b2fe4d143270daa4fe6d4ada8d/src/Tracking.cc#L1048), which makes me wonder if relocalization works at all in ORB-SLAM3:

```
                if (mState == RECENTLY_LOST)
                {
...
                        // TODO fix relocalization
                        bOK = Relocalization();
...
                }
```

For the test, I used the mono_kitti example applications, and passed 2 sequences to ORB-SLAM by simply appending the 2nd to the 1st.

richard-elvira, 2020-09-07:
The system tries to relocalize the frame in the map since the beginning of the lost track along 3 seconds, in another case a new map is created.

When a new map is created the system needs to initialize it, but in the monocular case can be not trivial.

wpbird007, 2020-09-29:
ORB_SLAM3/src/Tracking.cc

Line 3230 in ef97841

 vpMLPnPsolvers[i] = pSolver; 

you should after this line ,add nCandidates++; maybe fix the bug

closed 2020-10-19 by jdtardos