UZ-SLAMLab/ORB_SLAM3 #79: SIGSEGV while running Stereo-Inertial with ROS

rmvanarse, 2020-08-24:
I am using the system with a custom stereo rig and an IMU (calibrated using kalibr), from a rosbag. At certain specific points in the bag, the system crashes with a SIGSEGV. 
Following is the output of gdb backtrace:

```
(gdb) backtrace
#0  0x00007ffff66f9276 in ORB_SLAM3::Tracking::PredictStateIMU() ()
    at /home/aremvy/UNR_THESIS/marine_slam_ws/src/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../lib/libORB_SLAM3.so
#1  0x00007ffff6700da8 in ORB_SLAM3::Tracking::TrackWithMotionModel() ()
    at /home/aremvy/UNR_THESIS/marine_slam_ws/src/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../lib/libORB_SLAM3.so
#2  0x00007ffff67029b1 in ORB_SLAM3::Tracking::Track() ()
    at /home/aremvy/UNR_THESIS/marine_slam_ws/src/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../lib/libORB_SLAM3.so
#3  0x00007ffff670424c in ORB_SLAM3::Tracking::GrabImageStereo(cv::Mat const&, cv::Mat const&, double const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) ()
    at /home/aremvy/UNR_THESIS/marine_slam_ws/src/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../lib/libORB_SLAM3.so
#4  0x00007ffff66de25c in ORB_SLAM3::System::TrackStereo(cv::Mat const&, cv::Mat const&, double const&, std::vector<ORB_SLAM3::IMU::Point, std::allocator<ORB_SLAM3::IMU::Point> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) ()
    at /home/aremvy/UNR_THESIS/marine_slam_ws/src/ORB_SLAM3/Examples/ROS/ORB_SLAM3/../../../lib/libORB_SLAM3.so
#5  0x0000555555569f82 in ImageGrabber::SyncWithImu() ()
#6  0x00007ffff53486df in  () at /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#7  0x00007ffff78296db in start_thread (arg=0x7fffa2289700) at pthread_create.c:463
#8  0x00007ffff4a05a3f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```
By adding a series of print statements, I could trace the fault to be coming from this line:
https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/8ac600afe0c7f3f8a916d8cc0e16c7bda36027e5/src/Tracking.cc#L715

I further tried to reproduce the error with the EuRoC dataset, but wasn't able to do so. EuRoC works perfectly.

What might be causing this error? Is there a way to simply skip the faulty instances, as the system works fine for the rest of the duration

richard-elvira, 2020-09-07:
Try to check if the mpImuPreintegratedFrame is empty, the variable is loaded with the last preintegration measures between frames, it could happen that there are not IMU measures between 2 consecutive frames or ROS doesn't provide them right

rmvanarse, 2020-10-02:
Thank you! Turns out my rosbag had several pairs of consecutive frames without IMU measurements between them. Interpolating the IMU to add new messages between these frames solved the problem

closed 2020-10-02 by rmvanarse