UZ-SLAMLab/ORB_SLAM3 #80: Extending ORBSLAM3 to RGBD+Inertial

ankuPRK, 2020-08-24:
Hey guys I have some data where I have RGBD data along with IMU measurements. 

I was wondering if any feasibility issue will arise if I try to merge TUM RGDB system with EUROC stereo-imu such that the depth representation of TUM replaces Stereo of EUROC. If I remember correctly in ORBSLAM2 both Stereo and RGBD modalities were internally calling same tracking and mapping utilities.

jjgr3496, 2020-08-24:
We have not tested RGBD+inertial as we have not found any dataset that combines both of them.  However, as you have already noted, ORB-SALM2 shares the same pipeline for stereo and RGBD sensors. We keep the same pipeline for ORB-SLAM3 so I think it would be feasible to try a RGBD+inertial setup with ORB-SLAM3. We would be happy if you share with us any results you get in this direction!

ankuPRK, 2020-08-24:
Thanks! Sure if I get something meaningful will definitely share! :)

closed 2020-08-24 by ankuPRK

oym1994, 2020-12-08:
We are also working on RGBD-Inertial slam and recommend this  [OpenLORIS dataset](https://lifelong-robotic-vision.github.io/dataset/scene) . It comprises all kinds of sensor data needed by ORM_SLAM3, including mono RGBD-IMU, Stereo fisheye-IMU.

macTracyHuang, 2022-05-30:
> We are also working on RGBD-Inertial slam and recommend this [OpenLORIS dataset](https://lifelong-robotic-vision.github.io/dataset/scene) . It comprises all kinds of sensor data needed by ORM_SLAM3, including mono RGBD-IMU, Stereo fisheye-IMU.

hi, could you plz share how you implement rgbd-imu with ros? thx

wqx854987945, 2022-08-04:
> We are also working on RGBD-Inertial slam and recommend this [OpenLORIS dataset](https://lifelong-robotic-vision.github.io/dataset/scene) . It comprises all kinds of sensor data needed by ORM_SLAM3, including mono RGBD-IMU, Stereo fisheye-IMU.

Hi,in RGBD-Inertial slam ,what parameters do you set in imu.yaml(IMU.NoiseGyro/IMU.NoiseAcc/IMU.GyroWalk/IMU.AccWalk)? Could you plz show your  ".yaml"   about OpenLORIS dataset in ORBSLAM3?

liyuting-123, 2022-08-20:
> 我们也在研究 RGBD-Inertial slam 并推荐这个 [OpenLORIS 数据集 ](https://lifelong-robotic-vision.github.io/dataset/scene)。 它包含了 ORM_SLAM3 所需的各种传感器数据，包括单声道 RGBD-IMU、立体鱼眼-IMU。

hi, could you plz share how you implement rgbd-imu with ros? thx

liyuting-123, 2022-08-20:
> We are also working on RGBD-Inertial slam and recommend this [OpenLORIS dataset](https://lifelong-robotic-vision.github.io/dataset/scene) . It comprises all kinds of sensor data needed by ORM_SLAM3, including mono RGBD-IMU, Stereo fisheye-IMU.

hi, could you plz share how you implement rgbd-imu with ros? thx