UZ-SLAMLab/ORB_SLAM3 #81: cv::imread fail in mono_euroc

BeardyMan37, 2020-08-24:
`
cv::Mat im;
        int proccIm = 0;
        for(int ni=0; ni<nImages[seq]; ni++, proccIm++)
        {

            // Read image from file
            im = cv::imread(vstrImageFilenames[seq][ni],CV_LOAD_IMAGE_UNCHANGED);
            double tframe = vTimestampsCam[seq][ni];

            if(im.empty())
            {
                cerr << endl << "Failed to load image at: "
                     <<  vstrImageFilenames[seq][ni] << endl;
                return 1;
            }
`
In this codeblock of mono_euroc.cc, cv::imread fails to load the Euroc dataset images, and segmentation fault occurs. Any idea what might be the issue? I checked the EuRoC.yaml file following #3 and it already says the Camera.type is PinHole.

jjgr3496, 2020-08-25:
It looks like you are not calling the program with the correct arguments. Could you please copy here the command you are using?

BeardyMan37, 2020-08-25:
I ran the euroc_examples.sh.
`#!/bin/bash
pathDatasetEuroc='~/ORB_SLAM3/dataset/Euroc' #Example, it is necesary to change it by the dataset path

#------------------------------------
# Monocular Examples
echo "Launching MH01 with Monocular sensor"
./Examples/Monocular/mono_euroc ./Vocabulary/ORBvoc.txt ./Examples/Monocular/EuRoC.yaml "$pathDatasetEuroc"/MH01 ./Examples/Monocular/EuRoC_TimeStamps/MH01.txt dataset-MH01_mono`

BeardyMan37, 2020-08-25:
`~/ORB_SLAM3$ ./euroc_examples.sh 
Launching MH01 with Monocular sensor
num_seq = 1
file name: dataset-MH01_mono
Loading images for sequence 0...LOADED!

-------

ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Monocular

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name: 

Camera Parameters: 
- fx: 458.65399169921875
- fy: 457.29598999023438
- cx: 367.21499633789062
- cy: 248.375
- bf: 0
- k1: -0.28340810537338257
- k2: 0.073959067463874817
- p1: 0.00019359000725671649
- p2: 1.7618711353861727e-05
- fps: 20
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 1000
- Scale Levels: 8
- Scale Factor: 1.2000000476837158
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

Failed to load image at: ~/ORB_SLAM3/dataset/Euroc/MH01/mav0/cam0/data/1403636579763555584.png
./euroc_examples.sh: line 7: 20234 Segmentation fault      (core dumped) ./Examples/Monocular/mono_euroc ./Vocabulary/ORBvoc.txt ./Examples/Monocular/EuRoC.yaml "$pathDatasetEuroc"/MH01 ./Examples/Monocular/EuRoC_TimeStamps/MH01.txt dataset-MH01_mono
`
I was alson able to open 1403636579763555584.png manually, so the dataset is okay. It's also the first picture of the dataset.

choffmann, 2020-08-25:
Maybe investigate the core dump in gdb? See where it failed, which image it tried to load etc.

Btw... to enable core dumps to be generated, you have to execute `ulimit -c unlimited`. It is disabled by default in Ubuntu.

BeardyMan37, 2020-08-25:
> Maybe investigate the core dump in gdb? See where it failed, which image it tried to load etc.
> 
> Btw... to enable core dumps to be generated, you have to execute `ulimit -c unlimited`. It is disabled by default in Ubuntu.

`
        im = cv::imread(vstrImageFilenames[seq][ni],CV_LOAD_IMAGE_UNCHANGED);
`
In this line, cv::imread fails to read the png image provided by the dataset, and so, when the code returns 1 in the main function, it tries to free im, which is empty, and thus, it encounters the error,
Failed to load image at: ~/ORB_SLAM3/dataset/Euroc/MH01/mav0/cam0/data/1403636579763555584.png
free(): invalid pointer
This is as far I understood. I checked 1403636579763555584.png manually and it exists. Thus I'm having problem understanding why cv::imread fails, as I have OpenCV 3.2.0 installed.

choffmann, 2020-08-26:
Please provide canonical path. `~` is only valid in bash, AFAIK.

BeardyMan37, 2020-08-26:
Thank you. It worked. Closing the issue.

closed 2020-08-26 by BeardyMan37