UZ-SLAMLab/ORB_SLAM3 #85: Tracker stuck in initialization state

choffmann, 2020-08-27:
As said, tracker is stuck in initialization. So, no matches are found, even though ~5000 key points are found. I tested several data set from us (driving car). Still no success. Could you give me some pointer what might the problem here?

jjgr3496, 2020-09-01:
We are aware of some issue with monocular initialization even though it is the same as ORB-SLAM2. We are currently working in a new initialization algorithm so this should be solved shortly!

choffmann, 2020-09-04:
Actually, ORB-SLAM is now working on my data set. I played around with the settings and increasing the setting `ORBextractor.nFeatures: 3000` to 3000 solved my issue. Even 2000 worked for me. I was using images which are about ~2.5 bigger than the sample images.

closed 2020-09-04 by choffmann