UZ-SLAMLab/ORB_SLAM3 #88: Lapping area between images

ankuPRK, 2020-08-29:
Could someone help about what is the lapping area for fisheye stereo config file TUM_512.yaml?

```
# Lapping area between images
Lapping.left: 0
Lapping.right: 511

```
I was working on a dataset which uses t265 fisheye stereo camera so I wanted to know how to set this parameter. Thanks!

XiongLianga, 2020-09-01:
this can be set in the "thread threadLeft(&Frame::ExtractORB,this,0,imLeft,0,511)". can you tell me the  definition of lapping erea？

ankuPRK, 2020-09-01:
I don't know I tried looking it up but couldn't find anything. That's why I got confused.

jjgr3496, 2020-09-01:
@ankuPRK @XiongLianga  This will solved in a new realease this week. However, I will provide you with a short explanation:

Stereo cameras are usually rectified for the ease of finding stereo matches. This rectifying step is very well-known for pinhole cameras and it has negligible effects on the FoV of the original images.

However, rectifying stereo fisheye cameras yields to a trade back depending of the calibration of the rectified images:

- Keep the characteristic FoV of the fisheye cameras but greatly magnify objects at the boundaries of the image and loose resolution at the center of the image.
-Loose a huge quantity of FoV but make things in the center of the image look fine (like a pinhole camera).

These reasons discourage the use of rectification over fisheye images. Instead of that, we use the 2 raw stereo images modifying in a new stereo approch. In this approach, we consider the stereo rig as 2 monocular cameras constrianed by:

1. A constant relative SE(3) transformation.
2. Optionally an image region that observe the same portion of the scene (the lapping area).

Within this lapping area, we can get stereo, truescaled, points which combined with the relative pose of the cameras allows us to retrieve the real scale of the map. This lapping area is likely to be different depending of your stereo rig so we decided to include it as a parameter to identify in which regions of the image should consider features as "stereo" or "monocular".

This approach allows us to use all the features extracted in **both** images which renders a more robust stereo system. For example, if one camera gets completely occluded, the other one can be used for a monocular tracking in the meanwhile the other occluded.

Since the code is in a beta version and we ran a lot of experiments, there is still a lot of dead code and hardcoded parameters (in this case, the lapping area). That is why I apologize for the confusion. However, as said before, this will solved shortly with a parameter parser and a correct implementation of the lapping area.

ankuPRK, 2020-09-02:
Thank`you! The explanation makes perfect sense. Just one more follow-up doubt. In the yaml file lapping.left=0 and lapping.right=511, so is the origin of these coordinates center of left camera?

jjgr3496, 2020-09-07:
Please, take a look at the latest version. The parameters you have to set in the calibration file regarding the lapping area are the following:

- `Camera.lappingBegin` and `Camera2.lappingBegin`: these are the most left column in each image where the lapping area begins

- `Camera.lappingEnd` and Camera2.lappingEnd`: these are the most right column in each image where the lapping area ends

However, these parameters will eventually computed automatically in a future version from extrinsic and intrinsic calibration. If you have any doubt, please reopen this issue!

closed 2020-09-07 by jjgr3496

ankuPRK, 2020-09-07:
Thanks again this makes sense! We can keep the issue closed.

ihamieh, 2020-09-29:
Does ORB_SLAM3 update the camera exteinsic calibration?

tejalbarnwal, 2021-09-20:
@jjgr3496 
If the image width taken by cameras is w, will the camera1/2.lappingEnd always be w-1 and camera1/2.lappingBegin would always be 0?

tejalbarnwal, 2021-09-20:
@jjgr3496 are there any methods through which I can calculate the overlapping area between two cameras given their extrinsic and intrinsic parameters to provide them into the YAML file?

YanhaoZhang, 2022-02-08:
Hi @jjgr3496, thanks for your explanation on the lapping area. From `Frame.cc`, I saw that the default parameters are 0, does it mean we assume there is no overlap area among the stereo images? 
``` C++
thread threadLeft(&Frame::ExtractORB,this,0,imLeft,0,0);     
thread threadRight(&Frame::ExtractORB,this,1,imRight,0,0);
```
I am trying to do SLAM using multiple cameras with a fixed relative pose but without overlap area. In this case, I do not need to modify the default 0 parameter here, do I? Thank you.

jjgr3496, 2022-02-08:
> @jjgr3496 are there any methods through which I can calculate the overlapping area between two cameras given their extrinsic and intrinsic parameters to provide them into the YAML file?

Sorry, I have been a bit disconnected from the repo lately.

A method for computing the lapping area could be computing the FoV of both cameras and, considering the extrinsic parameters between cameras, check where both camera frustrums collide. With that commong area, one could reproject it to both cameras and check which pixels belongs to the lapping area.

> Hi @jjgr3496, thanks for your explanation on the lapping area. From `Frame.cc`, I saw that the default parameters are 0, does it mean we assume there is no overlap area among the stereo images? [...]

Those lines belongs to the PinHole stereo constructor (for which we do not make use of the lapping area and we set it to 0). The constructor for FishEye stereo cameras, for which we use the lapping area, can be found at Frame.h, lines 1159 & 1160, for which we read them from the calibration file!

Hope this solves your doubts!

YanhaoZhang, 2022-02-09:
> 

> Those lines belongs to the PinHole stereo constructor (for which we do not make use of the lapping area and we set it to 0). The constructor for FishEye stereo cameras, for which we use the lapping area, can be found at Frame.h, lines 1159 & 1160, for which we read them from the calibration file!

Got it! Thanks a lot @jjgr3496