UZ-SLAMLab/ORB_SLAM3 #89: Running ORB-SLAM3 with ROS - Problems!

ihamieh, 2020-08-31:
When I try to execute build_ros.sh script, the get the following error message: 

Ish@US:~$ cd ORB_SLAM3/
Ish@US:~/ORB_SLAM3$ chmod +x build_ros.sh
Ish@US:~/ORB_SLAM3$ ./build_ros.sh 
Building ROS nodes
mkdir: cannot create directory ‘build’: File exists
[rosbuild] Building package ORB_SLAM3
[rosbuild] Error from directory check: /opt/ros/kinetic/share/ros/core/rosbuild/bin/check_same_directories.py  /home/Ish/ORB_SLAM3/Examples/ROS/ORB_SLAM3
1
Traceback (most recent call last):
  File "/opt/ros/kinetic/share/ros/core/rosbuild/bin/check_same_directories.py", line 46, in <module>
    raise Exception
Exception
CMake Error at /opt/ros/kinetic/share/ros/core/rosbuild/private.cmake:102 (message):
  [rosbuild] rospack found package "ORB_SLAM3" at "", but the current
  directory is "/home/Ish/ORB_SLAM3/Examples/ROS/ORB_SLAM3".  You should
  double-check your ROS_PACKAGE_PATH to ensure that packages are found in the
  correct precedence order.
Call Stack (most recent call first):
  /opt/ros/kinetic/share/ros/core/rosbuild/public.cmake:177 (_rosbuild_check_package_location)
  CMakeLists.txt:4 (rosbuild_init)

-- Configuring incomplete, errors occurred!
See also "/home/Ish/ORB_SLAM3/Examples/ROS/ORB_SLAM3/build/CMakeFiles/CMakeOutput.log".
make: *** No targets specified and no makefile found.  Stop.

yutongwang123, 2020-09-01:
I got exactly the same issue, right now.....

jjgr3496, 2020-09-01:
This error comes because you forgot some steps mentioned in the ReadMe related with rose. More specifically:

1. Add the path including *Examples/ROS/ORB_SLAM3* to the ROS_PACKAGE_PATH environment variable. Open .bashrc file:
  ```
  gedit ~/.bashrc
  ```
and add at the end the following line. Replace PATH by the folder where you cloned ORB_SLAM3:

  ```
  export ROS_PACKAGE_PATH=${ROS_PACKAGE_PATH}:PATH/ORB_SLAM3/Examples/ROS
  ```

After doing this, you should be able to compile the ROS examples.

weblucas, 2020-09-14:
Easier, if you just want to use it.

https://github.com/VIS4ROB-lab/orb_slam_3_catkin

ihamieh, 2020-09-14:
@weblucas @jjgr3496 I was able to run this successfully! Thank you guys ...

closed 2020-09-14 by ihamieh

HorsonLiu, 2020-11-12:
Still error on Ubuntu 16.04.
`
Building ROS nodes
mkdir: cannot create directory ‘build’: File exists
[rosbuild] Building package ORB_SLAM3
[rosbuild] Error from directory check: /opt/ros/kinetic/share/ros/core/rosbuild/bin/check_same_directories.py  /home/add/workspace/slam/ORB_SLAM3/Examples/ROS/ORB_SLAM3
1
Traceback (most recent call last):
  File "/opt/ros/kinetic/share/ros/core/rosbuild/bin/check_same_directories.py", line 46, in <module>
    raise Exception
Exception
CMake Error at /opt/ros/kinetic/share/ros/core/rosbuild/private.cmake:102 (message):
  [rosbuild] rospack found package "ORB_SLAM3" at "", but the current
  directory is "/home/add/workspace/slam/ORB_SLAM3/Examples/ROS/ORB_SLAM3".
  You should double-check your ROS_PACKAGE_PATH to ensure that packages are
  found in the correct precedence order.
Call Stack (most recent call first):
  /opt/ros/kinetic/share/ros/core/rosbuild/public.cmake:177 (_rosbuild_check_package_location)
  CMakeLists.txt:4 (rosbuild_init)

-- Configuring incomplete, errors occurred!
See also "/home/add/workspace/slam/ORB_SLAM3/Examples/ROS/ORB_SLAM3/build/CMakeFiles/CMakeOutput.log".
make: *** No targets specified and no makefile found.  Stop.
`

owenwijaya177, 2021-06-04:
I did what you told @jjgr3496 but still gives the same error. Any suggestions?

Kanishk598, 2021-06-27:
> I did what you told @jjgr3496 but still gives the same error. Any suggestions?

Before executing the script `./build_ros.sh`, you should first source your `~/.bashrc` file using the command `source ~/.bashrc`. Now your script will not give the error.

MarcusDudeBro, 2021-07-02:
using source ~/.bashrc also doesn't fix the issue for me

Kanishk598, 2022-04-01:
> @MarcusDudeBro were you able to fix this issue?

This issue can be solved by the following method:
`export ROS_PACKAGE_PATH=${ROS_PACKAGE_PATH}:${PWD}/Examples/ROS`
Run the above command while being present in the ORB_SLAM3 repo. Now running the `build_ros.sh` script won't give any errors.

tungtungyan, 2022-05-18:
> > @MarcusDudeBro were you able to fix this issue?
> 
> This issue can be solved by the following method: `export ROS_PACKAGE_PATH=${ROS_PACKAGE_PATH}:${PWD}/Examples/ROS` Run the above command while being present in the ORB_SLAM3 repo. Now running the `./build_ros.sh` script won't give any errors.

My result is this.
echo $ROS_PACKAGE_PATH
/opt/ros/melodic/share:/media/ytl/2TB/catkin_orb/src/ORB_SLAM3/Examples/ROS:PATH/ORB_SLAM2_modified/Examples/ROS
How can I remove the medium part ? (/media/ytl/2TB/catkin_orb/src/ORB_SLAM3/Examples/ROS)

Kanishk598, 2022-05-19:
You can manually set the path by typing, whatever you want your ROS_PACKAGE_PATH to be. To do this, run the command `export ROS_PACKAGE_PATH=/your/desired/paths:/another/path/you/want/to/put:/yet/another/path`. Don't source your `.bashrc` file after doing this or else your path with reset with what is present in the file.

AndrewBloom, 2023-10-09:
I can see on ./build_ros.sh 
cd Examples/ROS/ORB_SLAM3

but this folder is not present on the repo, and seems it is in Examples_old instead

kanishkanarch, 2023-10-09:
Please try with `Examples_old` and see if it works.
Make sure you're using the correct branch of the repository.