UZ-SLAMLab/ORB_SLAM3 #90: Running ORB-SLAM3 with ZED2 always appears error.

yike-chen, 2020-09-03:
I has run [ORB-SLAM3](https://github.com/UZ-SLAMLab/ORB_SLAM3) with Stereo+IMU. I set the parameters following.
```
%YAML:1.0

#--------------------------------------------------------------------------------------------
# Camera Parameters. Adjust them!
#--------------------------------------------------------------------------------------------
Camera.type: "PinHole"

# Camera calibration and distortion parameters (OpenCV) 
# I get it from the topic of /zed2/zed_node/left/camera_info.
Camera.fx: 537.50732421875
Camera.fy: 537.50732421875
Camera.cx: 630.62646484375
Camera.cy: 362.08624267578125

# I use the rectification images, so set these with zeros.
Camera.k1: 0.0
Camera.k2: 0.0
Camera.p1: 0.0
Camera.p2: 0.0

Camera.width: 1280
Camera.height: 720

# Camera frames per second 
Camera.fps: 15.0

# IR projector baseline times fx (aprox.)
# I get it from the topic of /zed2/zed_node/right/camera_info with P(4).
Camera.bf: 64.62

# Color order of the images (0: BGR, 1: RGB. It is ignored if images are grayscale)
Camera.RGB: 1

# Close/Far threshold. Baseline times.
ThDepth: 40.0

# Deptmap values factor
DepthMapFactor: 1.0

# Transformation from camera 0 to body-frame (imu)
# I get it from topic of /zed2/zed_node/left_cam_imu_transform, and it seems more precise with STARTER CONSOLE information "Camera-IMU Transform" 
Tbc: !!opencv-matrix
   rows: 4
   cols: 4
   dt: f
   data: [0.999956756829, 0.00843152806614, 0.00392349541458, -0.00200000009499,
          -0.0084334428508, 0.99996540322, 0.000471741896564, -0.02300000377,
          -0.00391937794561, -0.000504810067849, 0.999992191792, -0.00200000009499,
         0.0, 0.0, 0.0, 1.0]

# IMU noise
# I get it from Project of **zed-examples/tutorials/tutorial 7 - sensor data/**.
IMU.NoiseGyro: 0.007 # 1.6968e-04 
IMU.NoiseAcc:  0.0016 # 2.0000e-3
IMU.GyroWalk:  0.0019474 
IMU.AccWalk:   0.0002509 # 3.0000e-3
IMU.Frequency: 400
```
I have set all parametes with ZED2 SDK provided, but I can`t run well with ORB-SLAM3(Stereo+IMU). I run well with ORB-SLAM3(Stereo), once I fuse the IMU data it damaged. The error can be found following.
```
New Map created with 313 points
build optimization graph
start optimization
end optimization
update Keyframes velocities and biases
IMU in Map 0 is initialized
PIOLKF: NOT ENOUGH EDGES
Fail to track local map!
IMU is not or recently initialized. Reseting active map...
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 3
mnInitialFrameId = 0
40 Frames set to lost
not IMU meas
First KF:10; Map init KF:0
New Map created with 318 points
build optimization graph
start optimization
end optimization
update Keyframes velocities and biases
IMU in Map 0 is initialized
PIOLKF: NOT ENOUGH EDGES
Fail to track local map!
IMU is not or recently initialized. Reseting active map...
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 44
mnInitialFrameId = 42
79 Frames set to lost
not IMU meas
First KF:20; Map init KF:10
New Map created with 388 points
Fail to track local map!
IMU is not or recently initialized. Reseting active map...
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 84
mnInitialFrameId = 82
83 Frames set to lost
not IMU meas
First KF:21; Map init KF:20
New Map created with 377 points
build optimization graph
start optimization
end optimization
update Keyframes velocities and biases
IMU in Map 0 is initialized
PIOLKF: NOT ENOUGH EDGES
Fail to track local map!
IMU is not or recently initialized. Reseting active map...
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 89
mnInitialFrameId = 87
123 Frames set to lost
not IMU meas
First KF:31; Map init KF:21
New Map created with 423 points
Fail to track local map!
IMU is not or recently initialized. Reseting active map...
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 130
mnInitialFrameId = 128
128 Frames set to lost
not IMU meas
First KF:33; Map init KF:31
New Map created with 271 points
Fail to track local map!
IMU is not or recently initialized. Reseting active map...
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 136
mnInitialFrameId = 134
130 Frames set to lost
not IMU meas
First KF:34; Map init KF:33
New Map created with 220 points
Fail to track local map!
IMU is not or recently initialized. Reseting active map...
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 139
mnInitialFrameId = 137
133 Frames set to lost
not IMU meas
First KF:35; Map init KF:34
New Map created with 377 points
Fail to track local map!
IMU is not or recently initialized. Reseting active map...
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 143
mnInitialFrameId = 141
137 Frames set to lost
not IMU meas
First KF:36; Map init KF:35
New Map created with 405 points
Fail to track local map!
IMU is not or recently initialized. Reseting active map...
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 148
mnInitialFrameId = 146
158 Frames set to lost
not IMU meas
First KF:42; Map init KF:36
New Map created with 328 points
Fail to track local map!
IMU is not or recently initialized. Reseting active map...
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 170
mnInitialFrameId = 168
161 Frames set to lost
not IMU meas
First KF:43; Map init KF:42
New Map created with 387 points
Fail to track local map!
IMU is not or recently initialized. Reseting active map...
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 174
mnInitialFrameId = 172
165 Frames set to lost
not IMU meas
First KF:44; Map init KF:43
New Map created with 359 points
Fail to track local map!
IMU is not or recently initialized. Reseting active map...
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 179
mnInitialFrameId = 177
169 Frames set to lost
not IMU meas
First KF:45; Map init KF:44
New Map created with 255 points
Fail to track local map!
IMU is not or recently initialized. Reseting active map...
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 184
mnInitialFrameId = 182
172 Frames set to lost
not IMU meas
First KF:46; Map init KF:45
New Map created with 385 points
Fail to track local map!
IMU is not or recently initialized. Reseting active map...
```
Can you give me some advices to solve it? Or I must calibrate it with Kalibr?

yike-chen, 2020-09-04:
[ORB-SLAM3](https://github.com/UZ-SLAMLab/ORB_SLAM3/issues/94) only supports Global Shutter cameras.

closed 2020-09-04 by yike-chen

rginjapan, 2020-10-27:
I think this explanation only can explain that the result is not so good when ORB-SLAM3 run on rolling shutter cameras, I still want to reopen this issue to let ORB-SLAM3 at least can run on ZED2 or ZEDmini.

> [ORB-SLAM3](https://github.com/UZ-SLAMLab/ORB_SLAM3/issues/94) only supports Global Shutter cameras.

awenbigprawn, 2021-05-06:
I'm suffering same problem, and I don't believe it is because of rolling shutter cameras. Because the pure Stereo works just fine, I think it's problem with IMU

antithing, 2021-08-12:
@awenbigprawn @yike-chen did you manage to solve this problem and run ORBSLAM3 with Zed in Stereo-inertial mode? Thanks!

awenbigprawn, 2021-08-12:
> @awenbigprawn @yike-chen did you manage to solve this problem and run ORBSLAM3 with Zed in Stereo-inertial mode? Thanks!

Or, you can messure the extrinsic of the imu very very carefully and precisely. Btw, zed2 camera's producer doesn't provide a precise translation matrix between imu and left camera. So, it's hard.😂

antithing, 2021-09-18:
@awenbigprawn thanks! Would you be able to share the code you use to run the Zed2 in stereo inertial mode? Thanks again!

RoZhong, 2021-11-30:
> 

I used Kalibr to calibrate the ZED2 monocular+imu today, and when I ran it, there would always be errors like the author's
[image]
.

amstrudy, 2024-05-23:
Commenting on this old thread to share what I ended up doing to get ORB SLAM Stereo-Inertial to run with my ZED2i camera. I couldn't find a single solution to this on here so hopefully this helps someone!

The main culprit is the `Tbc` matrix in the `.yaml` config file. To ensure that your problem is this matrix, try running just Stereo and you should get reasonable results (assuming your images have enough features). If everything explodes once you switch to Stereo-Inertial, then the `Tbc` matrix is likely the issue. This matrix is meant to be the transform between the camera and imu frame. In order to get this:

1. Run `rosrun tf tf_echo zed2i_left_camera_optical_frame zed2i_imu_link` while your bag file (or whatever) is running and grab the translation + quaternion. For me it was: `Translation: [0.023, -0.000, -0.002]` and `Rotation: in Quaternion [0.504, -0.497, 0.496, 0.503]`. Then I converted these to a 4 x 4 `R | t` matrix.
2. However, this is in `x` forward, `y` left, and `z` up coordinate frame. ORB SLAM expects `z` forward, `y` down, and `x` right. So, in order to rotate between the two I multiplied my `R | t` matrix by the rotations needed:
```math
\begin{bmatrix}0.0139993 & -0.9999020 & -0.000014 & 0.023 \\ -0.0019999 & -0.0000140 & -0.9999980 & 0 \\ 0.99990000 & 0.0139993 & -0.0019999 & -0.002 \\ 0 & 0 & 0 & 1 \end{bmatrix} \begin{bmatrix} 0 & -1 & 0 & 0 \\ 0 & 0 & -1 & 0 \\ 1 & 0 & 0 & 0 \\ 0 & 0 & 0 & 1 \end{bmatrix} = \begin{bmatrix} -0.000014 & -0.0139993 & 0.999902 & 0.023 \\ -0.999998 & 0.0019999 & 0.000014 & 0 \\ -0.0019999 & -0.9999 & -0.0139993 & -0.02 \\ 0 & 0 & 0 & 1 \end{bmatrix}
```
3. That resulting matrix is what you insert into your `.yaml` file. Here's my complete config file for the ZED2i for Stereo-Intertial:
```
%YAML:1.0

#--------------------------------------------------------------------------------------------
# Camera Parameters
#--------------------------------------------------------------------------------------------
Camera.type: "PinHole"

# Camera calibration and distortion parameters (OpenCV) 
Camera.fx: 524.7325439453125
Camera.fy: 524.7325439453125
Camera.cx: 631.94970703125
Camera.cy: 354.74462890625

Camera.k1: 0.0
Camera.k2: 0.0
Camera.k3: 0.0
Camera.p1: 0.0
Camera.p2: 0.0

Camera.bFishEye: 0

Camera.width: 1280
Camera.height: 720

# Camera frames per second 
Camera.fps: 5.0

# stereo baseline in meters times fx
Camera.bf: 62.9323025728 # 119.9321517944336 * 524.732543945 / 100

# Color order of the images (0: BGR, 1: RGB. It is ignored if images are grayscale)
Camera.RGB: 1

# Close/Far threshold. Baseline times.
ThDepth: 45.0

Tbc: !!opencv-matrix
    rows: 4
    cols: 4
    dt: f
    data: [-0.000014, -0.0139993, 0.999902, 0.023,
           -0.999998, 0.0019999, 0.000014, 0.0, 
           -0.0019999, -0.9999, -0.0139993, -0.002,
             0.0, 0.0, 0.0, 1.0]

# IMU noise
IMU.NoiseGyro: 0.007000000216066837 #0.028 #0.007000000216066837
IMU.NoiseAcc: 0.016 #0.0015999999595806003 #0.006 #0.0015999999595806003
IMU.GyroWalk: 0.0019474000437185168 #0.008 #0.0019474000437185168
IMU.AccWalk: 0.0002508999896235764 #0.001 #0.0002508999896235764
IMU.Frequency: 334

#--------------------------------------------------------------------------------------------
# ORB Parameters
#--------------------------------------------------------------------------------------------

# ORB Extractor: Number of features per image
ORBextractor.nFeatures: 8000

# ORB Extractor: Scale factor between levels in the scale pyramid 	
ORBextractor.scaleFactor: 1.5

# ORB Extractor: Number of levels in the scale pyramid	
ORBextractor.nLevels: 8

# ORB Extractor: Fast threshold
# Image is divided in a grid. At each cell FAST are extracted imposing a minimum response.
# Firstly we impose iniThFAST. If no corners are detected we impose a lower value minThFAST
# You can lower these values if your images have low contrast			
ORBextractor.iniThFAST: 20
ORBextractor.minThFAST: 7

#--------------------------------------------------------------------------------------------
# Viewer Parameters
#--------------------------------------------------------------------------------------------
Viewer.KeyFrameSize: 0.05
Viewer.KeyFrameLineWidth: 1
Viewer.GraphLineWidth: 0.9
Viewer.PointSize: 2
Viewer.CameraSize: 0.08
Viewer.CameraLineWidth: 3
Viewer.ViewpointX: 0
Viewer.ViewpointY: -0.7
Viewer.ViewpointZ: -1.8
Viewer.ViewpointF: 500
```

amstrudy, 2024-05-23:
Commenting on this old thread to share what I ended up doing to get ORB SLAM Stereo-Inertial to run with my ZED2i camera. I couldn't find a single solution to this on here so hopefully this helps someone!

The main culprint is the `Tbc` matrix in the `.yaml` config file. To ensure that your problem is this matrix, try running just Stereo and you should get reasonable results (assuming your images have enough features). If everything explodes once you switch to Stereo-Inertial, then the `Tbc` matrix is likely the issue. This matrix is meant to be the transform between the camera and imu frame. In order to get this:
1. Run `rosrun tf tf_echo zed2i_left_camera_optical_frame zed2i_imu_link while your bag file (or whatever) is running and grab the translation + quaternion. For me it was: `Translation: [0.023, -0.000, -0.002]` and `Rotation: in Quaternion [0.504, -0.497, 0.496, 0.503]`. Then I converted these to a 4 x 4 `R | t` matrix.
2. However, this is in `x` forward, `y` left, and `z` up coordinate frame. ORB SLAM expects `z` forward, `y` down, and `x` right. So, in order to rotate between the two I multiplied my `R | t` matrix by the rotations needed:
```math
\begin{bmatrix} 0.0139993 & -0.9999020 & -0.000014 & 0.023 \\ -0.0019999 & -0.0000149 & -0.9999980 & 0 \\ 0.99990000 & 0.0139993 & -0.0019999 & -0.002 \\ 0 & 0 & 0 & 1 \end{bmatrix} \begin{bmatrix} 0 & -1 & 0 & 0 \\ 0 & 0 & -1 & 0 \\ 1 & 0 & 0 & 0 \\ 0 & 0 & 0 & 1 \end{bmatrix} = \begin{bmatrix} -0.000014 & -0.0139993 & 0.999902 & 0.023 \\ -0.999998 & 0.0019999 & 0.000014 & 0 \\ -0.0019999 & -0.99990000 & -0.01399993 & -0.02 \\ 0 & 0 & 0 & 1 \end{bmatrix}
```
3. Add that your `.yaml` file and you should be good. Make sure to also get the proper framerates/noise params. Here is my full config file in case that is helpful to anyone:
```
%YAML:1.0

#--------------------------------------------------------------------------------------------
# Camera Parameters. Adjust them!
# These are taken from the Camera Information in Python API of ZED
#--------------------------------------------------------------------------------------------
Camera.type: "PinHole"

# Camera calibration and distortion parameters (OpenCV) 
Camera.fx: 524.7325439453125
Camera.fy: 524.7325439453125
Camera.cx: 631.94970703125
Camera.cy: 354.74462890625

Camera.k1: 0.0
Camera.k2: 0.0
Camera.k3: 0.0
Camera.p1: 0.0
Camera.p2: 0.0

Camera.bFishEye: 0

Camera.width: 1280
Camera.height: 720

# Camera frames per second 
Camera.fps: 5.0

# stereo baseline in meters times fx
Camera.bf: 62.9323025728 # 119.9321517944336 * 524.732543945 / 100

# Color order of the images (0: BGR, 1: RGB. It is ignored if images are grayscale)
Camera.RGB: 1

# Close/Far threshold. Baseline times.
ThDepth: 45.0

Tbc: !!opencv-matrix
    rows: 4
    cols: 4
    dt: f
    data: [-0.000014, -0.0139993, 0.999902, 0.023,
           -0.999998, 0.0019999, 0.000014, 0.0, 
           -0.0019999, -0.9999, -0.0139993, -0.002,
             0.0, 0.0, 0.0, 1.0]

# IMU noise
IMU.NoiseGyro: 0.007000000216066837 #0.028 #0.007000000216066837
IMU.NoiseAcc: 0.016 #0.0015999999595806003 #0.006 #0.0015999999595806003
IMU.GyroWalk: 0.0019474000437185168 #0.008 #0.0019474000437185168
IMU.AccWalk: 0.0002508999896235764 #0.001 #0.0002508999896235764
IMU.Frequency: 334

#--------------------------------------------------------------------------------------------
# ORB Parameters
#--------------------------------------------------------------------------------------------

# ORB Extractor: Number of features per image
ORBextractor.nFeatures: 8000

# ORB Extractor: Scale factor between levels in the scale pyramid 	
ORBextractor.scaleFactor: 1.5

# ORB Extractor: Number of levels in the scale pyramid	
ORBextractor.nLevels: 8

# ORB Extractor: Fast threshold
# Image is divided in a grid. At each cell FAST are extracted imposing a minimum response.
# Firstly we impose iniThFAST. If no corners are detected we impose a lower value minThFAST
# You can lower these values if your images have low contrast			
ORBextractor.iniThFAST: 20
ORBextractor.minThFAST: 7

#--------------------------------------------------------------------------------------------
# Viewer Parameters
#--------------------------------------------------------------------------------------------
Viewer.KeyFrameSize: 0.05
Viewer.KeyFrameLineWidth: 1
Viewer.GraphLineWidth: 0.9
Viewer.PointSize: 2
Viewer.CameraSize: 0.08
Viewer.CameraLineWidth: 3
Viewer.ViewpointX: 0
Viewer.ViewpointY: -0.7
Viewer.ViewpointZ: -1.8
Viewer.ViewpointF: 500
```