UZ-SLAMLab/ORB_SLAM3 #91: OpenCV Error

zhengweiSRC, 2020-09-03:
when I run with the command "rosrun ORB_SLAM3 Stereo_Inertial Vocabulary/ORBvoc.txt Examples/Stereo-Inertial/EuRoC.yaml true"
I find the following problem 
"
OpenCV Error: Assertion failed (0 <= _rowRange.start && _rowRange.start <= _rowRange.end && _rowRange.end <= m.rows) in Mat, file /build/opencv-L2vuMj/opencv-3.2.0+dfsg/modules/core/src/matrix.cpp, line 483
terminate called after throwing an instance of 'cv::Exception'
  what():  /build/opencv-L2vuMj/opencv-3.2.0+dfsg/modules/core/src/matrix.cpp:483: error: (-215) 0 <= _rowRange.start && _rowRange.start <= _rowRange.end && _rowRange.end <= m.rows in function Mat
"

jjgr3496, 2020-09-03:
I would need more information in order to help you. Could you please find in which part of the code is failing?

zhengweiSRC, 2020-09-07:
I am

> I would need more information in order to help you. Could you please find in which part of the code is failing?

closed 2020-09-07 by zhengweiSRC