UZ-SLAMLab/ORB_SLAM3 #92: Calibration results

ihamieh, 2020-09-03:
I made ORB-SLAM3 works on my own data but I'm not sure how to output the calibration results? How to output anything out of ORB-SLAM3 other than the visual viewer provided? 

Can someone let me know? I'm not sure this is provided or am I missing something? 

Thanks,
Ish

jjgr3496, 2020-09-03:
What do you mean with "output the calibration results"? What do you want to output?

ihamieh, 2020-09-04:
Both calibration results (camera to IMU 6 DoF alignment) and the vehicle path (odometry)!

jjgr3496, 2020-09-05:
You should provide the calibration to ORB-SLAM3, so you should know it in advance. However, if you are interested in printing it in the terminal, it is always printed in terminal just after loading the ORB vocabulary. 

Regarding the odometry or camera trajectory, whenever you pass  to ORB-SLAM3 a new image with one of the available methods (take a look at `System::TrackMonocular`, `System::TrackStereo`, `System::TrackRGBD`), you always get in return the estimated pose of the camera or vehicle.

ihamieh, 2020-09-09:
Thank you @jjgr3496 I'm not sure where to look for System::TrackMonocular, System::TrackStereo, System::TrackRGBD and I'm unable to record the camera odometry and trajectory! Is there a way to record this information? I'm thinking i might be able to rosbag all the topics to view it ... will that work?

jjgr3496, 2020-09-10:
If you have doubts about `System::TrackMonocular, System::TrackStereo, System::TrackRGBD`, you always check the example programs we provide in the Examples directory. For example, looking at line 112 at `Examples/Monocular/mono_euroc.cc`, you could change it to something like:

`cv::Mat cameraPose = SLAM.TrackMonocular(im,tframe);`

so you have stored in `cameraPose`the current pose of your camera. You can do whatever you want with it (from printing it, passing it to another library, store it in a file, ..., whatever you want).

ihamieh, 2020-09-11:
@jjgr3496 thank you for the update... how you do that if you are running ROS node? How you print out those messages?

jjgr3496, 2020-10-20:
With ROS is very similar. Please take a look at line 91 at `Examples/ROS/ORB_SLAM3/src/ros_mono.cc`. Like in my last comment, you can store the returning value (i.e. the current camera pose) in some variable and use it for whatever you want.

closed 2020-12-18 by jdtardos