UZ-SLAMLab/ORB_SLAM3 #93: Stereo sensor with fisheye camera model strange results

uupks, 2020-09-04:
Thank for your great work!

It can work well when i set my stereo camera work as monocular left or right, while Stereo mode ofthen gets lost and thus go to start a new map. Once more, there is a strange phenomenon that when i check the **Follow Camera** in the Pangolin window, it will turn up while i turn down the camera actually.

The following are camera settings.

<details>
<summary>Monocular</summary>

```
Camera.type: "KannalaBrandt8"

# Left Camera calibration and distortion parameters (OpenCV)
Camera.fx: 367.72052242449245796
Camera.fy: 368.00114970554136562
Camera.cx: 374.59032089471867266
Camera.cy: 248.09098687437946751

# Kannala-Brandt distortion parameters
Camera.k1: -0.00569343151132281
Camera.k2: -0.10314481138142903
Camera.k3: 0.28308572778313068
Camera.k4: -0.22417193067033600

# Camera frames per second 
Camera.fps: 20.0

Camera.width: 752
Camera.height: 480

# Color order of the images (0: BGR, 1: RGB. It is ignored if images are grayscale)
Camera.RGB: 1
```

</details>

<details>
<summary>Stereo</summary>

```
#--------------------------------------------------------------------------------------------
# Camera Parameters. Adjust them!
#--------------------------------------------------------------------------------------------
Camera.type: "KannalaBrandt8"

# Left Camera calibration and distortion parameters (OpenCV)
Camera.fx: 367.72052242449245796
Camera.fy: 368.00114970554136562
Camera.cx: 374.59032089471867266
Camera.cy: 248.09098687437946751

# Kannala-Brandt distortion parameters
Camera.k1: -0.00569343151132281
Camera.k2: -0.10314481138142903
Camera.k3: 0.28308572778313068
Camera.k4: -0.22417193067033600

# Right Camera calibration and distortion parameters (OpenCV)
Camera2.fx: 367.41815288058677424
Camera2.fy: 367.66897447623051676
Camera2.cx: 367.68055385733310914
Camera2.cy: 242.01520025315602425

# Kannala-Brandt distortion parameters
Camera2.k1: -0.00703249954579286
Camera2.k2: -0.06953558403643909
Camera2.k3: 0.11981566666675607
Camera2.k4: -0.06016161202070901

# Transformation matrix from right camera to left camera
#Tlr: !!opencv-matrix
#   rows: 3
#   cols: 4
#   dt: f
#   data: [ 1.0000,    0.0008,    0.0007,    0.1011,
#             -0.0008,    0.9989,    0.0469,    0.0019,
#             -0.0007,   -0.0469,    0.9989,    0.0010]

Tlr: !!opencv-matrix
  rows: 3
  cols: 4
  dt: f
  data: [    
          0.999948,   0.00431242,  -0.00925538,     0.119619,
          -0.00432544,     0.999989,  -0.00138333, -0.000363243,
          0.00924933,   0.00142335,     0.999956, -0.000248677
        ]

# Lapping area between images
Lapping.left: 0
Lapping.right: 511

# Camera frames per second
Camera.fps: 20.0

# Color order of the images (0: BGR, 1: RGB. It is ignored if images are grayscale)
Camera.RGB: 1

# Close/Far threshold. Baseline times.
ThDepth: 45

Camera.bf: 43.900173
```

</details>

And some screenshots 

<details>
  <summary>Mono Left</summary>
  
[image]
[image]

</details>

<details>
  <summary>Mono Right</summary>
  
[image]
[image]

</details>

<details>
  <summary>Stereo</summary>

[image]
[image]

</details>

Thanks for your time !

jjgr3496, 2020-09-07:
Could you please check if you still get the same errors with the latest version?

yike-chen, 2020-09-09:
Does your camera type is fisheye? Why your screenshoots is flat, while my image is fisheye image?
[image]

ihamieh, 2020-09-10:
@uupks where did you get those file? How did you generate it? I'm still learning this and I notice this while searching for fisheye with ORB-SLAM3! The I came to KannalaBrandt and from my search saw your work! I was wonderning if I'm using a mono fisheye should I just try the mono example you have posted? Thanks!

ihamieh, 2020-09-10:
@uupks how did you run those file? Where did you place your new yaml files? Could you please elaborate?

ihamieh, 2020-09-10:
@yike-chen KannalaBrandt8 camera type is a fisheye but I'm not sure how to execute this in ORB-SLAM3! Do you know?

yike-chen, 2020-09-11:
> @yike-chen KannalaBrandt8 camera type is a fisheye but I'm not sure how to execute this in ORB-SLAM3! Do you know?

You can follow the [README.md](https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/master/README.md)(6. ROS Examples).

yike-chen, 2020-09-11:
> @uupks how did you run those file? Where did you place your new yaml files? Could you please elaborate?

No matter where you put it, as long as you specify the right place.

uupks, 2020-09-11:
> Does your camera type is fisheye? Why your screenshoots is flat, while my image is fisheye image?
> [image]

My stereo camera is mynt camera. The official suggested calibrating it with KannalaBrandt8 model, and the KannalaBrandt8 model is the same as fisheye model in OpenCV.

uupks, 2020-09-11:
> @uupks how did you run those file? Where did you place your new yaml files? Could you please elaborate?

You can create a yaml file named with your camera name refer to the offical yamls, and place it at the correspond directory, sucn as Examples/Monocular and Examples/Stereo. Then run ros node with your yaml file.

This is what i run : 
`rosrun ORB_SLAM3 Stereo Vocabulary/ORBvoc.txt Examples/Stereo/innno_mynt.yaml false`

uupks, 2020-09-11:
> Could you please check if you still get the same errors with the latest version?

It seems normal with the latest version.  

Thanks !

ihamieh, 2020-09-11:
> You can create a yaml file named with your camera name refer to the offical yamls, and place it at the correspond directory, sucn as Examples/Monocular and Examples/Stereo. Then run ros node with [...]

@uupks I followed your steps and it runs but it stops in the middle of the video! This is my yaml file: 

%YAML:1.0

#--------------------------------------------------------------------------------------------
# Camera Parameters. Adjust them!
#--------------------------------------------------------------------------------------------

Camera.type: "KannalaBrandt8"

Camera.fx: 367.72052242449245796
Camera.fy: 368.00114970554136562
Camera.cx: 374.59032089471867266
Camera.cy: 248.09098687437946751

Camera.k1: -0.00569343151132281
Camera.k2: -0.10314481138142903
Camera.k3: 0.28308572778313068
Camera.k4: -0.22417193067033600

Camera.width: 752
Camera.height: 480

# Camera frames per second 
Camera.fps: 30.0

# IR projector baseline times fx (aprox.)
Camera.bf: 40.0

# Color order of the images (0: BGR, 1: RGB. It is ignored if images are grayscale)
Camera.RGB: 1

# Close/Far threshold. Baseline times.
ThDepth: 40.0

# Deptmap values factor
DepthMapFactor: 1.0

#--------------------------------------------------------------------------------------------
# ORB Parameters
#--------------------------------------------------------------------------------------------

# ORB Extractor: Number of features per image
ORBextractor.nFeatures: 1000

# ORB Extractor: Scale factor between levels in the scale pyramid 	
ORBextractor.scaleFactor: 1.2

# ORB Extractor: Number of levels in the scale pyramid	
ORBextractor.nLevels: 8

# ORB Extractor: Fast threshold
# Image is divided in a grid. At each cell FAST are extracted imposing a minimum response.
# Firstly we impose iniThFAST. If no corners are detected we impose a lower value minThFAST
# You can lower these values if your images have low contrast			
ORBextractor.iniThFAST: 20
ORBextractor.minThFAST: 7

#--------------------------------------------------------------------------------------------
# Viewer Parameters
#--------------------------------------------------------------------------------------------
Viewer.KeyFrameSize: 0.05
Viewer.KeyFrameLineWidth: 1
Viewer.GraphLineWidth: 0.9
Viewer.PointSize:2
Viewer.CameraSize: 0.08
Viewer.CameraLineWidth: 3
Viewer.ViewpointX: 0
Viewer.ViewpointY: -0.7
Viewer.ViewpointZ: -1.8
Viewer.ViewpointF: 500

lishanggui, 2020-09-22:
Hi @uupks Are you using realsense t265 for testing? Perform factory calibration on imu according to the official introduction of realsense. When t265 was placed still in different directions, his acceleration was not equal to the acceleration of gravity (the displayed value was 8-10). I want to ask if you have this problem? Will this affect the vio experiment.

thanks

closed 2020-10-11 by uupks